rockowitz/ddcutil #72: Sample programs make error (Makefile:15: missing separator)

aryoda, 2018-12-30:
I sample programs a really helpful to me to learn the `ddcutil` API.

I have just pulled the current version 0.9.4 and tried to build the sample programs in the sub folder
`src/sample_clients` but the recommended way of building them (as described here: http://www.ddcutil.com/api_main/#Documentation)

`make check # in the above sub folder`

fails with the message

> Makefile:15: *** missing separator.  Stop.

Line 15 contains an (auto)make macro code which seems to cause the problem (as also later starting at line 559):

`@SET_MAKE@`

There is no `autogen.sh` in the folder and calling `aclocal`, `autoconf` and `automake` manually does not solve the problem.

What am I missing to build the sample programs?

PS: The documentation in http://www.ddcutil.com/api_main/#Documentation mentions an old folder name for the sample programs (`src/sample_programs`, it is now `src/sample_clients`).

rockowitz, 2018-12-30:
autotools, !$#

Unless I can recreate the problem here I'm just stabbing in the dark.

The missing separator message typically occurs when spaces are used 
instead of a tab in a Makefile.am, but if you haven't touched 
Makefile.am I should see the error as well, which I do not.

Usually running the sequence aclocal, autoconf, automake, ./configure 
clears up problems when moving from one distribution to another, but 
you've done that.

What distribution are you running on?

What happens if you run "make check" from the top level directory 
instead of the sample_programs subdirectory? (though both should work)

I've fixed the documentation error.  Thanks for pointing it out.

Sanford

aryoda, 2018-12-30:
Thanks for trying it out on your side!

I think now the problem is on my side (using a git submodule with a pinned version of `ddcutil`)
since a fresh clone + build does work, **sorry for being too fast and opening an issue before trying this out myself before (waste of your time)** :-(

I will investigate the issue and append the solution when I find it.

I am using Ubuntu 18.04 (64 bit) BTW.

PS: The fresh clone + build did work (`make check` in the repo root did build the sample programs):

```
$ make

Making all in sample_clients
make[3]: Entering directory 'temp/ddcutil/src/sample_clients'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory 'temp/ddcutil/src/sample_clients'
make[2]: Leaving directory 'temp/ddcutil/src'

$ make check (in repo root folder)

make[2]: Entering directory 'temp/ddcutil/src/sample_clients'
make  laclient demo_capabilities demo_display_selection demo_feature_list demo_get_set_vcp demo_global_settings demo_profile_features demo_redirection demo_vcpinfo
make[3]: Entering directory 'temp/ddcutil/src/sample_clients'
  CC       clmain.o
  CCLD     laclient
  CC       demo_capabilities.o
  CCLD     demo_capabilities
  CC       demo_display_selection.o
  CCLD     demo_display_selection
  CC       demo_feature_list.o
  CCLD     demo_feature_list
  CC       demo_get_set_vcp.o
  CCLD     demo_get_set_vcp
  CC       demo_global_settings.o
  CCLD     demo_global_settings
  CC       demo_profile_features.o
  CCLD     demo_profile_features
  CC       demo_redirection.o
  CCLD     demo_redirection
  CC       demo_vcpinfo.o
  CCLD     demo_vcpinfo
make[3]: Leaving directory 'temp/ddcutil/src/sample_clients'
make[2]: Leaving directory 'temp/ddcutil/src/sample_clients'
make[1]: Leaving directory 'temp/ddcutil/src'
```

aryoda, 2018-12-30:
OK, I have fixed the problem.

It was related to how Git maintains submodules (I am using `ddcutil` as a submodule):
Some mess in there with different versions (git submodules are quite complicated IMHO).

Solution: I have applied the steps to update the submodule from upstream correctly now, see the good description here:

https://stackoverflow.com/questions/5828324/update-git-submodule-to-latest-commit-on-origin

After building the ddcutil the normal way I have applied a `make check` in the `ddcutil` (submodule) root folder (NOT in the src/sample_clients folder) and the sample programs were built correctly.

A good check for the correct `ddcutl` version of the submodule is:

```
$ # in the root repo folder
$ git submodule status
1834de60e8658c3c86f9d1e27b1e94830ffcab27 libddcutilcpp/ddcutil_submodule (v0.9.4)
```

closed 2018-12-30 by aryoda