rockowitz/ddcutil #74: The assertion failure in format_feature_detail_audio_treble_bass, LG 23EA73

Ovsyanka, 2019-01-09:
I got en error in output of `ddcutil interrogation`:

```
ddcutil: vcp_feature_codes.c:2031: format_feature_detail_audio_treble_bass: Assertion `vcp_version_gt(vcp_version, DDCA_VSPEC_V21)' failed.
```

Display: LG 23EA73
Video card: nvidia GTX-650
Video driver: proprietary.
[ddcutil-interrogate.log](https://github.com/rockowitz/ddcutil/files/2741785/ddcutil-interrogate.log)

rockowitz, 2019-01-10:
Thank you for reporting the bug.  It was introduced in the major code 
restructuring of release 0.9.3. There is a fix in the current 
development branch, 0.9.5.  If you are able to build from github, please 
download and build that branch and send the result of "ddcutil 
interrogate 2>&1" as an attachment.

Sanford

Ovsyanka, 2019-01-10:
I checked 0.9.5 and yes, the error gone.
[interrogate-0.9.5.txt](https://github.com/rockowitz/ddcutil/files/2744920/interrogate-0.9.5.txt)

rockowitz, 2019-01-10:
Thanks for testing.  Would you be willing to perform some additional 
tests for me?

As noted previously, your monitor never sets the "Unsupported feature" 
flag in a feature value response, instead returning all bytes 0 in a 
valid response.  I had added a check for this situation to monitor 
detection, but it doesn't seem to be working properly in your current 
interrogate output.    The latest version of branch 0.9.5 contains some 
debug statements to address this case.

Can you build using the latest 0.9.5 branch and send the output of the 
following command:

   ddcutil detect --verbose

Thanks,
Sanford

Ovsyanka, 2019-01-10:
Yes, I would be happy.

Here the file [detect-0.9.5.txt](https://github.com/rockowitz/ddcutil/files/2747212/detect-0.9.5.txt) and there was two errors in command line:

    Open failed for /dev/usb/hiddev0: errno=EACCES(13): Permission denied
    Open failed for /dev/usb/hiddev1: errno=EACCES(13): Permission denied

Maybe it would be more productive to use some realtime messenger for that. If you like it - just write me your suggestions on email. Because I don't think that it would be useful development process to be presented here.

rockowitz, 2019-01-27:
The /dev/usb/hiddev EACCESS errors are expected on ***ddcutil detect --verbose***.  They occur when checking to see if any hiddev device represents a monitor implementing the USB Monitor Control Class Specification for Monitor Control Command Set (MCCS) communication.

rockowitz, 2019-01-27:
As noted, the assertion failures when reporting VCP features x8F and x91 (Audio Treble/Bass) have been fixed in the 0.9.5 development branch.

closed 2019-01-27 by rockowitz

m-majetic, 2021-01-17:
Getting this error with version 0.9.9 on Debian 11 - bullseye.

```
Getting data for non-table VCP code 0x93 - Audio Balance L/R:
Raw value: opcode=0x93, mh=0x00, ml=0x64, sh=0x00, sl=0x32, max_val=100 (0x0064), cur_val=50 (0x0032)
ddcutil: vcp_feature_codes.c:1769: format_feature_detail_audio_treble_bass: Assertion `code_info->vcp_code == 0x8f || code_info->vcp_code == 0x91' failed.
```

I can provide more logging if you are interested in resolving this issue; again. :slightly_smiling_face:

rockowitz, 2021-01-17:
Fixed in upcoming release 1.0.0.  Build from branch 1.0.0-dev.

m-majetic, 2021-01-19:
@rockowitz 

Keep in mind that I have little to no experience in C and compiling, I am only following the guidelines on the ddcutil's website.

I got so far with building from the 1.0.0-dev branch:

```
make[3]: Entering directory '/home/marko/build_packages/ddcutil/src/app_sysenv'
  CC       query_sysenv.lo
  CC       query_sysenv_access.lo
  CC       query_sysenv_base.lo
  CC       query_sysenv_dmidecode.lo
  CC       query_sysenv_i2c.lo
query_sysenv_i2c.c: In function ‘try_single_getvcp_call’:
query_sysenv_i2c.c:204:14: error: statement with no effect [-Werror=unused-value]
       rc = 0;raw_scan_i2c_devices
              ^~~~~~~~~~~~~~~~~~~~
query_sysenv_i2c.c:204:34: error: expected ‘;’ before ‘}’ token
       rc = 0;raw_scan_i2c_devices
                                  ^
                                  ;
    }
    ~                              
cc1: all warnings being treated as errors
make[3]: *** [Makefile:547: query_sysenv_i2c.lo] Error 1
make[3]: Leaving directory '/home/marko/build_packages/ddcutil/src/app_sysenv'
make[2]: *** [Makefile:1086: all-recursive] Error 1
make[2]: Leaving directory '/home/marko/build_packages/ddcutil/src'
make[1]: *** [Makefile:700: all-recursive] Error 1
make[1]: Leaving directory '/home/marko/build_packages/ddcutil'
make: *** [Makefile:517: all] Error 2
```

rockowitz, 2021-01-19:
My apologies.  "raw_scan_i2c_devices" is a stray paste introduced in 
change 0b33ad, which should just have bumped the copyright year. A 
corrected query_sysenv_i2c.c file has been pushed to branch 1.0.0-dev.

m-majetic, 2021-01-19:
@rockowitz That worked.

Thanks !