rockowitz/ddcutil #75: Proprietary nvidia driver requires that module i2c_dev be loaded, contrary to documentation

Ovsyanka, 2019-01-09:
The interrogate output says about i2c_dev module that "Using only proprietary nvidia or fglrx driver. Module i2c_dev not required." But without it (i2c_dev module) the command `ddcutil detect` doesn't found monitor. And after I load i2c_dev module manually `modprobe i2c_dev` - then it works.

Display: LG 23EA73
Video card: nvidia GTX-650
Video driver: proprietary, nvidia-dkms  415.25-5

rockowitz, 2019-01-31:
The ***environment*** command (and hence the ***interrrogate*** command) in branch 0.9.5 has been fixed to regard the i2c_dev module as required when using the proprietary Nvidia driver.   Thank you for reporting the bug.

Note: The online documentation at www.ddcutil.com will not be updated until the next release.

Ovsyanka, 2019-01-31:
As I can see, that phrase still there in the environment output. I reinstalled [current](https://github.com/rockowitz/ddcutil/archive/0.9.5.zip) (670c598) 0.9.5 version and run it.
[environment-0.9.5.txt](https://github.com/rockowitz/ddcutil/files/2817172/environment-0.9.5.txt)

rockowitz, 2019-01-31:
Igor,

When I look at file query_sysenv_modules.c on github in branch 0.9.5, 
commit 670c598), line 181 has been changed from a call to 
only_nvidia_or_fglrx() to a call to only_fglrx(), and the message at 
line 184 is changed accordingly.

The changed code should not produce your output. What am I missing here?

Sanford

Ovsyanka, 2019-01-31:
Yeah, sorry, it was my mistake. I am not expirienced enought with building packages and eventually I used old source code.

Now I can confirm that this message doesn't appear anymore. I think I should close the issue then.

Thank you!

closed 2019-01-31 by Ovsyanka