rockowitz/ddcutil #76: Move typedefs from ddcutil_c_api.h to ddcutil_types.h to support decent API wrapping in C++

aryoda, 2019-01-12:
I am trying to wrap (encapsulate) all ddcutil C API (function) calls in a single C++ class to centralize all ddcutil calls (for central logging, error recognition and mocking).

To avoid inadvertently calling the ddcutil C API directly from C++ (instead of always using the C++ wrapper class) I want to get rid of including the `ddcutil.c_api.h` elsewhere except in my wrapper cpp file (but only including the `ddcutil_types.h` in the client classes of my wrapper class).

This does currently not work since I have found two `typedef`s in the `ddcutil_c_api.h`:

1. DDCA_Build_Option_Flags
2. DDCA_Capture_Option_Flags

Would it be possible to move these typedefs into to the `ddcutil_types_h` file?

THX!

rockowitz, 2019-01-14:
Aryoda,

These 2 typedefs are currently in in ddcutil.c_api.h for clarity because 
they are used only as input  or output of API functions, and are used 
nowhere else in the ddcutil code.  However, it  would not do great 
violence to the layering to move the definitions to ddcutil_types.h if 
that would help you.  I'll make the change in release 0.9.5.

Sanford

aryoda, 2019-01-14:
Great! I can send a pull request if tell me which branch to branch from and merge/"push" into...

I can also found another small documentation issue that I would fix with a (separate) pull request...

rockowitz, 2019-01-15:
Branch 0.9.5 is the current working branch.

It contains the typedef moves you requested for DDCA_Build_Option_Flags 
and DDCA_Capture_Option_Flags.

aryoda, 2019-01-17:
@rockowitz I need your advice regarding encapsulation of `ddcutil` calls in C++ for above typedefs...

I am currently wrapping the calls of `ddcutil` in a more OO-style API lto

* replace returned status codes with an exception
* return the real result object using a std::unique_ptr (smart pointer) with a custom deleter to avoid resource leaks

like this code example shows:

```
std::unique_ptr<void, DDCA_Status (*)(void *)>
DdcUtilWrapper::ddca_create_dispno_display_identifier(int dispno)
{
    DDCA_Display_Identifier did;

    DDCA_Status res = ::ddca_create_dispno_display_identifier(dispno, &did);

    if (res != DDCRC_OK)
        throw DDCException(108, res, "ddca_create_dispno_display_identifier: " + std::string(ddca_rc_desc(res)));

    std::unique_ptr<void, DDCA_Status (*)(void *)> did_up(did, &::ddca_free_display_identifier);

    return did_up;
}
```

The dirty part is, that he returned unique_ptr does not indicate that it stores a pointer to a display (handle) but just a pointer to a void type (= "untyped").

I would prefer having a typed unique_ptr like `std::unique_ptr<DDCA_Display_Identifier, DDCA_Status (*)(void *)>` instead but there is no typedef in `ddcutil` to provide this "alias", just:

`typedef void * DDCA_Display_Identifier;`

For me (just in C++) it would be better to have something like this:

`typedef void DDCA_Display_Identifier;  // without a "*" (pointer!)`

and a C function signature like

```
DDCA_Status ddca_create_dispno_display_identifier(
      int                      dispno,
      DDCA_Display_Identifier** did_loc);  // pointer-pointer now!
```

Obviously this would change the declared signature and existing client-side code.

I am also thinking about a C++-only solution

* using a typedef in C++ only for the unique_ptr
* using a typedef in C++ only the declare an alias for void
* ...

Do you have any idea for a possible (`ddcutil`-compatible) solution? Telling me that I am on the wrong track is also a valid feed-back ;-)

rockowitz, 2019-01-20:
Several comments:

1) Your DDCException class should allow for the possibility that the 
failing function has detailed error information, as described by the 
(recursive) DDC_Error_Detail struct and returned by function 
ddca_get_error_detail().  There are currently just a few API calls that 
provide detailed error information, but this is expected to grow.  It 
addresses situations where there can be multiple errors of interest to 
the callers, for example:
   - status code DDCRC_RETRY, what were the underlying errors? (not yet 
implemented)
   - errors parsing the capabilities string (ddca_get_error_detail())
   - errors on loading a VCP file (ddca_set_profile_related_values())

In branch 0.9.5, I have modified those functions in ddcutil_c_api.h that 
return a DDCA_Status to always clear the thread error detail at the 
start of execution.  Therefore, in your code that converts a status code 
to an exception, you can always call ddca_get_error_detail() without 
fear of getting error detail from a prior function call.  Of course, 
that function will (at this point at least) return NULL in most cases, 
as only a few API calls generated detailed error information.

2) In general, I've come to prefer not using pointers in typedefs, as it 
makes clear in function definitions that a pointer is being passed. The 
major exception is the case under consideration, where we're passing 
around an opaque reference. "typedef void DDCA_Display_Identifier;"  
would allow the following declaration, which is illegal.

     DDCA_Display_Identifier id;
     i.e.
     void id;

3) As a general comment, it appears you are attempting to wrap the 
entire API in a single DdcUtilWrapper class.  I would suggest this is 
not very OO.   Rather, think about what classes naturally arise.  For a 
very crude and incomplete class hierarchy, see the proof of concept code 
in source file src/cffi/ddc_cffi.py which is part of a my exploratory 
code for a building a python interface using cffi.

Regards,
Sanford

closed 2019-03-05 by rockowitz