rockowitz/ddcutil #78: Cannot change Gamma setting on AW3418DW

YamashitaRen, 2019-01-25:
Default value is "30720". I just changed it to "23040" using "ControlMyMonitor" on Windows.

```
# ddcutil -v getvcp 72
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Getting data for non-table VCP code 0x72 - Gamma:
Raw value: opcode=0x72, mh=0x00, ml=0x00, sh=0x5a, sl=0x00, max_val=0 (0x0000), cur_val=23040 (0x5a00)
VCP code 0x72 (Gamma                         ): SL: 0x00 ,  SH: 0x5a
```

If I try to change it back to "30720", I get an error : 
```
# ddcutil setvcp 72 30720
Number must be in range 0..255 (for now at least):  30720
Invalid VCP value: 30720
```
```# ddcutil setvcp 72 0x78
Verification failed for feature 72```

Only the values "0" and "0x00" are accepted, which I obviously don't want to use.

rockowitz, 2019-01-25:
Thank you for bringing this case to my attention.  I hope that you'll be 
able to help with testing the changes needed to properly handle feature 
x72 (Gamma).   (The spec for this feature is unusual in that it goes on 
for 4 pages, and it's unusual in that setting the high order byte matters.)

Can I ask you to send me the output of "ddcutil interrogate" for you 
system?   Please send it as an attachment.   Thank you.

Sanford

YamashitaRen, 2019-01-27:
Sure, let's do it !
[ddcutil.log](https://github.com/rockowitz/ddcutil/files/2800258/ddcutil.log)

rockowitz, 2019-01-28:
Great,

Branch 0.9.5 has been updated to allow for 2 byte numbers to be 
specified as setvcp values.
The numbers can be specified either in decimal or hexadecimal, high 
order byte first,

e.g. setvcp 72 30720
e.g. setvcp 72 x7800     (sh=x78, sl=x00)

That's the easy part.

What will likely not appear in release 0.9.5:
- interpreting the bytes on getvcp output
- capabilities string interpretation; the byte values associated with 
feature x72 in the capabilities string have a special interpretation, 
they are not  a list of allowed values
- feature metadata in the API will have to be extended to handle this case
- feature x72 will require a special widget for display in the GUI (ddcui)

I also notice a deviance from spec by this monitor that I've never seen 
before.  Instead of setting the "unsupported feature" flag when 
responding to a query of an unsupported feature, as per the DDC/CI  
spec, this monitor fails the IO operation with status -EIO.

Lastly, it looks like you're using the Nvidia proprietary driver, that 
module i2c_dev is loaded but not built into the kernel.  Can you try 
starting the system without i2c_dev loaded?  I ask because I had thought 
that module i2c_dev is unnecessary when using the proprietary nvidia 
driver, but a recent user report suggests that it is in fact required.   
All my systems here that use the nvidia driver have i2c_dev built into 
the kernel so that case is a bit hard for me to test.  Thanks.

Sanford

YamashitaRen, 2019-01-28:
Amazing !
I can confirm that setvcp now works with 2 byte numbers as values.

Actually my system doesn't load i2c_dev at boot (I have to change that). I had to load it manually in order to have ddcutil detect my monitor.
So, yes, I can confirm it. i2c_dev is required if using current nvidia driver.

Feel free to ping me if you need more testing.

rockowitz, 2019-01-31:
The current 0.9.5 branch contains additional  support for feature x72 
(gamma)

The getvcp command now interprets the reported gamma value.

The capabilities command interprets the "values" field for feature x72 
in the capabilities string returned by the monitor.

Comments appreciated.  Beyond does it work, could the message formatting 
be improved?

rockowitz, 2019-01-31:
The environment command has bee changed to require the i2c_dev driver be 
loaded when using the proprietary Nvidia driver.   This change is in the 
the latest 0.9.5 branch. The online documentation at www.ddcutil.com 
will be changed with the next release.  Thank you testing.

Sanford

rockowitz, 2019-02-04:
>
> Can't really comment about the formatting. I get the values, that's 
> all I need.
> Can't comment about the interpreted gamma value since my monitor OSD 
> doesn't have the setting...
>
> |ddcutil -v getvcp 72|
>
> |Getting data for non-table VCP code 0x72 - Gamma: Raw value: 
> opcode=0x72, mh=0x00, ml=0x00, sh=0x5a, sl=0x00, max_val=0 (0x0000), 
> cur_val=23040 (0x5a00) VCP code 0x72 (Gamma ): 0x5a00 - Mode: White 
> absolute adjustment (sl=0x00), gamma=1.90 (sh=0x5a) |
>
> Wouldn't it be better if we didn't get "max_val=0 (0x0000)" since the 
> max value is obviously wrong ?
The raw value is shown because you have specified verbose output. It's 
just that, the raw data, before any interpretation is applied.

> |ddcutil capabilities|
>
> |Feature: 72 (Gamma) Values: Full range of absolute adjustment 
> supported (0xff) Absolute tolerance: ideal (=0x00) Native gamma: 2.20 
> (0x78) |
>
> Are the first two values necessary ? Since we only change the gamma...

First, a general comma.  The term "Values" is a bit of a misnomer for 
feature x72.  Normally the values substring in the capabilities string 
is just a list of possible values for a simple non-continuous feature.  
In the case of feature x72, the values substring describes the possible 
values, since the set of possible values is huge. Values can be 
specified as absolute gamma, or as relative adjustments to native 
gamma.  They can be any value ("Full range"), limited to a range of 
values, or limited to specific values.  So all this information is 
needed to describe what values can be set.   For details, see pages 
43-46 of the MCCS spec (https://milek7.pl/ddcbacklight/mccs.pdf).
>
>     The environment command has bee changed to require the i2c_dev
>     driver be loaded when using the proprietary Nvidia driver.
>     What do you mean by this ? I don't have any error message if
>     i2c_dev isn't loaded (beside "Display not found")...
Please run "ddcutil environment --verbose" and send the output as an 
attachment.

YamashitaRen, 2019-02-04:
I see. Thanks for all the explanations.

> Please run "ddcutil environment --verbose" and send the output as an attachment.

Oh ! Sorry, I was using `ddcutil capabilities`. `ddcutil environment` indeed suggests to load i2c-dev.

closed 2019-03-05 by rockowitz