rockowitz/ddcutil #79: ddca_free_* functions

rockowitz, 2019-01-31:
In a private email, Jurgen Altfeld (aryoda) raised the following questions: 

1. Do I (always or never) have have to call `ddca_free_*` functions if the return status of a normal `ddca` function was **not** `DDCRC_OK`?

2. Do all `ddca_free_*` functions (which release resources) work idempotent so that I can call them more than once without any harm?

3. Is there an implicit order I have to stick to when calling different `ddca_free_*` functions?

This caused me to review the ddca_free_* functions and make some changes for consistency, which can be found in the latest 0.9.5 development branch.

Re question (1), if a function that creates a resource returns other than DDCRC_OK (0), then the resource is not created and need not be freed.

Re question (2), the answer is that freeing an already freed resource is an error.  (Consider the basic, and often underlying, case where memory is malloc'd, freed, and then freed again using the original pointer.  You'd expect a segfault.) 

Re question (3), the only case in which order matters is that a DDCA_Display_Ref should not be freed if it is associated with an open DDCA_Display_Handle.

More general comments:

ddca_free_* functions fall into two groups, those that specify internal library data structures and are referenced by an opaque pointer, and those that specify client side data structures and are referenced by a normal pointer. 

Those that specify client side data structures are essentially convenience functions, relieving the client programmer of the details of freeing what may be a complex data structure.  These functions always return void.  If passed a NULL pointer, they do nothing.  The functions in this group are: 
ddca_free_error_detail(), ddca_free_display_info_list(), ddca_free_parsed_capabilities(), ddca_free_table_vcp_value(), ddca_free_any_vcp_value(), and ddca_free_feature_metadata(). 

Functions that free internal library data structures take an opaque pointer as an argument.  These functions return a DDCA_Status.  They return DDCRC_ARG if the pointer is invalid.  If the pointer is NULL, the function does nothing and DDCRC_OK is returned. Depending on the function, additional status codes are possible.  Whether or not memory is actually freed is a hidden implementation detail, as some opaque pointers reference persistent internal data structures.    The functions in this group are: 
ddca_free_display_identifier(), ddca_free_display_ref(), ddca_close_display().

aryoda, 2019-02-03:
Hi Sanford, thanks for having taken the time for these great improvements in the documentation and API. The concept of internal ("opaque") and client-side data structures is very concise to understand and use.

What I am still worried about is the correct usage of the opaque `DDCA_Display_Ref` and `DDCA_Display_Handle` because a handle seems to contain a display reference internally.

My questions:

Can I safely assume that
1. a `DDCA_Display_Ref` is stable over time (unless I unplug the display of course)?
    In other words: When does a display reference become invalid?
2. the `DDCA_Display_Ref` that is internally used by a `DDCA_Display_Handle` is **not** freed if I free the display handle with `ddca_close_display`?
3. I can reuse an existing `DDCA_Display_Ref` for multiple subsequent calls of `ddca_open_display` and `ddca_close_display`
3. an "open" (= not freed) `DDCA_Display_Ref` does not block any function call of the `ddcutil` API
4. it makes no difference regarding the answers of 1 to 4 if I aquire the display reference via `ddca_get_display_info_list2` or `ddca_create_display_ref`?

**Background:** I want to minimize the time slot a display handle is open and "blocking" the communication channel to the display. Therefore I have to open and close it again and again each time I read or write display features (like you do it in your [VcpThread](https://github.com/rockowitz/ddcui/blob/master/src/nongui/vcpthread.cpp) class):

```
// DDCA_Display_Ref _dref;   // is an attribute of the class
DDCA_Display_Handle dh;
DDCA_Status ddcrc = ddca_open_display(this->_dref, &dh);
... // communicate with the display
ddcrc = ddca_close_display(dh);
```
There are two major ways of getting a display ref:
1. `ddca_get_display_info_list2` (in the member `dref`) to enumerate all available displays
2. `ddca_create_display_ref` (using a display identifier)

The documentation of (2) says:
> Normally, this is a permanently allocated #DDCA_Display_Ref
> created by monitor detection and does not need to be freed.
> Use #ddca_free_display_ref() to safely free.

rockowitz, 2019-02-04:
Again, more good questions.

Basically, a display reference contains the accumulated, persistent 
information about a display, e.g. its I2C bus number or USB device 
number, parsed EDID, parsed capabilities string, etc. (Some of this 
information, such as the capabilities string, is expensive to query, so 
I want to do it once and save the result.) A display handle, on the 
other hand, is something that you can perform IO on.   Put another way, 
the display reference both tells the display handle the operating system 
path to the display, and accumulates information obtained when a handle 
is open.

There's accumulated cruft in the API, reflecting the command line origin 
of ddcutil.   In the shared library all the connected monitors are 
"inventoried" and persistent display references are created for each.   
These are returned by ddca_get_display_info_list2().  At present, all 
that ddca_create_display_ref() does is find the existing persistent 
display reference.  It might better be named ddca_find_display_ref().  
And ddca_free_display_ref() is essentially a no-op because in the shared 
library all display references are (at present) persistent.

In the future, it would be nice if the shared library were able to 
detect when monitors are connected or disconnected.  In that case new 
display references would be created or existing ones marked as invalid.

> Hi Sanford, thanks for having taken the time for these great
> improvements in the documentation and API. The concept of internal
> ("opaque") and client-side data structures is very concise to [...]
Correct.  That is the intent.  (ddca_free_display_ref() surfaces an 
internal function that actually useful only in the command line 
version.  It should probably be removed from the API.)
>
>  1. the |DDCA_Display_Ref| that is internally used by a
>     |DDCA_Display_Handle| is *not* freed if I free the display handle
>     with |ddca_close_display|?
Correct.
>
>  1. I can reuse an existing |DDCA_Display_Ref| for multiple subsequent
>     calls of |ddca_open_display| and |ddca_close_display|
|Correct. That is the intent.|
>
>  1. an "open" (= not freed) |DDCA_Display_Ref| does not block any
>     function call of the |ddcutil| API
Correct. "open"  as a concept does not apply to a DDCA_Display_Ref. It 
is just the accumulated descriptive information about a display.

>  1. it makes no difference regarding the answers of 1 to 4 if I aquire
>     the display reference via |ddca_get_display_info_list2| or
>     |ddca_create_display_ref|?

Correct.
> *Background:* I want to minimize the time slot a display handle is
> open and "blocking" the communication channel to the display.
> Therefore I have to open and close it again and again each time I [...]

As a practical matter, repeatedly opening and closing a display for IO, 
which might seem profligate,  is actually cheap.  The time spent in 
sleeps mandated by the DDC/CI protocol dwarfs all other elapsed 
execution time.

aryoda, 2019-02-04:
Perfect, exactly what I need, thanks!

I am using the `dref` member of the `DDCA_Display_Info[_List]` typed argument from`ddca_get_display_info_list2` to get the my display references.

One last question to be absolutely sure:

Is the `DDCA_Display_Ref` in `DDCA_Display_Info->dref` provided by `ddca_get_display_info_list2`  still valid (usable) even after calling `ddca_free_display_info_list`?

I would say "yes", if it is an "persistent" internal data structure.

PS:

> In the future, it would be nice if the shared library were able to detect when monitors are connected or disconnected.
>  In that case new display references would be created or existing ones marked as invalid.

In fact I have the same problem in the C++ API too and I am thinking about marking objects as "invalid" to throw exceptions instead of deleting objects causing stale pointers at the client-side (if the client is storing a pointer eg. in a class attribute). Smart pointers do **not** protect from using stale pointers...

In real life notebooks with docking stations are a common use case of connecting/disconnecting (external) displays...

rockowitz, 2019-02-04:
> Perfect, exactly what I need, thanks!
>
> I am using the |dref| member of the |DDCA_Display_Info[_List]| typed
> argument from|ddca_get_display_info_list2| to get the my display
> references.
>
> One last [...]
Correct.  Struct DDCA_Display_Info contains a persistent 
DDCA_Display_Ref.  The remaining fields are information extracted from 
the display ref that may be of use to the client.  It just kind of 
grew.  Freeing a DDCA_Display_Info frees the copies of the extracted 
information, not the DDCA_Display_Ref.   An alternative, and perhaps 
clearer, design would be:

   ddca_get_display_refs()  that returns just an array of DDCA_Display_Ref

and then individual functions to extract individual bits of information, 
e.g.

   char * ddca_dref_get_model_name(DDCA_Display_Ref dref);

or perhaps a single function to extract all the publicly visible detail, 
e.g.

   DDCA_Status ddca_get_dref_detail(DDCA_Display_Ref dref, 
DDCA_Display_Details * details_loc);

where DDCA_Display_Detail is a struct similar to DDCA_Display_Info. But 
then the client is left with the task of associating the 
DDCA_Display_Detail with the DDCA_Display_Ref, so it would be convenient 
to include a DDCA_Display_Ref field in DDCA_Display_Detail, and we're 
sort of back where we started.

Thoughts?

> PS:
>
>     In the future, it would be nice if the shared library were able to
>     detect when monitors are connected or disconnected.
>     In that case new display references would be created or [...]

Agreed.  It would entail listening for udev device change notifications, 
acting on those that represent monitor changes, modifying internal data 
structures, and notifying the client of the changes.  A substantial hunk 
of work, I expect.

aryoda, 2019-02-04:
I have just pulled the changes and this small renaming from `ddca_create_display_ref()` to `ddca_get_display_ref()` is really a great thing to make it clear that there is nothing to free.

> Freeing a DDCA_Display_Info frees the copies of the extracted information, not the DDCA_Display_Ref.   An alternative, and perhaps clearer, design would be:

>      ddca_get_display_refs()  that returns just an array of DDCA_Display_Ref

> and then individual functions to extract individual bits of information [...]

As client programmer (using the `ddcutil` API) I have two basic use cases:
1. "Enumerate": Give me a list of all available displays with some helpful human-readable information
    (to pick one of the use case 2)
2. "Communicate": Pick a display and communicate with it (get and set values)

`ddca_get_display_info_list2` is absolutely OK for me for use case 1 since
- I get the `dispno`, `model_name` and `sn` to build a human readable "identifier" string stable over time to show the user
- I get the display handle to open the display the user has chosen so that I can get and set the display feature values

I personally see no need for an isolated `ddca_get_display_ref()` in my use cases - other use cases of other developers my require this (better named function) anyhow.

What would be really helpful is an update of the display ref related docs in the header files (as you already have noted in the commit message) but for now the thing is clear to me thanks to your answers above.

Thanks for taking the time - I still try to avoid looking into the `ddcutil` source code (except the demos) to get a feeling of the API from a knowing-nothing user perspective...

rockowitz, 2019-02-06:
Comments interspersed...

> I have just pulled the changes and this small renaming from
> |ddca_create_display_ref()| to |ddca_get_display_ref()| is really a
> great thing to make it clear that there is nothing to free. [...]

You've described my use case that led to DDCA_Display_Info_List 
containing more and more information.   Since the "fatness" of 
ddca_get_display_info_list2() doesn't seem to be an issue, we'll leave 
the API as is.
>
> What would be really helpful is an update of the display ref related 
> docs in the header files (as you already have noted in the commit 
> message) but for now the thing is clear to me thanks to your answers 
> above.
The inline documentation in ddcutil_types.h and ddcutil_c_api.h has been 
updated.   Also declarations for functions deprecated in release 0.9.0 
have been removed.

closed 2019-03-05 by rockowitz