rockowitz/ddcutil #80: ddcutil locks up kernel during any command

jrsa, 2019-02-18:
any ddcutil command deadlocks in the kernel and eventually I need to hold the power button. This is under ubuntu 18.04, on a MacBookPro10,1 with NVidia graphics and a Dell U2518D monitor.

This is the kernel log during one execution of `ddcutil detect`, from the first error to machine restart. Let me know if there is any more info i can provide, I did not yet try a verbose mode or anything like that.

[ddcutil_hang_kernlog.txt](https://github.com/rockowitz/ddcutil/files/2873865/ddcutil_hang_kernlog.txt)

rockowitz, 2019-02-18:
Thanks for the report.

Can you try running with the "--trace i2c" option and at least note the 
last few lines of output before the hang?  That will give me a better 
sense of the point where ddcutil is triggering the hang.

Relatedly, how far along does the "ddcutil environment" command get 
before it hangs?

Do you see a similar problem with ddccontrol or i2cdetect?

Sanford

jrsa, 2019-02-18:
`(invoke_i2c_writer) writer=read_writer, bytes_to_write=00
(invoke_i2c_writer) Returning rc=OK(0): success
(invoke_i2c_reader) reader=read_reader, bytect=1
(invoke_i2c_reader) Returning rc=OK(0): success
(invoke_i2c_reader) reader=read_reader, bytect=1`

the is the last thing i see from ddcutil

jrsa, 2019-02-18:
`ddcutil environment` prints the same thing before it hangs up, if i suppy the `--trace i2c` argument. If not, it doesn't print anything out.

interestly, I can also reproduce this hang by running `i2cdetect -y 10`. this command stops when it gets to address 0x37.

rockowitz, 2019-02-18:
That i2cdetect also hangs the system strongly indicates that the problem 
is in the nouveau driver. During display detection the /dev/i2c bus 
probing (currently) used by ddcutil is similar to that in i2cdetect.

A few more thoughts:

1) Your i2cdetect command probes device /dev/i2c-10.  Is that really 
where the external monitor is located?    If not, what happens when you 
use the proper bus number on i2cdetect?  Does the hang occur with all 
/dev/i2c buses, or just the one associated with the laptop display?

2) If i2cdetect does not hang with the proper bus number, you could try 
using the ddcutil --bus option to specify the /dev/i2c device number for 
the monitor.  This avoids probing all other /dev/i2c devices.

3) Display detection has been reworked in branch 0.9.5 to avoid display 
corruption seen on a Dell XPS 13 laptop.  I never got a kernel log in 
that case - it may be that your problem may be related.  If you can, 
please build from branch 0.9.5 and see if the problem still occurs.

Sanford

jrsa, 2019-02-18:
you are correct, I can successfully poke my external display by specifying i2c bus 11. I also switched to the 0.9.5 branch, but I suspect changing the bus is what made it work. I'm not sure if there is a way to handle the hang when scanning the bus, seems more like a kernel problem than a problem with this software. I would be curious, however, if there is a way to blacklist the problematic bus to prevent problems in the future.

rockowitz, 2019-02-18:
Let me see if I understand you correctly.  You're now using the 0.9.5 
branch.  Issuing a command such as "ddcutil getvcp 10 --bus 11" works.  
What happens if you issue the command "ddcutil detect --verbose" or 
"ddcutil environment --verbose"?   If either of these now work please 
send the output.  I ask because display detection in 0.9.5 goes to 
greater lengths to only probe "promising" /dev/i2c devices as reported 
by udev.  In particular, it avoids DDC/CI communication with Embedded 
DisplayPort (eDP) devices.  Laptop displays do not support DDC/CI, and 
recent laptop displays are eDP devices.

Assuming that that the environment command still hangs the system, 
instead please run the following commands:

sudo ls /sys/bus/i2c/devices
sudo cat /sys/bus/i2c/devices/*/name

Sanford

jrsa, 2019-02-20:
`ddcutil detect --verbose` and `ddcutil environment --verbose`, with ddcutil compiled from the 0.9.5 branch, still hang. interestingly, running with the `--bus 11` argument does *not* work on the PPA installed version (0.8.6 for me.) I have been using the 0.9.5 version with that argument and it works well.

devices:

i2c-0@
i2c-1@
i2c-10@
i2c-11@
i2c-12@
i2c-13@
i2c-14@
i2c-15@
i2c-16@
i2c-17@
i2c-18@
i2c-19@
i2c-2@
i2c-20@
i2c-3@
i2c-4@
i2c-5@
i2c-6@
i2c-7@
i2c-8@
i2c-9@

names:

nvkm-0000:01:00.0-bus-0000
nvkm-0000:01:00.0-bus-0001
nvkm-0000:01:00.0-bus-0002
nvkm-0000:01:00.0-bus-0005
nvkm-0000:01:00.0-bus-0006
nvkm-0000:01:00.0-bus-0007
nvkm-0000:01:00.0-bus-0008
nvkm-0000:01:00.0-bus-0009
nvkm-0000:01:00.0-aux-000a
nvkm-0000:01:00.0-aux-000b
nvkm-0000:01:00.0-aux-000c
nvkm-0000:01:00.0-aux-000d
0000:01:00.0
0000:01:00.0
0000:01:00.0
i915 gmbus ssc
i915 gmbus vga
i915 gmbus panel
i915 gmbus dpc
i915 gmbus dpb
i915 gmbus dpd

rockowitz, 2019-02-22:
Hi James,

As you note, the problem is in the nouveau driver.  The crash occurs in 
function g94_i2c_aux_transfer(), in kernel file 
drivers/gpu/drm/nouveau/nvkm/subdev/i2c/auxg94.c.  From the naming, it 
would seem that this code is intended to handle DisplayPort 
I2C-over-aux, and it's for the g94 chip.  The g94 is a relatively old 
chip, which is odd.  What model of Macbook Pro do you have?  In any 
event, this seems like a bug for the nouveau folks.

I was hoping to come up for a workaround to avoid probing the bus that 
causes the crash. Unfortunately, I don't see how to do that. ddcutil 
already ignores i2c buses belonging to the nouveau driver whose names do 
not start with "nvkm-", and all your remaining names look reasonable.

For now at least I'm afraid you're stuck with using the --bus option.  
If it's any consolation that's more efficient than having ddcutil 
examine all the I2C buses.

Regards,
Sanford

closed 2019-03-05 by rockowitz