rockowitz/ddcutil #81: Valgrind detects some memleaks

FedeDP, 2019-03-31:
Hi!
It seems like latest ddcutil (0.9.5) C api has some memleaks;
this is the output of a single run of my own program when built with ddcutil support:
[ddcutil_leak.txt](https://github.com/rockowitz/ddcutil/files/3026933/ddcutil_leak.txt)

Note that when building without ddcutil, no leak is shown.
Hopefully this will help you, thanks for this software!

FedeDP, 2019-03-31:
Moreover, this is just a plain "run program; wait; ctrl-c it (a signalfd is in place, so clean exit)".
It seems like there is some library init function marked with a "__attribute__((constructor))" that allocates some memory that is not freed by its destructor.

rockowitz, 2019-04-03:
The "leaks" that valgrind is reporting are for a persistent data 
structure constructed during library initialization that maps function 
addresses to names.  It is used by certain diagnostic routines.  You are 
correct that it is created by a library initialization function marked 
"attribute((constructor))", and that there is no corresponding 
destructor. There are many more dynamic data structures that are created 
during library execution.  All this memory will be released when the 
program using the library is terminated.  These allocations would 
represent memory leaks only if the library were to be dynamically 
unloaded by the application.   Is that a use case for your application?

Sanford

FedeDP, 2019-04-03:
I see; even if i know they are not proper "memleaks", i always find it a good design to actually free memory, even before program's exit. But that's me, and my fixation :D
Feel free to close this one if you feel!

closed 2025-02-09 by rockowitz