rockowitz/ddcutil #82: Change input with 2 cards and VGA passthrough

v44r, 2019-04-11:
I have a BenQ BL2400. The DP-1 input is connected to an Intel GPU, the DVI-1 input is connected to an AMD GPU that I passthrough to a Windows virtual machine. 

I'd like to switch inputs with a keyboard shortcut that I have set in the window manager in linux. When I'm in linux (Intel==>DP-1 input selected) I can switch to DVI-1 (AMD GPU==>Windows) without problems. This is the output of ddcutil detect while in linux before switching:
```
# sudo ddcutil detect
Display 1
   I2C bus:             /dev/i2c-6
   EDID synopsis:
      Mfg id:           BNQ
      Model:            BenQ BL2400
      Serial number:    49A02042SL0
      Manufacture year: 2010
      EDID version:     1.4
   VCP version:         2.1
```
However, while I'm showing Windows (DVI-1) if I run the same command from a linux terminal, I get:
```
# sudo ddcutil detect
Invalid display
   I2C bus:             /dev/i2c-6
   EDID synopsis:
      Mfg id:           BNQ
      Model:            BenQ BL2400
      Serial number:    49A02042SL0
      Manufacture year: 2010
      EDID version:     1.4
   DDC communication failed
```
and I cannot switch back with ddcutil. If I switch manually (with the button) to DP-1 and run ddcutil again the following executions are even worse:
```
# sudo ddcutil detect
No displays found
```
and I have to turn the monitor off and on to be able to control it again.

Maybe windows is messing with DDC? Is it a bad idea to try to do what I'm doing? Should I search for a windows utility to change back to linux in DP-1 or is it better to not even try to mess with DDC from two running OSs?

rockowitz, 2019-04-11:
What you're trying to do sometimes works, sometimes doesn't. It seems to 
depend on the monitor's DDC/CI implementation - whether DDC 
communication is active on the inactive video input.

Some things you might try:
- Avoid using DisplayPort. I2C on DVI and HDMI uses dedicated pins, 
whereas I2C on DIsplayPort is emulated over the Aux channel.
- Reference the display by its i2c bus number, e.g. "sudo ddcutil --bus 
6 setvcp 60 x0f"
- Use a Windows program within the VM to switch back

The latter almost always works.  I suggest using softMCCS 
<https://www.entechtaiwan.com/lib/softmccs.shtm> from EnTech, at least 
for testing.  Once you know that works you can look for a simpler 
Windows command line utility to perform the switching.

One thing that has me a bit perplexed is that there appears to be only 1 
monitor (the BL2400) connected to the system.  How are you issuing Linux 
commands when the monitor is switched to Windows?

If you run command "sudo ddcutil environment --verbose" and send the 
output as an attachment it will help me better understand your 
configuration.

Sanford

v44r, 2019-04-12:
Thanks for the tips. 

> One thing that has me a bit perplexed is that there appears to be only 1 monitor (the BL2400) connected to the system.  How are you issuing Linux commands when the monitor is switched to Windows?

I can change the keyboard focus from linux to windows and vice-versa pressing both Ctrl keys. So even when I'm showing windows I can (blindly) run linux commands. I run the command, nothing happens, so I switch inputs with the physical button and see the resulting error.

I tried switching DP and DVI (i.e. using DVI for linux and DP for windows) and using the bus number instead of the display number (/dev/i2c-4 right now), but I get the same result: when I change from linux (DVI) to windows (DP) no errors:

```
$ sudo ddcutil -b 4 setvcp 60 0x0f
$ 
```
and the monitor switches inputs accordingly. But when I try to switch back (I'm showing windows but linux has keyboard focus, so I run this blind):
```
$ sudo ddcutil -b 4 setvcp 60 0x03
DDC communication failed for monitor on I2C bus /dev/i2c-4
$
```
So, as you say, it seems that the monitor ignores DCC commands from non-selected inputs. With environment verbose, when the input connected to the linux graphics card is selected (currently DVI):
```
 Trying simple VCP read of feature 0x10...
      read() returned 6e 88 02 00 No protocol specified
10 00 00 64 00 1e de
      Attempt 1 to read feature succeeded.
      DDC communication succeeded
```
but when another input is selected:
```
Trying simple VCP read of feature 0x10...
      read() returned de de de de No protocol specified
de de de de de de de
      Attempt 1 to read feature failed. status = DDCRC_DDC_DATA(-3001): DDC data error.  Retrying...
      read() returned de de de de de de de de de de de
      Attempt 2 to read feature failed. status = DDCRC_DDC_DATA(-3001): DDC data error.  Retrying...
      read() returned de de de de de de de de de de de
      Attempt 3 to read feature failed. status = DDCRC_DDC_DATA(-3001): DDC data error.
      DDC communication failed.
```

I ended up installing ClickMonitorDDC in windows, which has a nice interface and support for hotkeys and assigned a combination to switch back to linux. Now I can switch back and forth without problems.

rockowitz, 2019-04-15:
Thanks for the update.  I'm glad you got it working.

Can I ask you to run "sudo ddcutil interrogate" and send me the output 
as a file?  I'd like to add a description of your monitor to the list at 
http://www.ddcutil.com/monitor_notes/

Sanford

kinghat, 2020-08-02:
im having this same issue:
host: manjaro
guest: win 10 with dedicated gtx 660 passed through

my issue is that i cant switch back from windows via ClickMonitorDDC or softMCCS which says it cant find DCC/CI: 

interrogate log: [ddcutil-interrogate.log](https://github.com/rockowitz/ddcutil/files/5013026/ddcutil-interrogate.log)