rockowitz/ddcutil #84: Trackpoint freezes after getting data from monitor.

lefthanded256, 2019-05-05:
When I execute command when ddcutil have to get data from monitor, like:
sudo ddcutil setvcp 10 - 5  --display 1 
sudo ddcutil getvcp 10 --display 1
It is sometimes freezes trackpoint in my Lenovo ThinkPad USB keyboard (SK-8855) and I have to unplug and plug usb cable. If you don't believe, i can record this ;).
I think that trackpoint also use I2C.

[interrogate.txt](https://github.com/rockowitz/ddcutil/files/3145410/interrogate.txt)
[env_verb.txt](https://github.com/rockowitz/ddcutil/files/3145411/env_verb.txt)

rockowitz, 2019-05-05:
Interesting problem.

That it's intermittent is particularly puzzling.  Does it have any 
correlation with the system being suspended?  Do you have an external 
mouse?  If so, does it hang when the trackpoint hangs?

Your display is at /dev/i2c-8.  Try using "--bus 8"  instead of 
"--display 1" to specify the display. This skips the display detection 
phase, during which ddcutil probes for USB devices communicating MCCS 
over USB.

Please run the command "sudo ddcutil usbenv --verbose" and send the 
output as a file (as you have done).  This used to be part of 
interrogate, but the output is voluminous and displays that communicate 
MCCS over USB are rare so it was just cluttering most reports.

I suggest you install the "i2cdetect" command.  It provides an 
independent way to scan i2c devices, and will be used by 
environment/interrogate if it is available.

What leads you think that the trackpoint is an I2C device?   The 
keyboard has only a USB connector, so I would expect the trackpoint to 
be communicating using USB "reports" as does a normal USB mouse.

Regards,
Sanford

lefthanded256, 2019-05-05:
I haven't any other mouse. System is not suspended it works normally, i can press keys. [Here is video from this situation](https://photos.app.goo.gl/NrfACMZqCmYJWC3e8)

I started using ddcutil with -bus option and for now it works ok. But with -device option it was not freezeing immediately after one execution. I will be testing it.

[usbenv.txt](https://github.com/rockowitz/ddcutil/files/3145613/usbenv.txt)

Meaby i should check "i2cdetect" later after we figure out what causing problem in this situation?

Ok, so i was wrong. I remember that i saw some logs about my machine and i saw that Thinkpad TrackPoin uses i2c interface, but it was probably on my laptop (Thinkpad T450). In laptop case keyboard with trackpoint is connected to motherboard not over usb but other unique connector which uses i2c.

This keyboard has marked combination of Fn+Home/End to control the brightness (but of course it was provided for Windows systems with Lenovo/Thinkpad devices). Can it be connected with this issue?

Thank you for help.

Mateusz

rockowitz, 2019-05-05:
>
> I haven't any other mouse. System is not suspended it works normally, 
> i can press keys. Here is video from this situation 
> <https://photos.app.goo.gl/NrfACMZqCmYJWC3e8>
So the pattern is the first ddcutil command works, the second causes the 
trackpoint to hang.  Physically disconnecting and reconnecting USB 
causes the trackpoint to be reinitialized, and things work again.

> I started using ddcutil with -bus option and for now it works ok. But 
> with -device option it was not freezeing immediately after one 
> execution. I will be testing it.

If the problem does not occur when using the --bus option then I think 
it's safe to say that something ddcutil is doing during monitor 
detection causes the hang.

Background:

Normally, Monitor Control Command Set commands are sent over the I2C bus 
using the DDC/CI protocol.  However, some high end monitors (e.g. Eizo, 
NEC) instead (or in addition) send the commands over the USB bus.  This 
is what ddcutil refers to as a USB connected monitor.  It does not mean 
that the video signal is sent over USB, just MCCS.

Now, there are USB standards for all this.   The "Device Class 
Definition for Human Interface Devices (HID)" specifies how 
communication works with devices like keyboards, mice, etc. There's also 
the "USB Monitor Control Class Specification", which treats MCCS 
communication as a special case of a USB HID device.  So ddcutil 
examines USB HID devices to see if any supports MCCS communication.  It 
tries to specially rule out the HID devices that are keyboards and mice, 
because there is all sorts of special handling for these devices.

What I suspect is that the trackpoint does not look like a normal mouse, 
so ddcutil sends commands to detect its capabilities that put it into an 
odd state.

> usbenv.txt <https://github.com/rockowitz/ddcutil/files/3145613/usbenv.txt>
It looks like the output is truncated. Did "ddcutil usbenv -v" crash, or 
was the output not fully captured for some other reason? If you haven't 
already, try redirecting stderr to stdout, e.g. "ddcutil usbenv -v > 
usbenv.txt  2>&1"
>
> Meaby i should check "i2cdetect" later after we figure out what 
> causing problem in this situation?

> Ok, so i was wrong. I remember that i saw some logs about my machine
> and i saw that Thinkpad TrackPoin uses i2c interface, but it was
> probably on my laptop (Thinkpad T450). In laptop case keyboard [...]

Doubtful.   The Fn+Home/End is just a special key sequence.  I imagine 
it is intended to be caught just like similar keys on a laptop, where 
there is hardware ACPI control of display brightness.
>
> Thank you for help.
>
> Mateusz

Thank you for the clarity of your reports.

lefthanded256, 2019-05-05:
So after executing "sudo ddcutil usbenv -v" few times trackpoint stop working again. Before i was spamming with command "sudo ddcutil setvcp 10 - 5 --bus 8" without any problem.
ddcutil usbenv -v > usbenv.txt  2>&1 - not crushes tracpoint, but same with sudo cause freeze problem (sudo ddcutil usbenv -v > sudo_usbenv.txt  2>&1).
[usbenv.txt](https://github.com/rockowitz/ddcutil/files/3145959/usbenv.txt)
[sudo_usbenv.txt](https://github.com/rockowitz/ddcutil/files/3145960/sudo_usbenv.txt)

rockowitz, 2019-05-05:
Please just use the --bus option for the next several days to confirm 
that avoids the problem.

On the assumption that it's the probing of the trackpoint that causes 
the problem, I'll review the USB related code to see how the problem 
might be better avoided.  As you might imagine, this is not a part of 
the code base that gets much attention.

Sanford

rockowitz, 2019-05-09:
Mateusz, 

Are you able to build from the git source?  I have pushed development branch 0.9.6 which contains special handling for your device, updated the usbenv diagnostics, and outputs a bunch of debugging code. 

Does using the --display option instead of --bus work now or does the trackpoint still hang? 

Please send output of "sudo ddcutil detect" and "sudo ddcutil usbenv -v".  If the hang still occurs, please send output of these commands before and after the hang.

Thanks, 
Sanford

lefthanded256, 2019-05-11:
So both "sudo ddcutil detect" and "sudo ddcutil usbenv -v" cause tracpoint freeze. 
[096.txt](https://github.com/rockowitz/ddcutil/files/3168863/096.txt)
I'm newb in building form sources so i executed --version command. I'm not shure if i buld it correctly.
Controlling relative brightens level with --bus cmd work without problem.

rockowitz, 2019-05-11:
Yes, you built successfully.  Version is 0.9.6 and I'm seeing the trace 
output.

There was already a function to always regard a USB device, based on 
it's vendor id/product id, as a display.  I added a corresponding 
function (deny_hid_monitor_by_vid_pid) to always ignore a device, and 
added the keyboard/trackpoint device to it. So ddcutil doesn't even 
query the device to determine whether it looks like a display.  And yet 
the trackpoint still locks up. Very puzzling.

Try building again with USB support totally disabled.  To do so, execute 
the following command before building:

   configure --disable-usb

The --version command should then indicate that ddcutil was built 
without support for USB displays.

Sanford

lefthanded256, 2019-05-11:
So now cmd like "sudo ddcutil setvcp 10 + 5 --display 1" works without freezing trackpoint.
Also i executed "sudo ddcutil detect" and "sudo ddcutil usbenv -v" without any problem.
[096_no_usb.txt](https://github.com/rockowitz/ddcutil/files/3168980/096_no_usb.txt)

rockowitz, 2019-05-11:
So something in the detection of USB connected displays is causing the 
failure.  At this point all ddcutil display detection does is open the 
usb hiddev device in read-only mode, get the vendor id/product id, and 
(for the keyboard/trackpoint) immediately decide to do nothing more with 
it.   Very strange.

rockowitz, 2019-05-12:
A couple hunches.

1) Does executing command "lsusb" freeze the trackpoint?

2) The latest 0.9.6 branch uses an alternative method of finding hiddev 
devices in the dev tree,
querying the file system instead of using the udev API.  Can you build 
it and see if anything changes?

Note that you'll need to execute configure again, either with parm 
"--enable-usb" or no parm.

Sanford

rockowitz, 2019-05-28:
Have you had a chance to test with the current 0.9.6 development branch?  If I don't hear back in the next few days I'm going to disable the diagnostic messages in the 0.9.6 branch and close the bug.

Sanford

lefthanded256, 2019-05-29:
So 0.9.6 without support for USB... seems to work fine with --display option. If it will make any problem in future i will mention it.

rockowitz, 2019-05-30:
The following message was sent via email but does not appear in the thread: 

Per your response, the debug trace messages, and associated code, relating to the SK-8855 problem have been disabled in the 0.9.6 branch.

closed 2019-05-30 by rockowitz

rockowitz, 2019-05-31:
Per your response, the debug trace messages, and associated code, 
relating to the SK-8855 problem have been disabled in the 0.9.6 branch.