rockowitz/ddcutil #86: ddcutil fails when called too quickly?

nvahalik, 2019-06-26:
I was using [Argos](https://github.com/p-e-w/argos) to write a script that sits in the Gnome Shell and would allow me to set some monitor configurations there.
After messing up on the script, I started noticing that the script stopped working and that subsequent calls to `ddcutil` just hung. Like... cannot `Ctrl+C` and cannot `kill -9 <pid-of-hung-ddcutil>`. The only thing that makes them disappear is a *hard* system reboot. The system will just hang if I try to reboot.

Note that everything else works perfectly and this issue only surfaced after scripting calls to ddcutil. I do have two monitors, if that matters.

Here's the output from `dmesg`, which appears to be relevant.

```[  364.474870] WARNING: CPU: 12 PID: 7810 at /build/linux-hwe-LWyLhJ/linux-hwe-4.18.0/drivers/gpu/drm/amd/amdgpu/../display/dc/gpio/gpio_base.c:64 dal_gpio_open_ex+0x10/0x40 [amdgpu]
[  364.474872] Modules linked in: nls_iso8859_1 amdkfd amd_iommu_v2 edac_mce_amd snd_hda_codec_realtek eeepc_wmi snd_hda_codec_generic kvm_amd asus_wmi sparse_keymap amdgpu snd_hda_codec_hdmi wmi_bmof video kvm mxm_wmi snd_hda_intel irqbypass snd_hda_codec crct10dif_pclmul snd_hda_core crc32_pclmul ghash_clmulni_intel snd_usb_audio snd_seq_midi pcbc snd_seq_midi_event snd_usbmidi_lib snd_hwdep snd_rawmidi chash joydev input_leds snd_pcm gpu_sched aesni_intel aes_x86_64 ttm snd_seq crypto_simd cryptd glue_helper drm_kms_helper snd_seq_device snd_timer drm snd fb_sys_fops syscopyarea sysfillrect sysimgblt soundcore ccp k10temp wmi mac_hid sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 hid_generic usbhid hid igb nvme i2c_piix4 i2c_algo_bit dca nvme_core ahci libahci gpio_amdpt gpio_generic
[  364.474919] CPU: 12 PID: 7810 Comm: ddcutil Not tainted 4.18.0-24-generic #25~18.04.1-Ubuntu
[  364.474920] Hardware name: System manufacturer System Product Name/PRIME X470-PRO, BIOS 4207 12/07/2018
[  364.474970] RIP: 0010:dal_gpio_open_ex+0x10/0x40 [amdgpu]
[  364.474971] Code: 48 89 e5 53 48 89 fb 48 8b 3f e8 2b 02 62 e4 48 c7 03 00 00 00 00 5b 5d c3 90 0f 1f 44 00 00 55 48 83 7f 08 00 48 89 e5 74 0e <0f> 0b e8 69 5d 51 e4 b8 05 00 00 00 5d c3 89 77 18 8b 57 14 4c 8d 
[  364.474998] RSP: 0018:ffffb8018d6d7b88 EFLAGS: 00010282
[  364.474999] RAX: 0000000000000000 RBX: ffff9ea7f8d98d00 RCX: 0000000000000037
[  364.475000] RDX: 0000000000000000 RSI: 0000000000000003 RDI: ffff9ea7f8d983c0
[  364.475001] RBP: ffffb8018d6d7b88 R08: 0000000000000000 R09: 0000000000000000
[  364.475002] R10: 0000000000000da0 R11: 0000000000000000 R12: 0000000000000003
[  364.475003] R13: 0000000000000000 R14: 0000000000000000 R15: ffffb8018d6d7d50
[  364.475005] FS:  00007fbdbd2c7b80(0000) GS:ffff9ea81ef00000(0000) knlGS:0000000000000000
[  364.475006] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  364.475007] CR2: 000055c07fcdde78 CR3: 000000069add2000 CR4: 00000000003406e0
[  364.475008] Call Trace:
[  364.475057]  dal_ddc_open+0x32/0xe0 [amdgpu]
[  364.475103]  dal_aux_engine_acquire+0x37/0x70 [amdgpu]
[  364.475148]  dal_i2caux_acquire_aux_engine+0x58/0x80 [amdgpu]
[  364.475189]  dal_i2caux_submit_aux_command+0x4b/0x1c0 [amdgpu]
[  364.475194]  ? load_balance+0x164/0x9a0
[  364.475239]  dal_ddc_service_read_dpcd_data+0x63/0xa0 [amdgpu]
[  364.475242]  ? syscall_return_via_sysret+0x13/0x83
[  364.475289]  dm_dp_aux_transfer+0xac/0xc0 [amdgpu]
[  364.475296]  drm_dp_i2c_do_msg+0x86/0x270 [drm_kms_helper]
[  364.475298]  ? __switch_to_asm+0x34/0x70
[  364.475303]  drm_dp_i2c_xfer+0xc0/0x250 [drm_kms_helper]
[  364.475305]  ? __switch_to_asm+0x40/0x70
[  364.475309]  __i2c_transfer+0x119/0x420
[  364.475311]  ? _cond_resched+0x19/0x40
[  364.475313]  i2c_transfer+0x5e/0xf0
[  364.475315]  i2c_transfer_buffer_flags+0x4d/0x70
[  364.475317]  i2cdev_read+0x77/0x120
[  364.475321]  __vfs_read+0x1b/0x40
[  364.475323]  vfs_read+0x8e/0x130
[  364.475325]  ksys_read+0x55/0xc0
[  364.475327]  __x64_sys_read+0x1a/0x20
[  364.475329]  do_syscall_64+0x5a/0x120
[  364.475332]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  364.475333] RIP: 0033:0x7fbdbbf46081
[  364.475334] Code: fe ff ff 48 8d 3d 67 9c 0a 00 48 83 ec 08 e8 a6 4c 02 00 66 0f 1f 44 00 00 48 8d 05 81 08 2e 00 8b 00 85 c0 75 13 31 c0 0f 05 <48> 3d 00 f0 ff ff 77 57 f3 c3 0f 1f 44 00 00 41 54 55 49 89 d4 53 
[  364.475360] RSP: 002b:00007ffe5c71a208 EFLAGS: 00000246 ORIG_RAX: 0000000000000000
[  364.475362] RAX: ffffffffffffffda RBX: 0000000000000014 RCX: 00007fbdbbf46081
[  364.475363] RDX: 0000000000000014 RSI: 000055c07fcec430 RDI: 0000000000000003
[  364.475364] RBP: 15abda1ca21f4c2f R08: 000055c07f85f4ee R09: 00007fbdbbfa4750
[  364.475364] R10: 00007ffe5c71a1d0 R11: 0000000000000246 R12: 0000000000000003
[  364.475365] R13: 000055c07fcec430 R14: 000055c07fcec740 R15: 0000000000000000
[  364.475367] ---[ end trace 8d9af36c08ebfc40 ]---
[  364.475374] int3: 0000 [#1] SMP NOPTI
[  364.475379] CPU: 12 PID: 7810 Comm: ddcutil Tainted: G        W         4.18.0-24-generic #25~18.04.1-Ubuntu
[  364.475381] Hardware name: System manufacturer System Product Name/PRIME X470-PRO, BIOS 4207 12/07/2018
[  364.475385] RIP: 0010:kgdb_breakpoint+0x14/0x20
[  364.475386] Code: ea ff ff ff c3 90 0f 1f 44 00 00 55 31 c0 48 89 e5 5d c3 0f 1f 00 0f 1f 44 00 00 55 48 89 e5 f0 ff 05 68 7f 88 01 0f ae f8 cc <0f> ae f8 f0 ff 0d 5a 7f 88 01 5d c3 0f 1f 44 00 00 55 48 89 e5 e8 
[  364.475414] RSP: 0018:ffffb8018d6d7b78 EFLAGS: 00000202
[  364.475416] RAX: 0000000000000000 RBX: ffff9ea7f8d98d00 RCX: 0000000000000037
[  364.475418] RDX: 0000000000000000 RSI: 0000000000000003 RDI: ffff9ea7f8d983c0
[  364.475419] RBP: ffffb8018d6d7b78 R08: 0000000000000000 R09: 0000000000000000
[  364.475421] R10: 0000000000000da0 R11: 0000000000000000 R12: 0000000000000003
[  364.475422] R13: 0000000000000000 R14: 0000000000000000 R15: ffffb8018d6d7d50
[  364.475425] FS:  00007fbdbd2c7b80(0000) GS:ffff9ea81ef00000(0000) knlGS:0000000000000000
[  364.475427] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  364.475428] CR2: 000055c07fcdde78 CR3: 000000069add2000 CR4: 00000000003406e0
[  364.475430] Call Trace:
[... 21 lines cut ...]
[  364.475753]  __x64_sys_read+0x1a/0x20
[  364.475755]  do_syscall_64+0x5a/0x120
[  364.475758]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  364.475759] RIP: 0033:0x7fbdbbf46081
[  364.475760] Code: fe ff ff 48 8d 3d 67 9c 0a 00 48 83 ec 08 e8 a6 4c 02 00 66 0f 1f 44 00 00 48 8d 05 81 08 2e 00 8b 00 85 c0 75 13 31 c0 0f 05 <48> 3d 00 f0 ff ff 77 57 f3 c3 0f 1f 44 00 00 41 54 55 49 89 d4 53 
[  364.475784] RSP: 002b:00007ffe5c71a208 EFLAGS: 00000246 ORIG_RAX: 0000000000000000
[  364.475786] RAX: ffffffffffffffda RBX: 0000000000000014 RCX: 00007fbdbbf46081
[  364.475788] RDX: 0000000000000014 RSI: 000055c07fcec430 RDI: 0000000000000003
[  364.475789] RBP: 15abda1ca21f4c2f R08: 000055c07f85f4ee R09: 00007fbdbbfa4750
[  364.475791] R10: 00007ffe5c71a1d0 R11: 0000000000000246 R12: 0000000000000003
[  364.475792] R13: 000055c07fcec430 R14: 000055c07fcec740 R15: 0000000000000000
[  364.475794] Modules linked in: nls_iso8859_1 amdkfd amd_iommu_v2 edac_mce_amd snd_hda_codec_realtek eeepc_wmi snd_hda_codec_generic kvm_amd asus_wmi sparse_keymap amdgpu snd_hda_codec_hdmi wmi_bmof video kvm mxm_wmi snd_hda_intel irqbypass snd_hda_codec crct10dif_pclmul snd_hda_core crc32_pclmul ghash_clmulni_intel snd_usb_audio snd_seq_midi pcbc snd_seq_midi_event snd_usbmidi_lib snd_hwdep snd_rawmidi chash joydev input_leds snd_pcm gpu_sched aesni_intel aes_x86_64 ttm snd_seq crypto_simd cryptd glue_helper drm_kms_helper snd_seq_device snd_timer drm snd fb_sys_fops syscopyarea sysfillrect sysimgblt soundcore ccp k10temp wmi mac_hid sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 hid_generic usbhid hid igb nvme i2c_piix4 i2c_algo_bit dca nvme_core ahci libahci gpio_amdpt gpio_generic
[  364.475824] ---[ end trace 8d9af36c08ebfc41 ]---
[  364.475827] RIP: 0010:kgdb_breakpoint+0x14/0x20
[  364.475828] Code: ea ff ff ff c3 90 0f 1f 44 00 00 55 31 c0 48 89 e5 5d c3 0f 1f 00 0f 1f 44 00 00 55 48 89 e5 f0 ff 05 68 7f 88 01 0f ae f8 cc <0f> ae f8 f0 ff 0d 5a 7f 88 01 5d c3 0f 1f 44 00 00 55 48 89 e5 e8 
[  364.475851] RSP: 0018:ffffb8018d6d7b78 EFLAGS: 00000202
[  364.475853] RAX: 0000000000000000 RBX: ffff9ea7f8d98d00 RCX: 0000000000000037
[  364.475854] RDX: 0000000000000000 RSI: 0000000000000003 RDI: ffff9ea7f8d983c0
[  364.475856] RBP: ffffb8018d6d7b78 R08: 0000000000000000 R09: 0000000000000000
[  364.475857] R10: 0000000000000da0 R11: 0000000000000000 R12: 0000000000000003
[  364.475859] R13: 0000000000000000 R14: 0000000000000000 R15: ffffb8018d6d7d50
[  364.475861] FS:  00007fbdbd2c7b80(0000) GS:ffff9ea81ef00000(0000) knlGS:0000000000000000
[  364.475862] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  364.475864] CR2: 000055c07fcdde78 CR3: 000000069add2000 CR4: 00000000003406e0
```

rockowitz, 2019-07-01:
Looking at the trace, the hang is deep in the amdgpu driver, more 
specifically in the DisplayPort I2C-over-aux handling.  As a result, a 
ddcutil  call to i2c is hanging.

ddcutil could be rewritten to run i2c system calls in a separate thread, 
catch the SIGTERM and SIGKILL signals, and terminate execution, but that 
would add a lot of complexity for a problem that has never been reported 
before and appears to be caused by a driver bug.

Taking a cursory look at the amdgpu driver forums, it appears that there 
have been significant changes in this area in kernel 4.19. (You're on 
4.18.) See, for example, bug 105724 at bugs.feedesktop.org.

If the problem does not resolve with 4.19, I suggest you post a bug to 
bugs.freedesktop.org. The initial response will likely ask if you've 
installed the latest development driver.

Are your monitors daisy-chained DisplayPort devices?  That's a 
particularly problematic area.

Please run command "ddcutil interrogate" and send the output as an 
attachment.  That may give me some further insight into what is going on.

Sanford

nvahalik, 2019-07-02:
Here you go!
[interrogate-output.txt](https://github.com/rockowitz/ddcutil/files/3350993/interrogate-output.txt)

nvahalik, 2019-07-02:
No, my displays both have their own DP connection. I'm using a Radeon RX 580 series card. It's got 2x DP and 2x HDMI outputs.

nvahalik, 2019-07-03:
Kernel updated to 4.20.17. Still an issue. This time I tried to set the brightness of both monitors at the same time. The first time, the first one ran and the second one hung. Then both hung.

Looks like it failed at the exact same spot.

```
[ 1290.270967] WARNING: CPU: 11 PID: 13323 at drivers/gpu/drm/amd/amdgpu/../display/dc/gpio/gpio_base.c:64 dal_gpio_open_ex+0x29/0x40 [amdgpu]
[ 1290.270968] Modules linked in: i2c_dev uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media edac_mce_amd kvm_amd kvm irqbypass snd_hda_codec_realtek nls_iso8859_1 amdgpu snd_hda_codec_generic snd_hda_codec_hdmi snd_hda_intel crct10dif_pclmul snd_hda_codec snd_usb_audio snd_hda_core snd_usbmidi_lib crc32_pclmul snd_hwdep chash ghash_clmulni_intel amd_iommu_v2 snd_pcm gpu_sched snd_seq_dummy ttm snd_seq_oss snd_seq_midi drm_kms_helper snd_seq_midi_event snd_rawmidi drm snd_seq drm_panel_orientation_quirks cfbfillrect cfbimgblt snd_seq_device aesni_intel snd_timer cfbcopyarea eeepc_wmi fb_sys_fops syscopyarea asus_wmi aes_x86_64 snd sysfillrect crypto_simd sparse_keymap input_leds joydev sysimgblt cryptd glue_helper mxm_wmi video wmi_bmof soundcore ccp k10temp fb fbdev mac_hid sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 hid_generic usbhid hid igb i2c_algo_bit i2c_piix4 dca nvme i2c_core ahci nvme_core libahci gpio_amdpt wmi
[ 1290.271002]  gpio_generic
[ 1290.271005] CPU: 11 PID: 13323 Comm: ddcutil Not tainted 4.20.17-042017-generic #201903190933
[ 1290.271006] Hardware name: System manufacturer System Product Name/PRIME X470-PRO, BIOS 4207 12/07/2018
[ 1290.271075] RIP: 0010:dal_gpio_open_ex+0x29/0x40 [amdgpu]
[ 1290.271076] Code: 00 0f 1f 44 00 00 55 48 83 7f 08 00 48 89 e5 75 19 89 77 18 8b 57 14 4c 8d 47 08 89 f1 8b 77 10 48 8b 3f e8 f9 04 00 00 5d c3 <0f> 0b e8 e0 1a 87 c3 b8 05 00 00 00 5d c3 66 0f 1f 84 00 00 00 00
[ 1290.271077] RSP: 0018:ffffb2af4fe83b70 EFLAGS: 00010286
[ 1290.271079] RAX: 0000000000000000 RBX: ffff920037f87380 RCX: 0000000000000000
[ 1290.271080] RDX: 0000000000000000 RSI: 0000000000000003 RDI: ffff920037f87ca0
[ 1290.271080] RBP: ffffb2af4fe83b70 R08: 0000000000000000 R09: 0000000000000001
[ 1290.271081] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000003
[ 1290.271082] R13: 0000000000000000 R14: 0000000000000001 R15: 0000000000000030
[ 1290.271083] FS:  00007fd33ac28b80(0000) GS:ffff92003ecc0000(0000) knlGS:0000000000000000
[ 1290.271084] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1290.271085] CR2: 00007f762abaa040 CR3: 000000063cfe0000 CR4: 00000000003406e0
[ 1290.271085] Call Trace:
[ 1290.271155]  dal_ddc_open+0x32/0xe0 [amdgpu]
[ 1290.271224]  acquire+0x3b/0x80 [amdgpu]
[ 1290.271286]  dc_link_aux_transfer+0x98/0x150 [amdgpu]
[ 1290.271355]  dm_dp_aux_transfer+0x9f/0x130 [amdgpu]
[ 1290.271361]  drm_dp_i2c_do_msg+0x8b/0x290 [drm_kms_helper]
[ 1290.271367]  drm_dp_i2c_xfer+0xc4/0x250 [drm_kms_helper]
[ 1290.271372]  __i2c_transfer+0x14a/0x450 [i2c_core]
[ 1290.271376]  ? _cond_resched+0x19/0x30
[ 1290.271379]  i2c_transfer+0x5a/0xd0 [i2c_core]
[ 1290.271383]  i2c_transfer_buffer_flags+0x4d/0x70 [i2c_core]
[ 1290.271386]  i2cdev_write+0x56/0xa0 [i2c_dev]
[ 1290.271389]  __vfs_write+0x3a/0x190
[ 1290.271391]  ? apparmor_file_permission+0x1a/0x20
[ 1290.271394]  ? security_file_permission+0x31/0xc0
[ 1290.271396]  vfs_write+0xab/0x1b0
[ 1290.271397]  ksys_write+0x55/0xc0
[ 1290.271399]  __x64_sys_write+0x1a/0x20
[ 1290.271402]  do_syscall_64+0x5a/0x110
[ 1290.271404]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[ 1290.271405] RIP: 0033:0x7fd3398a7154
[ 1290.271407] Code: 89 02 48 c7 c0 ff ff ff ff c3 66 2e 0f 1f 84 00 00 00 00 00 66 90 48 8d 05 b1 07 2e 00 8b 00 85 c0 75 13 b8 01 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 54 f3 c3 66 90 41 54 55 49 89 d4 53 48 89 f5
[ 1290.271407] RSP: 002b:00007ffee3079758 EFLAGS: 00000246 ORIG_RAX: 0000000000000001
[ 1290.271409] RAX: ffffffffffffffda RBX: 0000000000000001 RCX: 00007fd3398a7154
[ 1290.271409] RDX: 0000000000000001 RSI: 00007ffee30797b7 RDI: 0000000000000003
[ 1290.271410] RBP: 00007ffee30797b7 R08: 00007ffee31480a0 R09: 00007ffee3148080
[ 1290.271411] R10: 000000000008bdb0 R11: 0000000000000246 R12: 15adf9c617cac805
[ 1290.271411] R13: 0000000000000003 R14: 0000000000000000 R15: 000055d724af4b88
[ 1290.271413] ---[ end trace b558afb76afe3f3f ]---
[ 1290.271420] int3: 0000 [#1] SMP NOPTI
[ 1290.271424] CPU: 11 PID: 13323 Comm: ddcutil Tainted: G        W         4.20.17-042017-generic #201903190933
[ 1290.271425] Hardware name: System manufacturer System Product Name/PRIME X470-PRO, BIOS 4207 12/07/2018
[ 1290.271429] RIP: 0010:kgdb_breakpoint+0x14/0x20
[ 1290.271431] Code: 00 5d c3 0f 1f 00 0f 1f 44 00 00 55 31 c0 48 89 e5 5d c3 0f 1f 00 0f 1f 44 00 00 55 48 89 e5 f0 ff 05 18 fb 61 01 0f ae f8 cc <0f> ae f8 f0 ff 0d 0a fb 61 01 5d c3 0f 1f 44 00 00 55 48 89 e5 e8
[ 1290.271432] RSP: 0018:ffffb2af4fe83b60 EFLAGS: 00000202
[ 1290.271434] RAX: 0000000000000000 RBX: ffff920037f87380 RCX: 0000000000000000
[ 1290.271435] RDX: 0000000000000000 RSI: 0000000000000003 RDI: ffff920037f87ca0
[ 1290.271436] RBP: ffffb2af4fe83b60 R08: 0000000000000000 R09: 0000000000000001
[ 1290.271437] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000003
[ 1290.271438] R13: 0000000000000000 R14: 0000000000000001 R15: 0000000000000030
[ 1290.271440] FS:  00007fd33ac28b80(0000) GS:ffff92003ecc0000(0000) knlGS:0000000000000000
[... cut ...]
[ 1290.272236] R13: ffff920031a62c00 R14: ffffffff8555dc40 R15: 0000000000000002
[ 1290.272238] FS:  0000000000000000(0000) GS:ffff92003ecc0000(0000) knlGS:0000000000000000
[ 1290.272239] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1290.272240] CR2: 00007f762abaa040 CR3: 00000003fca0a000 CR4: 00000000003406e0
[ 1290.272242] Call Trace:
[ 1290.272246]  do_idle+0x22c/0x280
[ 1290.272249]  cpu_startup_entry+0x1d/0x20
[ 1290.272252]  start_secondary+0x1ab/0x200
[ 1290.272255]  secondary_startup_64+0xa4/0xb0
[ 1290.272257] ---[ end trace b558afb76afe3f41 ]---
```

rockowitz, 2019-07-03:
Nick,

I'm afraid everything looks quite in order.   DisplayPort Multi-Stream 
Transport and docking stations are common problem areas for drivers, but 
neither of those apply to you.  So my suggestion remains to report the 
problem to the amdgpu folks.

Regards,
Sanford

rockowitz, 2019-07-03:
When you say "at the same time", any chance you were executing 2 
instances of ddcutil in parallel?

If you are executing ddcutil sequentially, what happens if you insert a 
sleep between the 2 calls?

Sanford

nvahalik, 2019-07-05:
Correct. If I call this:

```
/usr/bin/ddcutil -d 1 setvcp 10 50 &
/usr/bin/ddcutil -d 2 setvcp 10 50 &
```
The second command will hang.

However, changing it to this:

```
/usr/bin/ddcutil -d 1 setvcp 10 50
sleep .1
/usr/bin/ddcutil -d 2 setvcp 10 50
```
Has resulted in no hangs when calling the batch command.

Lomomn, 2020-04-13:
I would also like to report issue when calling ddcutil too quickly. I was wondering if there is a built-in command which prevents ddcutil from running if another instance is already running.

rockowitz, 2020-04-13:
When you say "issue", what precisely is the behavior you observe?  What 
commands are you executing?  Can you send a log of the output.  Also, 
please run "ddcutil interrogate" and send the output as a file.  Thank you.

closed 2025-02-09 by rockowitz