rockowitz/ddcutil #88: Unable to detect second monitor as I2C device in MST chain

alfietham, 2019-08-17:
First of all, thanks for this wonderful util!!

**Use case & Setup:** My goal is to be able to control the brightness of 2 Dell Ultrasharp monitors at the same time such that the brightness are always in sync.

I've been able to successfully achieve that with one of the monitors plugged via DP and another via HDMI, and running `ddcutil --bus X setvcp 10 xx` in sequence for each monitor. Thus, monitor settings should not be the issue. 

However, I'd like to get rid of the HDMI and chain the second monitor via DP MST. Hoping that both monitors can be targeted in a similar manner by setvcp on each monitor to control it's brightness. (unless there's another better approach)

**Issue / Symptoms:** Unable to detect second monitor as an I2C device. 
- 2nd monitor shows up as DP-6, but not as i2c device
```   
EDID: ...20200139  Mfg: DEL  Model: DELL U2417H    SN:  
      I2C device:    Not found
      XrandR output:  DP-2-1-8
      DRM connector:  DP-6
      UDEV name:      (null)
      UDEV syspath:   (null)
      sysfs drm path: /sys/class/drm/card0/card0-DP-6
      sysfs drm I2C:  (null)
```
- Module i2c-dev is built into kernel
- Video driver module: i915
- Running `ddcutil detect` yields only the first display. With:
```
Display 1
   I2C bus:             /dev/i2c-5
      I2C address 0x50 (EDID) present: true 
      Is eDP device:                   false
      /sys/bus/i2c/devices/i2c-5/name: DPDDC-C
   EDID synopsis:
      Mfg id:           DEL
      Model:            DELL U2419HC
      Serial number:    
      Manufacture year: 2019
      EDID version:     1.4
      ...
   VCP version:         2.1
   ...
   Controller mfg:      RealTek
   Firmware version:    1.3
   Monitor returns DDC Null Response for unsupported features: false
```

**Other notes:**
- Seems similar to #61, except that my video driver is i915
- From reading the FAQ and other issues, it seems it might have something to do with how drivers handle MST? But I wasn't sure about that, or if there are any better ways around this

**Debug logs**
[ddcutil environment --verbose](https://gist.github.com/alfietham/46d2d003e53adbcc77497ce4099eaaf5)

rockowitz, 2019-08-17:
The i915 driver doesn't implement I2C over MST.  It's a long standing 
problem. See https://bugs.freedesktop.org/show_bug.cgi?id=100954

alfietham, 2019-08-19:
Ah, got it. Thanks for the response, and the bug link!

ptsneves, 2022-02-27:
It is now working since kernel >=5.10. Confirmed it works.

closed 2022-02-27 by rockowitz