rockowitz/ddcutil #89: Acer G235H DDC not detected

parkerlreed, 2019-08-17:
Using an RX 480 on Arch Linux. i2c-dev is probed. ddcutil finds the monitor but doesn't see the DDC. Monitor is connected via DVI

According to page 9 of the service manual, DDC is supported over DVI http://ekb.acer.com.cn/KB/%E6%8A%80%E6%9C%AF%E6%96%87%E4%BB%B6/MN/G235H/SG_G235H_BOOK.pdf

[image]

It seems to be DDC2B

```
[parker@e55 ~]$ sudo ddcutil detect --verbose
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Invalid display
   I2C bus:             /dev/i2c-8
      I2C address 0x50 (EDID) present: true 
      Is eDP device:                   false
      /sys/bus/i2c/devices/i2c-8/name: AMDGPU DM i2c hw bus 4
   EDID synopsis:
      Mfg id:           ACR
      Model:            G235H
      Serial number:    LK60C0134032
      Manufacture year: 2010
      EDID version:     1.3
      Product code:     288
      Extra descriptor: Unspecified
      Video input definition: 0x80 - Digital Input
      Supported features:
         DPMS standby
         DPMS suspend
         DPMS active-off
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.640, 0.329
      Green x,y:        0.300, 0.600
      Blue  x,y:        0.150, 0.060
      Extension blocks: 0
   EDID source: 
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 04 72 20 01 9a 39 91 02   .........r ..9..
      +0010   1d 14 01 03 80 34 1d 78 ea de 95 a3 54 4c 99 26   .....4.x....TL.&
      +0020   0f 50 54 bb 8c 10 b3 00 71 4f 81 00 95 00 81 80   .PT.....qO......
      +0030   81 c0 d1 c0 01 01 1a 36 80 a0 70 38 1f 40 30 20   .......6..p8.@0 
      +0040   25 00 09 25 21 00 00 1f 00 00 00 fd 00 38 4b 1f   %..%!........8K.
      +0050   54 12 00 0a 20 20 20 20 20 20 00 00 00 ff 00 4c   T...      .....L
      +0060   4b 36 30 43 30 31 33 34 30 33 32 0a 00 00 00 fc   K60C0134032.....
      +0070   00 47 32 33 35 48 0a 20 20 20 20 20 20 20 00 e6   .G235H.       ..
   DDC communication failed
   Is DDC/CI enabled in the monitor's on-screen display?
```

rockowitz, 2019-08-18:
> image

At the I2C level, ddcutil is at least partly able to communicate with 
the monitor.  It successfully reads the EDID at slave address x50 of the 
I2C bus.  DDC is at slave address x37, and that communication is failing 
for some reason.  (I assume you've double checked that DDC is enabled in 
the on screen display. )

The "ddcutil environment" command analyzes the system environment and 
makes suggestions as to what the problem may be.  If you still have a 
problem, run the following command as root and send the output as an 
attached file:

# ddcutil environment --verbose 2>&1

Sanford

parkerlreed, 2019-08-18:
Yeah the only environment suggestion was just group/ownership, but I was testing with sudo anyways.

Changed all the i2c to 777 for testing and same result.

Here's the output. 
[out.txt](https://github.com/rockowitz/ddcutil/files/3513289/out.txt)

parkerlreed, 2019-08-18:
And here's the OSD for confirmation 
[image]

parkerlreed, 2019-08-18:
One thing I notice is that the service manual DVI EDID closesly relates to the reported EDID but differs in a few spots. I wonder if editing the current EDID to match and loading it at boot might work.

parkerlreed, 2019-08-18:
Ahh yeah that method doesn't work as it reads the EDID directly from i2c.

rockowitz, 2019-08-19:
The one thing that catches my eye is that I2C-DDC communication fails on 
/dev/i2c-8 with system error EIO: Input/output error. This is the same 
error that all the other buses return when trying to read a non-existent 
EDID. It's not that there's bad DDC communication, i.e. with errors, it 
appears that the monitor is simply not responding on slave address x37.

Sanford

rockowitz, 2019-08-19:
The data returned for the EDID has no impact on DDC communication.  It 
is normal for monitors to have different EDIDs on different inputs.  The 
EDID on a VGA connection will contain timing information.  The EDID for 
a digital connection will not.

parkerlreed, 2019-08-19:
I was pointing out that the DVI EDID differed. Ignoring the others. 

So there's probably nothing that could be done?

rockowitz, 2019-08-19:
Any chance you have another monitor that you can plug in?  That would 
indicate whether the problem is monitor or driver related.

parkerlreed, 2019-08-19:
I do not. This is the first decent monitor I've had.

liangyongxiang, 2019-08-24:
I have a similar problem:
 [ddcutil detect --verbose](https://gist.github.com/liangyongxiang/8884291bc762bb7ddc6c221a530d1592)
[ddcutil environment --verbose](https://gist.github.com/liangyongxiang/1ee1cf66fc1d31bb33d43f4b1ebea246)

rockowitz, 2019-08-25:
It appears that the monitor, at least on its DisplayPort connection, is 
not responding to DDC/CI communication, i.e. I2C slave address x37.  I 
assume you've checked that DDC/CI is enabled in the on screen display.

Both the monitor and laptop appear to have VGA connections. The first 
step would be to test using VGA instead of DisplayPort. Particularly, 
the i2cdetect command, which "ddcutil environment" uses, does not 
support DisplayPort i2c buses.

A second step, would be to test with a different monitor.

Does either of these steps fix the problem?

If not, please send the output of "ddcutil env --verbose" for each of 
these configuration.

Sanford

liangyongxiang, 2019-08-30:
> Particularly

I finally have time to test it today!
I connect VGA and DisplayPort at the same time, so I can set the brightness. And I tried to set the brightness with DisplayPort only under Windows 10 and it works.

In addition, i2cdetect does not support DisplayPort i2c buses. Is there a software solution?

rockowitz, 2019-09-01:
>
>     Particularly
>
> I finally have time to test it today!
> I connect VGA and DisplayPort at the same time, so I can set the 
> brightness.
What I think you're saying is that if you select the VGA connection, 
e.g. by using the "--display" option, then "ddcutil setvcp 10" works.  
Please send the output of "ddcutil environment --verbose" for this case.

I also gather that you do not have another monitor to test with.

> And I tried to set the brightness with DisplayPort only under Windows 
> 10 and it works.
Windows drivers are generally more robust than Linux drivers for I2C 
communication.  In particular, they drop the I2C bus speed if necessary.

Regarding drivers, I have 2 suggestions:

1) You do not appear to have set the special Nvidia settings.  See 
http://www.ddcutil.com/nvidia.  I doubt this will make a difference in 
your case, though.
2) What happens if you use the nouveau driver instead of the Nvidia driver.

> In addition, i2cdetect does not support DisplayPort i2c buses. Is 
> there a software solution?
I know of no significant alternative to i2cdetect for probing I2C buses 
independent of ddcutil.   There's i2cio on github, but it just reports 
every slave address as being active for a DisplayPort connection.  You 
can try ddccontrol, occasionally it detects a monitor that ddcutil does 
not, but usually it's the other way around.

Sanford

liangyongxiang, 2019-09-01:
> Please send the output of "ddcutil environment --verbose" for this case.

After I configured NVIDIA, I tried the previous command again. This is my log: https://gist.github.com/liangyongxiang/6346d7e9121593df9165b727f6d14cf6

> I also gather that you do not have another monitor to test with.

Yes, I only have one monitor

rockowitz, 2019-09-03:
Since DDC communication works on the VGA connection, and also on 
DisplayPort in Windows, my best guess is that there's something a bit 
off in the monitor's DDC implementation on the DisplayPort connection, 
which the Windows driver can handle, but the Nvidia driver cannot.   
About the only thing I have to suggest is trying the nouveau driver.

Sanford

liangyongxiang, 2019-09-04:
I tried nouveau before but failed. I will continue to try the nouveau drive on weekends.

rockowitz, 2019-09-04:
Do you mean you can't get the nouveau driver to work, or that it works 
but doesn't solve the DisplayPort i2c problem?

liangyongxiang, 2019-09-04:
In the first case, I can't get the nouveau driver to work.

parkerlreed, 2019-09-04:
Ran across another monitor at work. Also on an AMD GPU (RX 560)

```
Invalid display

   I2C bus:             /dev/i2c-11
      I2C address 0x50 (EDID) present: true 
      Is eDP device:                   false
      /sys/bus/i2c/devices/i2c-11/name: AMDGPU DM i2c hw bus 1
   EDID synopsis:
      Mfg id:           AOC
      Model:            2279WH
      Serial number:    AHXK39A004047
      Manufacture year: 2019
      EDID version:     1.3
      Product code:     8825
      Extra descriptor: Unspecified
      Video input definition: 0x80 - Digital Input
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.653, 0.337
      Green x,y:        0.322, 0.611
      Blue  x,y:        0.153, 0.061
      Extension blocks: 1
   EDID source: 
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 05 e3 79 22 cf 0f 00 00   ..........y"....
      +0010   0b 1d 01 03 80 30 1b 78 2a 5a 65 a7 56 52 9c 27   .....0.x*Ze.VR.'
      +0020   0f 50 54 bf ef 00 d1 c0 b3 00 95 00 81 80 81 40   .PT............@
      +0030   81 c0 01 01 01 01 02 3a 80 18 71 38 2d 40 58 2c   .......:..q8-@X,
      +0040   45 00 dd 0c 11 00 00 1e 00 00 00 fd 00 32 4c 1e   E............2L.
      +0050   53 11 00 0a 20 20 20 20 20 20 00 00 00 fc 00 32   S...      .....2
      +0060   32 37 39 57 48 0a 20 20 20 20 20 20 00 00 00 ff   279WH.      ....
      +0070   00 41 48 58 4b 33 39 41 30 30 34 30 34 37 01 27   .AHXK39A004047.'
   DDC communication failed
   Is DDC/CI enabled in the monitor's on-screen display?
```

OSD reports DDC enabled. 

Page 62 here claims it should be on HDMI https://aocmonitorap.com/upload_files/manual/I2279VWHE/English.pdf

Page 63 notes that it is also DDC2B

[ddc-det.txt](https://github.com/rockowitz/ddcutil/files/3576113/ddc-det.txt)
[ddc-env.txt](https://github.com/rockowitz/ddcutil/files/3576114/ddc-env.txt)

rockowitz, 2019-09-05:
Looks like a DDC data error.   That is, I2C slave address x37 is active, 
but reading from it returns invalid data.

To confirm:

Install package i2c-tools, which contains command i2cdetect.  Use it to 
see what slave addresses are responsive on bus /dev/i2c-11: "i2cdetect 
-y 11"

Try running "ddcutil detect --verbose --ddc".  This will report DDC data 
errors.  (The -ddc option is not particularly useful in the environment 
command because the reads of feature x10 there use simple, hard coded, 
I2C communication,  as opposed to the normal, layered code that checks 
the --ddc option.

I doubt it is the driver.  I'm running on an AMD 460 right now, which 
has the same architecture as the 560.

Sanford

parkerlreed, 2019-09-05:
2279WH

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- 37 -- -- 3a -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- 49 -- -- -- -- -- -- 
50: 50 51 -- -- -- -- -- -- -- 59 -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --     

Output level:               Verbose
Reporting DDC data errors:  true
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 88 02 01 00 01 ff ff ff 00 00 b4 00 00 00 00 00 00 00
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 88 02 01 00 01 ff ff 00 00 b4 00 00 00 00 00 00 00 00
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 88 02 01 00 01 ff ff 00 00 00 b4 00 00 00 00 00 00 00
DDC: All zero response detected in ddc_i2c_write_read_raw
```

And here's the previous. It doesn't report any errors in the detect output

G235H

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- 37 -- -- 3a -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- 49 -- -- -- -- -- -- 
50: 50 -- -- -- -- -- -- -- -- 59 -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --           

[parker@e55 ~]$ sudo ddcutil detect --verbose --ddc
Output level:               Verbose
Reporting DDC data errors:  true
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

```

rockowitz, 2019-09-06:
Re the 2279WH, this appears to be a monitor problem. The video card is 
reading invalid data from the I2C bus.  The giveaway is the double bytes 
in response error.  The monitor is just not keeping up.  You can try 
undocumented option "--sleep-stats 2", which will double the time that 
ddcutil sleeps between I2C writes and reads from the amount required by 
the DDC protocol.  But I doubt this will solve the problem.  The real 
solution would be to slow the I2C bus clock.  This was proposed for the 
i915 driver but never implemented.  Nor is it implemented in any of the 
other Linux drivers. I suspect that the Windows drivers do this when 
communicating on slave address x37 (DDC), which would be at least one 
reason they work when Linux drivers don't.

Re the G235H, I'm not sure what to day.  It seems that all bytes come 
back 0 for any read request to the monitor.  I'll continue to ponder 
this, but it's most likely just a problem in how the monitor implements 
DDC.

Note that the "--stats" option will dump out lots of statistics. 
Included will be a count of how many packets were read that had all 
bytes 0.

Sanford

parkerlreed, 2019-09-06:
I see what sleep stats are mentioned in the code but the argument doesn't seem to be exposed

```
[parker@stealth ~]$ sudo ddcutil detect --verbose --ddc --sleep-stats 2
Option parsing failed: Unknown option --sleep-stats
No command specified
```

rockowitz, 2019-09-06:
Sorry.  Brain infarct.  The options to use are:
   --sleep-strategy 2
   --stats

parkerlreed, 2019-09-06:
```
Output level:               Verbose
Reporting DDC data errors:  true
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: Unexpected source address 0x51, should be 0x6e
DDC: i2c_response_bytes: 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51 51
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: Double byte in packet.
DDC: i2c_response_bytes: 6e 6e 88 02 01 00 01 ff ff 00 00 b4 00 00 00 00 00 00 00 00
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
DDC: All zero response detected in ddc_i2c_write_read_raw
Invalid display
   I2C bus:             /dev/i2c-6
      I2C address 0x50 (EDID) present: true 
      Is eDP device:                   true 
      /sys/bus/i2c/devices/i2c-6/name: DPDDC-A
   EDID synopsis:
      Mfg id:           SHP
      Model:            LQ125T1JW02
      Serial number:    Unspecified
      Manufacture year: 2014
      EDID version:     1.4
      Product code:     5167
      Extra descriptor: Unspecified
      Video input definition: 0xa5 - Digital Input (DisplayPort)
      Supported features:
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.312, 0.328
      Red   x,y:        0.640, 0.329
      Green x,y:        0.300, 0.600
      Blue  x,y:        0.149, 0.060
      Extension blocks: 0
   EDID source: 
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 4d 10 2f 14 00 00 00 00   ........M./.....
      +0010   2d 18 01 04 a5 1c 10 78 06 de 50 a3 54 4c 99 26   -......x..P.TL.&
      +0020   0f 50 54 00 00 00 01 01 01 01 01 01 01 01 01 01   .PT.............
      +0030   01 01 01 01 01 01 56 5e 00 a0 a0 a0 29 50 30 20   ......V^....)P0 
      +0040   35 00 15 9b 10 00 00 18 e4 3e 00 a0 a0 a0 29 50   5........>....)P
[... 94 lines cut ...]

   Sleep Event type      Count
   SE_WRITE_TO_READ         20
   SE_POST_OPEN              2
   SE_POST_WRITE             0
   SE_POST_READ             20
   SE_DDC_NULL               0
   SE_POST_SAVE_SETTINGS     0

Call Stats:
   Type                                     Count    Millisec  (      Nanosec)
   write calls       (IE_WRITE)                46         547  (    547203224)
   read calls        (IE_READ)                 25         146  (    146780992)
   open file calls   (IE_OPEN)                 15           0  (       345429)
   close file calls  (IE_CLOSE)                15           0  (       170267)
   other I/O calls   (IE_OTHER)                22           0  (        98579)
   Totals:                                    123         694  (    694598491)

Sleep Call Stats:
   Total sleep calls:                                      52
   Requested sleep time milliseconds :                   3600
   Actual sleep milliseconds (nanosec):                  3603  (   3603890858)

Total elapsed milliseconds (nanoseconds):                4472  (   4472564741)
```

liangyongxiang, 2019-09-10:
> Sorry. In the first case, I can't get the nouveau driver to work.
I tried to connect to the monitor using iGPU's miniDP interface to HDMI. It can work.
https://gist.github.com/liangyongxiang/d4fa97c5588c5d51e22305c286a5eb58

rockowitz, 2019-09-11:
What happens if you connect the laptop's Intel graphics mini-DP to the 
DP connector on the monitor?

liangyongxiang, 2019-09-11:
> 
> What happens if you connect the laptop's Intel graphics mini-DP to the DP connector on the monitor?
> […](#)

I bought a line of mini-dp to dp yesterday, I will test this way later.

liangyongxiang, 2019-09-14:
Today I tried to use minidp to dp on iGPU, the work is normal:

https://gist.github.com/liangyongxiang/8b42352a83803df4d57a87aafa6111df

rockowitz, 2019-09-15:
To summarize:
- Using the on-chip Intel GPU
   - DDC communication works in all combinations, including DisplayPort
- Using Nvidia gpu:
   - you have only tested with the proprietary Nvidia driver, not with 
nouveau
   - DDC communication fails on DisplayPort
   - DDC communication succeeds on other ports

So is it a hardware issue or a problem with the Nvidia driver? To 
distinguish this you need to test with nouveau.

If you conclude the problem is in the Nvidia driver, the proper place to 
post is on the Linux forum at Nvidia:

https://devtalk.nvidia.com/default/board/98/linux/

You may find these threads of interest:

https://devtalk.nvidia.com/default/topic/572292/-solved-does-gddccontrol-work-for-anyone-here-nvidia-i2c-monitor-display-ddc/

https://devtalk.nvidia.com/default/topic/1025917/has-anyone-ever-gotten-ddc-ci-to-work-with-pascal-gtx10x0-cards-over-displayport-/

https://devtalk.nvidia.com/default/topic/1038239/linux/survey-gddccontrol-issues-with-nvidia-drivers-i2c-monitor-display-ddc-/

Regards,
Sanford

liangyongxiang, 2019-09-16:
I installed nouveau and uninstalled the nvidia driver, and then running ddcutil in the text terminal also failed. And I can't run X normally:
https://gist.github.com/liangyongxiang/64506c6f87385eab9cc8ec06c055b4f5

Now based on iGPU I use (wayland+sway) or (xorg+i3) to work very well. I gave up using nvidia.

rockowitz, 2019-09-17:
If the problem is independent of the driver, then it would seem that it 
lies in the nvidia gpu displayport connector. But if you can't run X 
normally then that suggests something else may be at play.

This problem clearly is external to ddcutil.  I'm afraid that there's no 
more help I can give you.  Good luck.

Regards,
Sanford

parkerlreed, 2019-11-27:
Well I finally ran across a supported monitor, so at least I know the driver side is working. (Acer SA230)

closed 2020-11-16 by rockowitz

Grawp, 2021-12-20:
> Well I finally ran across a supported monitor, so at least I know the driver side is working. (Acer SA230)

Have you got the SA230 working? It's not working for me on both HDMI and DVI (without adapters).

parkerlreed, 2022-03-15:
@Grawp As far as I recall it worked fine via i2c on an AMD GPU. Don't have that monitor currently as it had to be returned. 

(Sorry for the late reply. GitHub buries single notification repos out of sight)

parkerlreed, 2022-03-16:
@Grawp Turns out I'm bad a remembering what I have when I'm not at the hardware...

```
[parker@wolfcola ~]$ sudo modprobe i2c-dev
[sudo] password for parker: 
[parker@wolfcola ~]$ sudo ddcutil detect --verbose
Display 1
   I2C bus:  /dev/i2c-4
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      Is LVDS device:                     false
      /sys/bus/i2c/devices/i2c-4/name:   AMDGPU DM i2c hw bus 3
   EDID synopsis:
      Mfg id:               ACR
      Model:                SA230
      Product code:         1406
      Serial number:        
      Binary serial number: 2440050594 (0x917037a2)
      Manufacture year:     2019,  Week: 17
      EDID version:         1.3
      Extra descriptor:        
      Video input definition:    0x80 - Digital Input
      Supported features:
         DPMS standby
         DPMS suspend
         DPMS active-off
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.647, 0.341
      Green x,y:        0.318, 0.618
      Blue  x,y:        0.152, 0.059
      Extension blocks: 1
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 04 72 7e 05 a2 37 70 91   .........r~..7p.
      +0010   11 1d 01 03 80 3c 22 78 ee d9 05 a5 57 51 9e 27   .....<"x....WQ.'
      +0020   0f 50 54 b3 0c 00 71 4f 81 80 81 c0 81 00 95 00   .PT...qO........
      +0030   b3 00 d1 c0 01 01 2a 44 80 a0 70 38 27 40 30 20   ......*D..p8'@0 
      +0040   35 00 fd 1e 11 00 00 1a 02 3a 80 18 71 38 2d 40   5........:..q8-@
      +0050   58 2c 45 00 fd 1e 11 00 00 1e 00 00 00 fd 00 30   X,E............0
      +0060   4c 1f 54 12 00 0a 20 20 20 20 20 20 00 00 00 fc   L.T...      ....
      +0070   00 53 41 32 33 30 0a 20 20 20 20 20 20 20 01 c0   .SA230.       ..
   VCP version:         2.2
   Controller mfg:      RealTek
   Firmware version:    0.1
   Monitor returns DDC Null Response for unsupported features: false

```