rockowitz/ddcutil #91: Memleak in is_edp_device()

FedeDP, 2019-08-27:
Hi!

I spotted a memleak in i2c_bus_core.c::is_edp_device(): https://github.com/rockowitz/ddcutil/blob/master/src/i2c/i2c_bus_core.c#L375.
As you can see, `GPtrArray * lines = execute_shell_cmd_collect(cmd);` is never freed.
We should add a `g_ptr_array_free(lines, true);` at L385.

I cannot provide a PR right now, if you can wait i will.

rockowitz, 2019-08-27:
Federico,

Thanks for pointing out the leak.  It's a bit disconcerting, given that 
the code is run through Coverity Scan, but also minor, since this 
function is called exactly once for each detected display. (Not quite 
true, as the "ddcutil environment" command can call it a few times for 
each display, but that's of no import.)

No need to send a PR.  The patch is trivial enough to do.  It will go 
into the next release.  The bug doesn't warrant re-releasing. Actually, 
the proper solution, as the comment says, is to recode the function to 
use the Linux API instead of collecting the output of a shell command.

Regards,
Sanford

FedeDP, 2019-09-17:
I see, thanks! (Sorry for the huge delay, i was sure i had responded!)

closed 2019-10-25 by rockowitz