rockowitz/ddcutil #92: unable to set 100% brightness

532910, 2019-09-19:
```
# ddcutil getvcp 0x10
VCP code 0x10 (Brightness                    ): current value =   255, max value =  7000

# ddcutil setvcp 0x10 1000
Number must be in range 0..255 (for now at least):  1000
Invalid VCP value: 1000
```

`# ddcutil setvcp 0x10 255` sets brightness to 3.6%

rockowitz, 2019-09-19:
> |# ddcutil getvcp 0x10 VCP code 0x10 (Brightness ): current value = 
> 255, max value = 7000 # ddcutil setvcp 0x10 1000 Number must be in 
> range 0..255 (for now at least): 1000 Invalid VCP value: 1000 |
>
> |# ddcutil setvcp 0x10 255| sets brightness to 3.6%

You've encountered a very interesting case, which I want to explore more.

First, it appears that you're using an older release of ddcutil.  As of 
0.9.6, setvcp recognizes values in the range 0..65535, or 
x0000..xffff.   So you should be able to  issue the command "setvcp x10 
7000".  (But you're creating libddcutil.so.1, which implies that you are 
using 0.9.6 or 0.9.7.  This is confusing.)

More interesting is that your monitor appears to actually be using the 
high order bytes (MH, SH) for continuous VCP values.  This is an area of 
ambiguity in the specs.  The DDC/CI spec references them, but the MCCS 
(Monitor Control Command Set) spec does not.  In practice, I've never 
before encountered a monitor that appears to be using the high order 
byte for continuous (e.g. x10) values.  So that I can understand your 
system better, please run the following command and send the output as 
an attachment:
   sudo ddcutil interrogate 2>&1

Thanks.

Sanford

532910, 2019-09-19:
I had run ddcutil from debian accidentally, instead of builded one. git/master works fine.

closed 2019-09-19 by 532910