rockowitz/ddcutil #93: radeon support

532910, 2019-09-19:
I have a NEC LCD2090UXi monitor that works fine with i915 (except #92), but not with `[AMD/ATI] Cedar GL [FirePro 2460]` and radeon driver:

```
# ddcutil detect --verbose
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Display 1
   I2C bus:             /dev/i2c-1
      I2C address 0x50 (EDID) present: true 
      Is eDP device:                   false
      /sys/bus/i2c/devices/i2c-1/name: Radeon i2c hw bus 0x90
   EDID synopsis:
      Mfg id:           NEC
      Model:            LCD2090UXi
      Serial number:    86113300YB
      Manufacture year: 2008
      EDID version:     1.3
      Product code:     26288
      Extra descriptor: Unspecified
      Video input definition: 0x80 - Digital Input
      Supported features:
         DPMS standby
         DPMS suspend
         DPMS active-off
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.639, 0.342
      Green x,y:        0.290, 0.615
      Blue  x,y:        0.146, 0.072
      Extension blocks: 0
   EDID source: 
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 38 a3 b0 66 01 01 01 01   ........8..f....
      +0010   17 12 01 03 80 29 1f 78 ea a6 a5 a3 57 4a 9d 25   .....).x....WJ.%
      +0020   12 50 54 bf ef 80 31 59 45 59 61 59 71 4f 81 59   .PT...1YEYaYqO.Y
      +0030   81 99 a9 40 81 80 48 3f 40 30 62 b0 32 40 40 c0   ...@..H?@0b.2@@.
      +0040   13 00 98 32 11 00 00 1e 00 00 00 fd 00 32 55 1f   ...2.........2U.
      +0050   5c 11 00 0a 20 20 20 20 20 20 00 00 00 fc 00 4c   \...      .....L
      +0060   43 44 32 30 39 30 55 58 69 0a 20 20 00 00 00 ff   CD2090UXi.  ....
      +0070   00 38 36 31 31 33 33 30 30 59 42 0a 20 20 00 30   .86113300YB.  .0
   VCP version:         Detection failed
   Controller mfg:      DDC communication failed
   Firmware version:    DDC communication failed
   Monitor returns DDC Null Response for unsupported features: true
```

rockowitz, 2019-09-19:
Please send the interrogate output, or at least the output of "ddcutil 
probe --verbose".  I'd like to review all the features of your monitor 
so see if any others report continuous values  in the 0..65535 range  or 
seem to have other unexpected behavior.  Thanks.

532910, 2019-09-19:
<details>
<summary>radeon:</summary>

```
# ddcutil probe --verbose
Unable to get capabilities for monitor on [i2c: fh=3, busno=1].  Maximum DDC retries exceeded.
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Mfg id: NEC, model: LCD2090UXi, sn: 86113300YB

Capabilities for display on bus /dev/i2c-1

May support table reads:   true

Scanning all VCP feature codes for display [i2c: fh=3, busno=1]

Getting data for non-table VCP code 0x00 - Unknown feature:
VCP code 0x00 (Unknown feature               ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]
VCP code 0x01 (Degauss                       ): Write-only feature

Getting data for non-table VCP code 0x02 - New control value:
VCP code 0x02 (New control value             ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0x03 - Soft controls:
VCP code 0x03 (Soft controls                 ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]
VCP code 0x04 (Restore factory defaults      ): Write-only feature
VCP code 0x05 (Restore factory brightness/contrast defaults): Write-only feature
VCP code 0x06 (Restore factory geometry defaults): Write-only feature

Getting data for non-table VCP code 0x07 - Unknown feature:
VCP code 0x07 (Unknown feature               ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]
VCP code 0x08 (Restore color defaults        ): Write-only feature

Getting data for non-table VCP code 0x09 - Unknown feature:
VCP code 0x09 (Unknown feature               ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]
VCP code 0x0a (Restore factory TV defaults   ): Write-only feature

Getting data for non-table VCP code 0x0b - Color temperature increment:
VCP code 0x0b (Color temperature increment   ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0x0c - Color temperature request:
VCP code 0x0c (Color temperature request     ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0x0d - Unknown feature:
VCP code 0x0d (Unknown feature               ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0x0e - Clock:
VCP code 0x0e (Clock                         ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0x0f - Unknown feature:
VCP code 0x0f (Unknown feature               ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0x10 - Brightness:
VCP code 0x10 (Brightness                    ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0x11 - Flesh tone enhancement:
VCP code 0x11 (Flesh tone enhancement        ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0x12 - Contrast:
VCP code 0x12 (Contrast                      ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0x13 - Backlight control:
VCP code 0x13 (Backlight control             ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

[... 2052 lines cut ...]

Readable features declared in capabilities string: 02 0b 0c 10 12 14 16 18 1a 20 30 68 6c 6e 70 87 8c 9b 9c 9d 9e 9f a0 aa b6 df e3 f4 f5 f6 fa fb fc fe ff

MCCS (VCP) version reported by capabilities: 2.0
MCCS (VCP) version reported by feature 0xDf: 2.0

Features declared as readable capabilities but not found by scanning:
   Feature x0c - Color temperature request
   Feature x16 - Video gain: Red
   Feature x18 - Video gain: Green
   Feature x1a - Video gain: Blue
   Feature x9b - 6 axis hue control: Red
   Feature x9c - 6 axis hue control: Yellow
   Feature x9d - 6 axis hue control: Green
   Feature x9e - 6 axis hue control: Cyan
   Feature x9f - 6 axis hue control: Blue
   Feature xa0 - 6 axis hue control: Magenta

Features found by scanning but not declared as capabilities:
   Feature xd6 - Power mode

Unable to calculate color temperature from VCP features x0B and x0C
```

</details>

rockowitz, 2019-09-20:
The radeon output is quite pathological.  Are the intel and radeon 
outputs connected to the monitor at the same time?   If so, was the 
display set to the intel input when the radeon test was done?

On 9/19/19 10:36 AM, sergio wrote:
> radeon:
> |# ddcutil probe --verbose Unable to get capabilities for monitor on 
> [i2c: fh=3, busno=1]. Maximum DDC retries exceeded. Output level: 
> Verbose Reporting DDC data errors: false Trace groups active: none 
> Traced functions: none Traced files: none Force I2C slave address: 
> false User defined features: disabled Mfg id: NEC, model: LCD2090UXi, 
> sn: 86113300YB Capabilities for display on bus /dev/i2c-1 May support 
> table reads: true Scanning all VCP feature codes for display [i2c: 
> fh=3, busno=1] Getting data for non-table VCP code 0x00 - Unknown 
> feature: VCP code 0x00 (Unknown feature ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] VCP code 0x01 (Degauss ): 
> Write-only feature Getting data for non-table VCP code 0x02 - New 
> control value: VCP code 0x02 (New control value ): Invalid response. 
> status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC 
> Null Message, Display_Handle[i2c: fh=3, busno=1] Getting data for 
> non-table VCP code 0x03 - Soft controls: VCP code 0x03 (Soft controls 
> ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all 
> tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1] 
> VCP code 0x04 (Restore factory defaults ): Write-only feature VCP code 
> 0x05 (Restore factory brightness/contrast defaults): Write-only 
> feature VCP code 0x06 (Restore factory geometry defaults): Write-only 
> feature Getting data for non-table VCP code 0x07 - Unknown feature: 
> VCP code 0x07 (Unknown feature ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] VCP code 0x08 (Restore 
> color defaults ): Write-only feature Getting data for non-table VCP 
> code 0x09 - Unknown feature: VCP code 0x09 (Unknown feature ): Invalid 
> response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries 
> returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1] VCP code 
> 0x0a (Restore factory TV defaults ): Write-only feature Getting data 
> for non-table VCP code 0x0b - Color temperature increment: VCP code 
> 0x0b (Color temperature increment ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] Getting data for non-table 
> VCP code 0x0c - Color temperature request: VCP code 0x0c (Color 
> temperature request ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] Getting data for non-table 
> VCP code 0x0d - Unknown feature: VCP code 0x0d (Unknown feature ): 
> Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all 
> tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1] 
> Getting data for non-table VCP code 0x0e - Clock: VCP code 0x0e (Clock 
> ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all 
> tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1] 
> Getting data for non-table VCP code 0x0f - Unknown feature: VCP code 
> 0x0f (Unknown feature ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] Getting data for non-table 
> VCP code 0x10 - Brightness: VCP code 0x10 (Brightness ): Invalid 
> response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries 
> returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1] Getting 
> data for non-table VCP code 0x11 - Flesh tone enhancement: VCP code 
> 0x11 (Flesh tone enhancement ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] Getting data for non-table 
> VCP code 0x12 - Contrast: VCP code 0x12 (Contrast ): Invalid response. 
> status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC 
> Null Message, Display_Handle[i2c: fh=3, busno=1] Getting data for 
> non-table VCP code 0x13 - Backlight control: VCP code 0x13 (Backlight 
> control ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] Getting data for non-table 
> VCP code 0x14 - Select color preset: VCP code 0x14 (Select color 
> preset ): Invalid response. status 
[... 886 lines cut ...]
> Message, Display_Handle[i2c: fh=3, busno=1] Getting data for non-table 
> VCP code 0xf7 - Manufacturer Specific: VCP code 0xf7 (Manufacturer 
> Specific ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] Getting data for non-table 
> VCP code 0xf8 - Manufacturer Specific: VCP code 0xf8 (Manufacturer 
> Specific ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] Getting data for non-table 
> VCP code 0xf8 - Manufacturer Specific: VCP code 0xf8 (Manufacturer 
> Specific ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
[... cut ...]
> VCP code 0xf9 - Manufacturer Specific: VCP code 0xf9 (Manufacturer 
> Specific ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] Getting data for non-table 
> VCP code 0xfa - Manufacturer Specific: VCP code 0xfa (Manufacturer 
> Specific ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] Getting data for non-t

532910, 2019-09-21:
Both test were done with the display set to the corresponding input.
Intel is compute stick with the only one monitor. Radeon is quad-head FirePro 2460 with 3 LCD2090UXi connected.

532910, 2019-09-21:
This test was done with only one monitor connected to radeon.

```
% xrandr                   
Screen 0: minimum 320 x 200, current 1200 x 1600, maximum 8192 x 8192
DisplayPort-0 connected primary 1200x1600+0+0 left (normal left inverted right x axis y axis) 408mm x 306mm
   1600x1200     60.00*+
   1280x1024     85.02    75.02    60.02  
   1280x960      85.00  
   1152x864      75.00  
   1024x768      85.00    75.03    70.07    60.00  
   832x624       74.55  
   800x600       85.06    72.19    75.00    60.32    56.25  
   640x480       85.01    75.00    72.81    66.67    59.94  
   720x400       70.08  
DisplayPort-1 disconnected (normal left inverted right x axis y axis)
DisplayPort-2 disconnected (normal left inverted right x axis y axis)
DisplayPort-3 disconnected (normal left inverted right x axis y axis)
```

<details>
<summary>radeon:</summary>

```
# ddcutil probe --verbose
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Mfg id: NEC, model: LCD2090UXi, sn: 86113300YB

Capabilities for display on bus /dev/i2c-1
Unable to get capabilities for monitor on [i2c: fh=3, busno=1].  Maximum DDC retries exceeded.

May support table reads:   true

Scanning all VCP feature codes for display [i2c: fh=3, busno=1]

Getting data for non-table VCP code 0x00 - Unknown feature:
VCP code 0x00 (Unknown feature               ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]
VCP code 0x01 (Degauss                       ): Write-only feature

Getting data for non-table VCP code 0x02 - New control value:
VCP code 0x02 (New control value             ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0x03 - Soft controls:
VCP code 0x03 (Soft controls                 ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]
VCP code 0x04 (Restore factory defaults      ): Write-only feature
VCP code 0x05 (Restore factory brightness/contrast defaults): Write-only feature
VCP code 0x06 (Restore factory geometry defaults): Write-only feature

Getting data for non-table VCP code 0x07 - Unknown feature:
VCP code 0x07 (Unknown feature               ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]
VCP code 0x08 (Restore color defaults        ): Write-only feature

Getting data for non-table VCP code 0x09 - Unknown feature:
VCP code 0x09 (Unknown feature               ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]
VCP code 0x0a (Restore factory TV defaults   ): Write-only feature

Getting data for non-table VCP code 0x0b - Color temperature increment:
VCP code 0x0b (Color temperature increment   ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0x0c - Color temperature request:
VCP code 0x0c (Color temperature request     ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0x0d - Unknown feature:
[... 790 lines cut ...]
Getting data for non-table VCP code 0xfd - Manufacturer Specific:
VCP code 0xfd (Manufacturer Specific         ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0xfd - Manufacturer Specific:
VCP code 0xfd (Manufacturer Specific         ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0xfe - Manufacturer Specific:
VCP code 0xfe (Manufacturer Specific         ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0xfe - Manufacturer Specific:
VCP code 0xfe (Manufacturer Specific         ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0xff - Manufacturer Specific:
VCP code 0xff (Manufacturer Specific         ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Getting data for non-table VCP code 0xff - Manufacturer Specific:
VCP code 0xff (Manufacturer Specific         ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Unable to read or parse capabilities.
Skipping comparison of declared capabilities to observed features

Unable to calculate color temperature from VCP features x0B and x0C
```

</details>

rockowitz, 2019-09-21:
First, thank you for your help in diagnosing this problem.

It seems that there's something wrong at the DDC or I2C level in the 
radeon case. Every request to the monitor returns a DDC Null Response,

 From the definition of the Null Response in the DDC/CI spec, this is a 
specific byte sequence intended for the following cases:
- (1) To detect that the display is DDC/CI capable (by reading it at the 
I2C slave address)
- (2) To tell the host that the display does not have any answer to give 
to the host (not ready or not expected)
- (3) The "Enable Application Report" has not been sent before using 
Application Messages.

And from the Communications Protocol section of the spec:

Display shall reply with a "null" message if it cannot reply in time or 
the command is not supported.

Some monitors misuse the Null Response to indicate that a feature is 
unsupported, in response to a Get VCP Feature request (getvcp command).  
This is incorrect.  The command (getvcp) is supported, and the monitor 
should respond with the unsupported feature bit set in a normal reply.  
Since the unsupported feature bit is properly set when using i915, we 
know that such misuse is not case for the monitor's DDC/CI 
implementation.  So we're left with case (2) above.

Note: The test that ddcutil performs to determine whether the monitor 
misuses the Null Response is this way is not unambiguous. In your case 
ddcutil gets it wrong and thinks the Null Response to a feature request 
indicates unsupported feature.

Conceivably, the monitor's DDC/CI implementation is marginal, being good 
enough when driven by the i915's I2C implementation but not when driven 
by radeon driver.

First, help me understand your (radeon) environment by running the full 
set of diagnostics:

   sudo ddcutil interrogate --verbose 2>&1

Second, try running with DDC error reports enabled, and retry statistics 
turned on:

   ddcutil getvcp 10 --ddc --stats

Third, try executing the getvcp command with undocumented option 
--sleep-strategy 2.  This doubles the amount of time (from that mandated 
by the spec) that ddcutil waits between sending a request and reading 
the response.

   ddcutil getvcp 10 --ddc --stats --sleep-strategy 2

Based on the output of these commands I expect that I'll either ask you 
to run ddcutil with some specific tracing turned on, and/or put up a 
patched copy on github for you to try.

Thanks again,
Sanford

On 9/21/19 6:55 AM, sergio wrote:
>
> This test was done with only one monitor connected to radeon.
>
> |% xrandr Screen 0: minimum 320 x 200, current 1200 x 1600, maximum 
> 8192 x 8192 DisplayPort-0 connected primary 1200x1600+0+0 left (normal 
> left inverted right x axis y axis) 408mm x 306mm 1600x1200 60.00*+ 
> 1280x1024 85.02 75.02 60.02 1280x960 85.00 1152x864 75.00 1024x768 
> 85.00 75.03 70.07 60.00 832x624 74.55 800x600 85.06 72.19 75.00 60.32 
> 56.25 640x480 85.01 75.00 72.81 66.67 59.94 720x400 70.08 
[... 908 lines cut ...]
> Specific ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] Getting data for non-table 
> VCP code 0xf2 - Manufacturer Specific: VCP code 0xf2 (Manufacturer 
> Specific ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] Getting data for non-table 
> VCP code 0xf2 - Manufacturer Specific: VCP code 0xf2 (Manufacturer 
> Specific ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] Getting data for non-table 
> VCP code 0xf3 - Manufacturer Specific: VCP code 0xf3 (Manufacturer 
> Specific ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] Getting data for non-table 
> VCP code 0xf3 - Manufacturer Specific: VCP code 0xf3 (Manufacturer 
> Specific ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] Getting data for non-table 
> VCP code 0xf4 - Manufacturer Specific: VCP code 0xf4 (Manufacturer 
> Specific ): Invalid response. status 
> code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null 
> Message, Display_Handle[i2c: fh=3, busno=1] Getting data for non-table 
> VCP code 0xf4 - Manufacturer Specific: VCP code 0xf4 (Manufacturer 
> Specific ): Inval

532910, 2019-09-24:
> First, thank you for your help in diagnosing this problem.

Same here! (:
[ddcutil.txt](https://github.com/rockowitz/ddcutil/files/3649589/ddcutil.txt)

```# ddcutil interrogate --verbose``` see attached file due to ```your comment is too long (maximum is 65536 characters). ```

<details>
<summary>

```# ddcutil getvcp 10 --ddc --stats```
</summary>

```
VCP (aka MCCS) version for display is undetected or less than 2.0. Output may not be accurate.
VCP code 0x10 (Brightness                    ): Invalid response. status code=DDCRC_ALL_RESPONSES_NULL(-3011): all tries returned DDC Null Message, Display_Handle[i2c: fh=3, busno=1]

Maximum Try Settings:
Operation Type                    Current  Default
Write only exchange tries:              4        4
Write read exchange tries:             10       10
Multi-part read exchange tries:         8        8
Multi-part write exchange tries:        8        8

Retry statistics for ddc write only
   No tries attempted

Retry statistics for ddc write/read
   Max tries allowed: 10
   Successful attempts by number of tries required:
       1:    0
       2:    0
       3:    0
       4:    0
       5:    0
       6:    0
       7:    0
       8:    0
       9:    0
      10:    0
   Total:                              0
   Failed due to max tries exceeded:   0
   Failed due to fatal error:          7
   Total attempts:                     7

Retry statistics for multi-part read exchange
   No tries attempted

Retry statistics for multi-part write exchange
   No tries attempted

DDC Related Errors:  
Count   Status Code                          Description
   16   DDCRC_NULL_RESPONSE          (-3002) received DDC null response 
    7   DDCRC_ALL_RESPONSES_NULL     (-3011) all tries returned DDC Null Message 
Total errors: 23

Errors Wrapped in Retry:  
Count   Status Code                          Description
   16   DDCRC_NULL_RESPONSE          (-3002) received DDC null response 
Total errors: 16

Sleep Strategy Stats:
   Total IO events:        127
   IO error count:          23
   Total sleep events:      45

   Sleep Event type      Count
   SE_WRITE_TO_READ         16
   SE_POST_OPEN              4
[... 85 lines cut ...]
   SE_WRITE_TO_READ         16
   SE_POST_OPEN              4
   SE_POST_WRITE             0
   SE_POST_READ             16
   SE_DDC_NULL               9
   SE_POST_SAVE_SETTINGS     0

Call Stats:
   Type                                     Count    Millisec  (      Nanosec)
   write calls       (IE_WRITE)                46          89  (     89766109)
   read calls        (IE_READ)                 19         155  (    155764445)
   open file calls   (IE_OPEN)                 17           0  (       216154)
   close file calls  (IE_CLOSE)                17           0  (       125326)
   other I/O calls   (IE_OTHER)                28           0  (        63141)
   Totals:                                    127         245  (    245935175)

Sleep Call Stats:
   Total sleep calls:                                      57
   Requested sleep time milliseconds :                   5000
   Actual sleep milliseconds (nanosec):                  5004  (   5004177289)

Total elapsed milliseconds (nanoseconds):                5295  (   5295273348)
zsh: exit 1     ddcutil getvcp 10 --ddc --stats --sleep-strategy 2
```
</details>

rockowitz, 2019-09-27:
I have uploaded a new ddcutil branch, 0.9.8, which contains a temporary, 
hardcoded change that forces increasingly long waits for each retry when 
the DDC Null Message is received.

Try executing the command:
# ddcutil getvcp 10 --stats

I expect you'll see the following message:
   Extended delay as recovery from DDC Null Response

and multiple messages regarding increasingly long sleeps.

Let me know if this resolves the problem.   (I doubt it will, but it's 
worth a try.)

If it does, please send the output.

Sanford

532910, 2019-09-27:
There was no `DDC Null Response` message, but a lot of `Sleep event type` / `sleep_multiplier`.

<detauls>
<summary>The full output:</summary>

```
(set_sleep_multiplier) Setting sleep_multiplier = 2
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 2, sleep_time_millis = 100
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 2, sleep_time_millis = 100
(set_sleep_multiplier) Setting sleep_multiplier = 3
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 3, sleep_time_millis = 150
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 3, sleep_time_millis = 150
(set_sleep_multiplier) Setting sleep_multiplier = 4
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 4, sleep_time_millis = 200
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 4, sleep_time_millis = 200
(set_sleep_multiplier) Setting sleep_multiplier = 5
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 5, sleep_time_millis = 250
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 5, sleep_time_millis = 250
(set_sleep_multiplier) Setting sleep_multiplier = 6
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 6, sleep_time_millis = 300
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 6, sleep_time_millis = 300
(set_sleep_multiplier) Setting sleep_multiplier = 7
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 7, sleep_time_millis = 350
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 7, sleep_time_millis = 350
(set_sleep_multiplier) Setting sleep_multiplier = 1
(set_sleep_multiplier) Setting sleep_multiplier = 2
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 2, sleep_time_millis = 100
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 2, sleep_time_millis = 100
(set_sleep_multiplier) Setting sleep_multiplier = 3
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 3, sleep_time_millis = 150
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 3, sleep_time_millis = 150
(set_sleep_multiplier) Setting sleep_multiplier = 4
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 4, sleep_time_millis = 200
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 4, sleep_time_millis = 200
(set_sleep_multiplier) Setting sleep_multiplier = 5
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 5, sleep_time_millis = 250
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 5, sleep_time_millis = 250
(set_sleep_multiplier) Setting sleep_multiplier = 6
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 6, sleep_time_millis = 300
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 6, sleep_time_millis = 300
(set_sleep_multiplier) Setting sleep_multiplier = 7
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 7, sleep_time_millis = 350
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 7, sleep_time_millis = 350
(set_sleep_multiplier) Setting sleep_multiplier = 1
(set_sleep_multiplier) Setting sleep_multiplier = 2
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 2, sleep_time_millis = 100
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 2, sleep_time_millis = 100
(set_sleep_multiplier) Setting sleep_multiplier = 3
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 3, sleep_time_millis = 150
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 3, sleep_time_millis = 150
(set_sleep_multiplier) Setting sleep_multiplier = 4
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 4, sleep_time_millis = 200
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 4, sleep_time_millis = 200
(set_sleep_multiplier) Setting sleep_multiplier = 5
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 5, sleep_time_millis = 250
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 5, sleep_time_millis = 250
(set_sleep_multiplier) Setting sleep_multiplier = 6
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 6, sleep_time_millis = 300
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 6, sleep_time_millis = 300
(set_sleep_multiplier) Setting sleep_multiplier = 7
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 7, sleep_time_millis = 350
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 7, sleep_time_millis = 350
(set_sleep_multiplier) Setting sleep_multiplier = 1
(set_sleep_multiplier) Setting sleep_multiplier = 2
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 2, sleep_time_millis = 100
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 2, sleep_time_millis = 100
(set_sleep_multiplier) Setting sleep_multiplier = 3
(call_tuned_sleep) Sleep event type: SE_WRITE_TO_READ, sleep_multiplier = 3, sleep_time_millis = 150
(call_tuned_sleep) Sleep event type: SE_POST_READ, sleep_multiplier = 3, sleep_time_millis = 150
(set_sleep_multiplier) Setting sleep_multiplier = 4
[... 122 lines cut ...]
   Sleep Event type      Count
   SE_WRITE_TO_READ         49
   SE_POST_OPEN              4
[... cut ...]
   close file calls  (IE_CLOSE)                18           0  (       144190)
   other I/O calls   (IE_OTHER)                28           0  (        68164)
   Totals:                                    195         454  (    454504448)

Sleep Call Stats:
   Total sleep calls:                                     156
   Requested sleep time milliseconds :                  35100
   Actual sleep milliseconds (nanosec):                 35111  (  35111713325)

Total elapsed milliseconds (nanoseconds):               35635  (  35635244955)
```
</details>

rockowitz, 2019-09-27:
I neglected to tell you to use the --verbose option.  However the final 
line of output indicates all the failures and DDC Null Message, and the 
statistics confirm that.  So slowing down DDC communication by 
increasing sleep times between writes and reads is not going to solve 
the problem.

So we're left with a couple possibilities:

1) It may be that the standard I2C bus timing is just too fast for your 
older card.  Unfortunately, there's no way for a userspace program like 
ddcutil to tell radeon driver to slow down.

2) According to the interrogate output, each of your displays is 
connected to a DisplayPort output on the video card.  However, per the 
specs the LCD2090UXi monitor does not have a DisplayPort input.   So you 
must be using DisplayPort to DVI converters or cables.  DisplayPort I2C 
transmission is frequently problematic - the I2C signal is carried 
logically on the DisplayPort AUX channel.   See, for example, this long 
running thread for the i915 driver: 
https://bugs.freedesktop.org/show_bug.cgi?id=100954

I can't tell if your FirePro card has any DVI or HDMI connectors. If it 
does, try connecting one of your monitors that way.

rockowitz, 2019-09-30:
The 0.9.8 branch has been updated.
- The special handling and diagnostic messages relating to your problem have been disabled or removed.
- Undocumented command line option **-y** no longer affects sleep strategy. 
- There is a new command line option **--sleep-multplier** that provides a more general way to alter DDC mandated sleeps.  It takes a floating point number as an argument, e.g. ***--sleep-multiplier 2***, ***--sleep-multiplier .5***.  This option causes the normal sleep times mandated by the DDC/CI spec to be multiplied by the value specified.  For example, if  **sleep-multiplier** is 1.5, then sleeps that would normally be 50 ms are now 75 ms.

Brief testing shows that a **--sleep-multiplier** value greater than 1 stabilizes communication with at least one problematic monitor.  Other monitors can communicate reliably with values as small as .1, greatly speeding up **ddcutil** execution time.

532910, 2019-09-30:
https://en.wikipedia.org/wiki/DisplayPort#DisplayPort_dual-mode_(DP++)

LCD2090UXi are connected to FirePro 2460 directly (via dvi-dvi cable and passive minidp-dvi pigtail that comes with videocard)

532910, 2019-09-30:
I've

```
% git clone https://github.com/rockowitz/ddcutil.git
% cd ddcutil
% git checkout origin/0.9.8
% ./autogen.sh
```
and it replied `ddcutil 0.9.8`, but:

`Unknown option --sleep-multplier`

532910, 2019-09-30:
```
% s ./src/ddcutil --version
ddcutil 0.9.8
```

and `--sleep-multiplier=number` is present in `--help` output

532910, 2019-09-30:
Oh, just a typo mistake, that I copied from
> There is a new command line option **--sleep-multplier** that provides

532910, 2019-09-30:
One more amazing experiment. I have an old VGA-only DELL E156FP monitor and Apple miniDP-VGA adapter (it's active, I'm absolutely sure). And this monitor (via this adaptor) successfully works with FirePro 2460, (can change brightness level).

532910, 2019-09-30:
(And NEC 2090UXi via this Apple miniDP-VGA adapter also works fine.)

532910, 2019-09-30:
Hm, I've just remembered, that I've heard about DVI-HDMI adapters, that are passive of course, as HDMI has electrically same interface as DVI (digital part), but contains IC connected to I2C/DDC line.

Multimeter shows that
1. something is present on  DDC lines in my DVI-MINIDP adapters,
2. there are no directly connected data lines, only shields.

I'm sorry bothering you for and grateful for your help. Let me know if you need some help with the project.

closed 2019-09-30 by 532910

rockowitz, 2019-10-01:
Recall that the DDC signals are carried over the DisplayPort AUX channel 
(pins 15, 17).  The AUX channel is a more general signaling mechanism 
than I2C.  There's a protocol for carrying logical I2C signals (along 
with other other signals) over the AUX channel.  There are no hardwired 
connections.

What kind of help might you be able to provide?  There's a lot of work 
to be done on ddcui.  The existing code is essentially an extended proof 
of concept.  It needs to be cleaned up and hardened both in its use of 
Qt and C++.  I pretty much jumped from C to OO programming in Java, and 
the C++ code reflects that.  ddcui needs to be packaged for both debian 
and rpm based distros.  Plus, as you've seen, the use of cmake and qmake 
needs to be generalized.

Sanford

532910, 2019-10-15:
I solved the issue!!! It was due to `hw_i2c` `radeon` module parameter. It is `0` by default, but it was set to `1` in my `/etc/modprobe.d/radeon.conf` I don't remember how and for.
I think you should add this case into documentation. Moreover, other video modules may have other parameterers that may affect ddcutil. (Once again, just the fact: DP->VGA adapter works with any `hw_i2c` parameter's value, DP->DVI works only with `hw_i2c=0`. There is `auxch` parameter that doesn't affect ddcutil at all.)

> What kind of help might you be able to provide?
I don't know (: I think my strengths are see the flaws in everything and solve problems.
I don't do much programming and don't like c++. I never created rpm packages and not a debian maintainer, but can take this task.