rockowitz/ddcutil #95: getvcp doesn't get updated values

p3lim, 2019-10-14:
I have one monitor (HP Z24n) with two computers connected to it, both by DP.
```
$ ddcutil -d 1 capabilities
MCCS version: 2.2
VCP Features:
  ...
  Feature: 60 (Input Source)
    Values:
        03: DVI-1
        0f: DisplayPort-1
        10: DisplayPort-2
        11: HDMI-1
  ...
```

Computer 1 is connected to DP-1, computer 2 to DP-2, both get the same capabilities.
If I run `ddcutil -d 1 getvcp 60` when the monitor is set to input DP-1 it returns `0x10`, as expected. Running `ddcutil -d 1 setvcp 60 0x0f` correctly sets the input on the monitor, but running the same `getvcp` command on computer 1 still yields `0x10`, not `0x0f`. On computer 2 the `getvcp` command returns `0x0f` as expected when the input is for it, but also doesn't change to `0x10` when I run the `setvcp` command with `0x10` there.

TL;DR: `getvcp` doesn't seem to get the latest/correct data, or I'm misunderstanding how it's supposed to work. I just need to see what input is currently active.

rockowitz, 2019-10-14:
What you're seeing is not unexpected monitor behavior.

Feature x60 is unusual, in that it instructs the monitor to take its 
input from somewhere else. How the monitor responds to additional 
commands in this situation depends on the monitor's DDC implementation.  
Some monitors have one set of DDC settings and respond the same on 
whatever input getvcp or setvcp comes on. Others will return an error if 
not on the current input.  And in your case the monitor is apparently 
returning the old value for the input.

In short, how the monitor responds to requests not on its current input 
depends on the monitor.

Sanford

p3lim, 2019-10-15:
That's a shame. Is there any other way, with ddcutil or not, that you're aware of that would be able to detect if there are monitors active on the system? It's DP, which afaik signals the system when it's not connected, but I'm unsure about the details.

A bit offtopic for ddcutil, but it would be of great help regardless.

rockowitz, 2019-10-16:
DDC communication is a separate question from detecting monitors.  If a 
monitor is connected, its EDID can be read on the /dev/i2cN associated 
with the monitor at slave address x50.  This (I dare to say) always 
works.  It does not matter what input is selected on the monitor.  
Basically, what ddcutil does is check every I2C bus for the presence of 
an EDID. (It's actually more sophisticated, but that's sufficient for 
this discussion.) If an EDID can be read then ddcutil has found a 
monitor.  It then checks whether DDC communication is working at slave 
address x37.  That may or may not be the case.  For example laptop 
displays have an EDID, but never support DDC.

If DDC communication fails for a monitor, "ddcutil detect" reports it as 
an invalid display, and does not assign a display number.

The /dev/i2c devices are just one way the video driver can use the I2C 
bus to get the EDID without
Subsystems other than I2C report the EDID without explicitly using a 
/dev/i2c device.  For example, X11's xrandr command reports connected 
monitors.  surfacing a reference to a /dev/i2c device. To see many of 
the ways that an EDID can be accessed from userspace, the the Device 
Identifier Cross Reference Report at (nearly) the end of the output from 
"ddcutil environment --verbose".

- Sanford

closed 2019-10-25 by rockowitz