rockowitz/ddcutil #96: "DDC communication failed", display is detected though

marmaladeseismic, 2019-10-26:
With Raspbian 9 on RPi 2 using HDMI, ddcutil it seems is able to detect some basic info about a display but communication fails e.g. in `getvcp`. Attached are files I saw you requested for similar issues.

Let me know if I should provide anything else!

Also thank you for this great utility, I'm using it to automatically adjust monitor brightness in my office over the day :) hope I can do the same with the TV soon!

[sudo ddcutil detect --verbose --ddc --force-slave-address](https://github.com/rockowitz/ddcutil/files/3775071/detect_verbose.txt)
[sudo ddcutil environment --verbose](https://github.com/rockowitz/ddcutil/files/3775070/environment_verbose.txt)

rockowitz, 2019-10-26:
First, a basic question.  Do you know for certain that the Sony TV 
supports DDC? The i2cdetect command shows slave address x50 (EDID) is 
active on /dev/i2c-2, but x37 (DDC) is not.

marmaladeseismic, 2019-10-27:
No I'm not certain. I didn't realize the difference between EDID and DDC. So all the info gathered then was via EDID?

Any recommendation on determining whether DDC is supposed to work? I can't find anything this technical in the product documentation or TV menus.

Also curious if it isn't enabled then what is the chance that the firmware is capable but the feature is disabled, and then I dunno JTAG?..

rockowitz, 2019-10-28:
Most monitors the support DDC state so in their (full) specifications, 
and have a switch in their on-screen display (OSD) that controls whether 
it is enabled. Occasionally I encounter a monitor that supports DDC but 
does not have a setting in the OSD, in which case DDC is always enabled.

If you can, hook up a regular computer (i.e. not a Pi) to the TV and run 
i2cdetect for the appropriate /dev/i2c bus number. (Alternatively, run 
"ddcutil environment --verbose"  which will probe all potential bus 
numbers for you.)
I doubt you'll see anything different, but I suggest it because properly 
configuring the Pi is finicky.

Sanford

marmaladeseismic, 2019-11-06:
I finally got around to trying another machine, unfortunately it's using nvidia drivers but it does work with a modern monitor over DisplayPort. Still, HDMI to TV did not work. Same messages as the RPi. I guess the TV doesn't support DDC (currently).

closed 2019-11-06 by marmaladeseismic