rockowitz/ddcutil #97: ddcutil not working on Raspberry Pi 4

fm99, 2019-10-26:
ddcutil not working on Raspberry Pi 4

I have tried to use ddcutil (v0.9.2 from default raspian buster repo) on my new Pi4 - unfortunately without success: When I enter "ddcutil detect" I get the message "No diplays found".

Since I use Kodi, I must use on the "GL (Fake KMS) OpenGL desktop driver with fake KMS" (see https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=251645).

I then read the documentation about the Raspberry Pi on www.ddcutil.com and played around with the i2c related dtparam settings. Still no success.

Finally, I downloaded the source code of the latest version of ddcutil (v0.9.7) and compiled it. That didn't help either.

Please find enclosed my current /boot/config.txt and various ddcutil diagnostic reports:

- [boot-config.txt](https://github.com/rockowitz/ddcutil/files/3775326/boot-config.txt)

- [ddcutil detect --verbose.txt](https://github.com/rockowitz/ddcutil/files/3775327/ddcutil.detect.--verbose.txt)

- [ddcutil environment --verbose.txt](https://github.com/rockowitz/ddcutil/files/3775328/ddcutil.environment.--verbose.txt)

- [ddcutil interrogate.txt](https://github.com/rockowitz/ddcutil/files/3775329/ddcutil.interrogate.txt)

Thank you for any help.

rockowitz, 2019-10-27:
Are you and user marmaladeseismic the same?   Rather a coincidence to 
see to posts re the Raspberry Pi on the same day.

Looking at your environment command output, there are no slave addresses 
active on /dev/i2c-2, including x50 and x37.  This is the case whether 
using ddcutil's I2C access, or command i2cdetect.  So it would appear to 
be an I2C configuration problem.  The Raspberry Pi is not a platform I 
normally test on - the last time I looked at it in any depth was over a 
year ago.   A couple questions come to mind, but they're pretty basic.  
What happens if you do not use Kodi?  Does ddcutil work for you on a Pi 3?

Note that for testing purposes, you can use command i2cdetect.  If 
i2cdetect does not find slave addresses x50 and x37 on the I2C bus, 
there's no point in trying to use ddcutil.

Sorry that I can't be of more help.  If you do find a solution please 
post it here so that others can see it and I can include it in the 
documentation.

Sanford

fm99, 2019-10-27:
Hi Sanford

Thanks for your answer.

As for your questions/point:

1. "Are you and user marmaladeseismic the same?" => No, I'm not that other user, it's just a coincidence.
2. "What happens if you do not use Kodi?" => If I configure the "Original non-GL desktop driver" (i.e. if no i2c related dtparam line is set in /boot/config.txt) ddcutil doesn't work either ("No diplays found").
3. "Does ddcutil work for you on a Pi 3?" => I have two Pi 3B+, one running debian/raspbian 9 (stretch) and one running debian/raspbian 10 (buster). On both Pis ddcutil runs as expected, i.e. I can read/change the settings of my monitor without any problems.
4. "Sorry that I can't be of more help.  If you do find a solution please post it here so that others can see it and I can include it in the documentation." => That is not a problem. I was pleased that you reacted so quickly to my request. If I find a solution, I will describe it as a comment here. Thanks again for your help!

bangom, 2020-01-02:
> Hi Sanford
> 
> Thanks for your answer.
> 
> As for your questions/points:
> 
> 1. "Are you and user marmaladeseismic the same?" => No, I'm not that other user, it's just a coincidence.
> 2. "What happens if you do not use Kodi?" => If I configure the "Original non-GL desktop driver" (i.e. if only one or more I2C dtparam lines but no GL driver related dtoverlay line is set in /boot/config.txt), ddcutil doesn't work either ("No diplays found").
> 3. "Does ddcutil work for you on a Pi 3?" => I have two Pi 3B+, one running debian/raspbian 9 (stretch) and one running debian/raspbian 10 (buster). On both Pis ddcutil runs as expected, i.e. I can read/change the settings of my monitor without any problems.
> 4. "Sorry that I can't be of more help.  If you do find a solution please post it here so that others can see it and I can include it in the documentation." => That is not a problem. I was pleased that you reacted so quickly to my request. If I find a solution, I will describe it as a comment here. Thanks again for your help!
> 
> Update: I found this discussion on the kernel source tree for the Pi: [raspberrypi/linux#3152](https://github.com/raspberrypi/linux/issues/3152).
> As I understand it, I2C support for the Pi4 is still incomplete at the moment, so ddcutil cannot be used on a Pi4 for now :-/

IMHO, this is not a ddcutil problem. I2C on RPI4 is OK in general...  The problem with ddcutil on RPI4 is that HDMI DDC/CI (I2C) is not accessible in userspace (/dev/i2c-2) due to completely new HW comparing to RPI3 (RPI4 drivers need to be fixed).

As a workaround, you can physically connect HDMI DDC/CI to free i2c bus on RPI4 and control it with ddcutil - details see my post here: https://github.com/raspberrypi/linux/issues/3152.

rockowitz, 2020-01-02:
Thanks for the detailed report.  I'll update the Pi documentation at 
www.ddcutil.com/raspberry.

- Sanford

rockowitz, 2020-01-02:
Page www.ddcutil.com/raspberry has been updated to reflect this 
information.

Thanks again,
- Sanford

robertalexa, 2021-03-26:
@rockowitz Hi,

Could you please offer your input on this?

To my knowledge, kernel 5.10 should have fixed this issue and ddcutil should now be working. However the [documentation](https://www.ddcutil.com/raspberry/) for ddcutil raspberry still says that it is unavailable for RPI4.

I am to assume that [this](https://github.com/raspberrypi/linux/issues/3152#issuecomment-768433812) comment is actually correct and the documentation above mentioned is outdated?
> Just to make it easier for everyone, a quick step by step:
> 
> 1. run `rpi-update`, this should install rpi firmware 5.10
> 2. in `/boot/config.txt` look for the `[pi4]` section and set this value `dtoverlay=vc4-kms-v3d`
> 3. reboot
> 4. `ddcutil detect` should now detect your screen (is ddc capable of course...)

Many thanks,
Rob

rockowitz, 2021-03-27:
@robertalexa You are correct.  The ddcutil Raspberry Pi documentation is badly out of date. I was not aware that I2C was now working over HDMI on the Pi 4.  I've been meaning to put together a Pi 4 for testing, but that's backed up behind a lot of other things (like getting out a new release).  

If you'd be willing to send me modified content for ddcutil's Pi page, I'll modify the page accordingly.

robertalexa, 2021-03-27:
@rockowitz at the minute i am stacked, but i am hoping that some time this weekend i will find the time to test this at my end, make note of requirements and after that i will post another reply for the documentation. Leave this with me for a bit

robertalexa, 2021-03-27:
@rockowitz hey man. Just had a chance to have a play with this and everything is good :)

In order for this to work:
1. make sure you are at least on firmware 5.10 - this is now mainstream, so `sudo apt update && sudo apt upgrade` will be sufficient.
2. In `/boot/config.txt` look for the `[pi4]` section and set replace `dtoverlay=vc4-fkms-v3d` with `dtoverlay=vc4-kms-v3d`
3. In `/etc/modules` add `i2c_dev`
4. Reboot
5. Install ddcutil - `sudo apt install ddcutil`
6. Done

Check if everything is working and your monitor is detected: `ddcutil detect`
The output should be similar to this
`Display 1
   I2C bus:             /dev/i2c-12
   EDID synopsis:
      Mfg id:           BNQ
      Model:            BenQ GW2270
      Serial number:    AAF03393019
      Manufacture year: 2015
      EDID version:     1.3
   VCP version:         2.1
`
Find out what commands you can run on your monitor: `ddcutil capabilities`
Check the [command overview page](https://www.ddcutil.com/commands/) for how to read and write your monitor parameters.

-------------
Hope this is good enough to put on the Documentation page. 
With this in mind, you can probably close the Issue as everything is now working.

Offtopic: do you have any experience with controlling screen brightness based on room light using a LDR? Thanks

rockowitz, 2021-03-29:
Robert, 

I have updated the [ddcutil Raspberry Pi page](https://www.ddcutil.com/raspberry/) based on your documentation.  Let me know if it looks right to you.

I have no direct experience controlling screen brightness using a LDR, but it seems to me there are 3 parts to the question: 
- Reading the sensor (presumably on Pi). Googling finds numerous pages on how to do this, which I'm sure you're aware of.
- Setting monitor VCP feature x10 (brightness).  In C, you can use the i2c-dev API directly, or libddcutil which handles errors, retries, etc.  If coding in Python, just use its API to invoke ddcutil commands. Note that you'll probably want to read the current and maximum value of feature x10 as well.  Keep in mind that different monitors have different maximum values.  100 and 255 are common, but I've even seen instance where the value is a 2 byte number.
- The logic for determining the value of feature x10 based on the LDR output.  Well, that's up to you.

robertalexa, 2021-03-29:
@rockowitz I have seen the docs earlier today and it looks good :) nice one for updating it!

As to the LDR, i know how to achieve it on paper, i am just not sure how often i should be sending updates to a monitor without any risks. If you think about a phone for example, I am sure you know how the screen brightness gradually adjusts when changing the ambient light. It is after this effect that I am after.

So to achieve sth as smooth as that, there will be more than one single VCP value involved, probably 1 by 1 for the duration of say 1 second. But I am unsure if there are any risks associated with "flooding" the monitor with such quick successive changes.

Anyway, thanks for your input, it is much appreciated :)

rockowitz, 2021-03-29:
@robertalexa offtopic LDR discussion moved to issue #197

closed 2025-02-09 by rockowitz