rockowitz/ddcutil #99: ddcutil works with amdgpu.dc=0 but not with amdgpu.dc=1

Futureknows, 2019-12-13:
With amdgpu.dc=0 (disabled) ddcutil works correctly for adjusting all monitor settings. I need amdcpu.dc=1 to have audio through displayport to my monitor. When booted with amdcpu.dc=1, DDcutil usually works the first time it is used, then it no longer detects the monitor, until reboot. I2C permissions are correct.

Is there any workaround to have amdgpu.dc=1 and still have ddcutil working?

rockowitz, 2019-12-14:
This is, as they say, an interesting problem.  I've never heard of it 
before.  It's almost certainly an amdgpu driver issue, but let's see if 
we can drill down some.

amdgpu.dc controls whether the Display Core driver is enabled.  It takes 
3 values:
    1 = force enable
    0 = force disable
   -1 = enable by default for supported asics

My understanding is that Display Core is supported only in Vega, Raven 
Ridge, and later devices.

What happens if you boot with amdgpu.dc=-1 ? Is Display Core enabled or 
disabled?  If it is, "Display Core initialized" should be found in file 
/var/log/dmesg

I've tried booting this workstation with parm amdgpu.cd=1, and have not 
encountered the problem you describe.   The card is an old R9 280x, 
which predates Display Core.

If you are using a DisplayPort connection, what happens if you switch to 
DVI or HDMI (assuming that's possible).

I have a set of additional questions, but it would be best if you run 
the following command and send the output as an attachment.
#  ddcutil environment --verbose

Please ensure that package i2c-tools is installed beforehand.

Sanford

Futureknows, 2019-12-16:
Thanks for the detailed response. My RX580 seems haunted. I reinstalled the amdgpu and cleaned up the .conf files and booted with amdgpu.dc=1, and now OS sees both DP audio and DDC is functioning. 

The XFX RX580 has some notorious problems including dropping the DP signal entirely until machine is hard reset. Many people reported dc=0 solved the problem so I was running that way. My monitor doesn't have HDMI2.0 so I have to use DP to get 4k 60hz.

A related condition I found interesting is I hooked up a 5k monitor with a dp1.4 cable and with dc=0 the OS saw the monitor as 2560 max. When booting with dc=1 it detected and perfectly works at 5k60hz on a single DP1.4 cable (which is glorious).

Will keep posted if the DDC problem returns I have a feeling it is related to when the PC _and_ the monitor both enter deep sleep, then upon resume DDC link fails to reinitialize.

Futureknows, 2019-12-17:
Problem has returned with amdgpu.dc=1. 
**ddcutil environment --verbose** hangs with following output:

`ddcutil environment --verbose
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled
`

rockowitz, 2019-12-17:
Try running:
   ddcutil environment --verbose --trace env

At least that may give me an idea of where the hang occurs.

Sanford

Futureknows, 2019-12-17:
Same output, the parser freezes during the trace. 

`$ ddcutil environment --verbose --trace env
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        ENV
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled
`

It works the first time after booting.  However, with DC=0 it never seems to lose communication.

rockowitz, 2019-12-17:
Scratch that last request to run with "--trace env".   There won't be 
much useful information.

Do you have valgrind installed?   If so please run command "valgrind 
ddcutil environment -v".  If it hangs, I assume ctrl-c should terminate 
ddcutil, and valgrind should show where the hang is.

Re the amdgpu.dc problem, your original report said that ddcutil may 
work, then no longer detects the monitor.  Is it the case that ddcutil 
reads the EDID, then decides that the display doesn't support DDC and 
reports it as invalid, or that even the EDID read fails and ddcutil 
doesn't see the monitor at all?   It's conceivable that when using the 
Display Core driver that the I2C bus timing is different.  I've seen 
this "sometimes it works, sometimes it doesn't" behavior with marginal 
monitors.  And sometimes, slowing down the communication addresses the 
problem. To do so, use the "--sleep-multiplier" option, e.g. "--sleep 
multiplier 4".  This causes ddcutil to increase (or decrease) the sleeps 
mandated by the DDC protocol.

Sanford

Futureknows, 2019-12-18:
The intermittent thing is certainly when I am seeing. It works on first boot then at some point just stops working. I ran **ddcutil environment --verbose --trace env** on a fresh booth (with DDC working). The output follows:

[user@ASRock ~]$ ddcutil environment --verbose --trace env
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        ENV
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

The following tests probe the runtime environment using multiple overlapping methods.
   Current time (local): 2019-12-17 17:37:48 PST
   Current time (UTC):   2019-12-18 01:37:48
   Seconds since boot:   62

*** Basic System Information ***

ddcutil version: 0.9.8

/proc/version:
   Linux version 5.3.15-300.fc31.x86_64 (mockbuild@bkernel03.phx2.fedoraproject.org) (gcc version 9.2.1 20190827 (Red Hat 9.2.1-1) (GCC)) #1 SMP Thu Dec 5 15:04:01 UTC 2019

Architecture:     x86_64
Distributor id:   Fedora
Release:          31
Found a known architecture

/proc/cmdline:
   BOOT_IMAGE=(hd0,msdos3)/vmlinuz-5.3.15-300.fc31.x86_64 root=/dev/mapper/fedora-root ro rd.lvm.lv=fedora/root rd.luks.uuid=luks-efd71399-44c9-4ec7-95f5-6aaff92d9cd7 rd.lvm.lv=fedora/swap rhgb quiet radeon.cik_support=0 radeon.si_support=0 amdgpu.cik_support=1 amdgpu.si_support=1 amdgpu.dpm=1 amdgpu.dc=1 amdgpu.audio=1 CONFIG_SND_DYNAMIC_MINORS=y zswap.enabled=1 zswap.max_pool_percent=25 zswap.compressor=lz4 zswap.zpool=z3fold usbcore.autosuspend=-1

Compiler information:
   C standard: 201710
   gcc compatible compiler:
      Compiler version: 9.2.1

Processor information as reported by lscpu:
   Architecture:                    x86_64
   CPU op-mode(s):                  32-bit, 64-bit
   Byte Order:                      Little Endian
   Address sizes:                   43 bits physical, 48 bits virtual
   CPU(s):                          24
   On-line CPU(s) list:             0-23
   Thread(s) per core:              2
   Core(s) per socket:              12
   Socket(s):                       1
   NUMA node(s):                    1
   Vendor ID:                       AuthenticAMD
   CPU family:                      23
   Model:                           113
   Model name:                      AMD Ryzen 9 3900X 12-Core Processor
   Stepping:                        0
   Frequency boost:                 enabled
   CPU MHz:                         2196.087
   CPU max MHz:                     3800.0000
   CPU min MHz:                     2200.0000
   BogoMIPS:                        7585.87
   Virtualization:                  AMD-V
   L1d cache:                       384 KiB
   L1i cache:                       384 KiB
   L2 cache:                        6 MiB
   L3 cache:                        64 MiB
   NUMA node0 CPU(s):               0-23
   Vulnerability Itlb multihit:     Not affected
   Vulnerability L1tf:              Not affected
   Vulnerability Mds:               Not affected
   Vulnerability Meltdown:          Not affected
   Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disabled via prctl and seccomp
   Vulnerability Spectre v1:        Mitigation; usercopy/swapgs barriers and __user pointer sanitization
   Vulnerability Spectre v2:        Mitigation; Full AMD retpoline, IBPB conditional, STIBP conditional, RSB filling
[... 1491 lines cut ...]
   Drivers detected:              amdgpu
   /dev/i2c device numbers:       0 1 2 3 4 5 6 7 8 9 10 11
   sysfs_i2c_devices_exist:       true
   /sys/bus/i2c device numbers:   0 1 2 3 4 5 6 7 8 9 10 11
   dev_i2c_devices_required:      true
   module_i2c_dev_needed:         true
   module_i2c_dev_builtin:        false
   loadable_i2c_dev_exists:       true
   i2c_dev_loaded_or_builtin:     true
   group_i2c_checked:             true
   group_i2c_exists:              true
   dev_i2c_common_group_name:     i2c
   all_dev_i2c_has_group_i2c:     true
   any_dev_i2c_has_group_i2c:     true
   all_dev_i2c_is_group_rw:       true
   any_dev_i2c_is_group_rw:       true
   cur_uname:                     user
   cur_uid:                       1000
   cur_user_in_group_i2c:         true
   cur_user_any_devi2c_rw:        true
   cur_user_all_devi2c_rw:        true

Configuration suggestions:
   Current user has RW access to all /dev/i2c-N devices.
   Skipping further group and permission checks.

rockowitz, 2019-12-18:
In an earlier reply I wrote:  "Is it the case that ddcutil reads the 
EDID, then decides that the display doesn't support DDC and reports it 
as invalid, or that even the EDID read fails and ddcutil doesn't see the 
monitor at all?" What I neglected to say is that this is the behavior to 
be seen on the "ddcutil detect" command.  Even better, use "ddc detect 
--verbose".

The only thing I see of interest in your environment command output is 
that ddccontrol is running.  It's conceivable that there's some 
interference .  Lastly, on the chance that it's a timing problem, you 
might try swapping in another monitor.

Sanford

Futureknows, 2019-12-18:
Hi Sanford, thanks for the help! Some qbits are incoming! I suspect it's not a flaky monitor as it happens on two (NEC and Asus), and it didn't happen with either monitor on my other Intel CPU/GPU (same OS drive). I removed DDccontrol, it is quite possible they are conflicting. I will also try the -sleep-multiplier argument, it makes good sense as I have ddcutil bound to keyboard shortcuts that could rapidly repeat and overrun the buffer. New command is working for now.

`ddcutil setvcp 10 + 10 --sleep-multiplier 4`

Futureknows, 2019-12-27:
`ddcutil setvcp 10 + 10 --sleep-multiplier 4`

Has worked for over a week. I'm closing the issue thanks for this solution!

closed 2019-12-27 by Futureknows

clefru, 2020-10-22:
I am seeing the same problem with amdgpu.dc=1 but not with amdgpu.dc=0 on my 5.8.16 kernel with an RX480 hooking up a monitor over Dual Link DVI. Here is the output of ddcutil and ddccontrol https://pastebin.com/cPHjVTk1. Using the --sleep-multiplier 4 argument gives me exactly the same result.

Note that proper DDC support with amdgpu.dc=1, when I hook up the monitor over DisplayPort.

rockowitz, 2020-10-26:
I have just rebooted my system that has a R9 280x card, kernel version 
5.4.0.  With amdgpu.dc=1, DVI, HDMI, and DP connected displays all 
work.   I'm afraid I have no insight into this problem, which clearly is 
an amdgpu driver issue. I'll add an entry to the the ddcutil FAQ 
<http://www.ddcutil.com/faq/>

Note that the --sleep-multiplier option is of no use here.  Simply 
reading the EDID fails.  Option --sleep-multiplier applies to the DDC 
protocol on slave address x37.  Reading the EDID on slave address x50 is 
a simple read operation with no waits.

Polynomial-C, 2022-01-20:
Hi,

just wanted to say that this is still an issue with kernel-5.15.x (5.15.16 at time of this writing) and an AMD Radeon RX590 card.

rockowitz, 2022-01-20:
@Polynomial-C Please clarify what "still an issue" means here.  Given the entire thread, it's not entirely clear.  Does ddcutil always fail, intermittently fail, fail after return from sleep?   With amdgpu.dc=0, amdgpu.dc=1.

FWIW, on my RX580, ddcutil works whether or not Display Core is enabled. 

@Futureknows If you're still using that RX580 and it's the 4GB model, I have a favour to ask.  I recently picked up a RX580 that had been used for mining. At least one video bios has been modified and I'd like to return the card to stock so I can be certain I'm seeing "normal" behaviour.  If you're familiar with the process (simple to do using amdvbflash), could you dump both the "performance" and "quiet" bioses and send them to me somehow, e.g. as email attachments to rockowitz@minsoft.com.  Thank you very much.

SmallJoker, 2022-03-17:
@rockowitz RX570. `amdgpu.dc=0` works, `amdgpu.dc=1` does not.

Specs:
 * ddcutil 1.1.0 (latest from your testing PPA)
      * *kind request to update your PPA for Ubuntu 20.04 Focal, and 22.04 when it releases*
 * RX570 Ellesmere, amdgpu driver
 * GPU BIOS 115-C940PI0-100

**Default boot options**

```
$ i2cdetect -l
i2c-3	unknown   	AMDGPU DM i2c hw bus 1          	N/A
i2c-1	unknown   	AMDGPU DM i2c hw bus 0          	N/A
i2c-4	unknown   	AMDGPU DM i2c hw bus 2          	N/A
i2c-2	unknown   	dmdc                            	N/A
i2c-0	unknown   	SMBus I801 adapter at f000      	N/A
i2c-5	unknown   	AMDGPU DM i2c hw bus 3          	N/A

$ sudo ddcutil detect
Invalid display
   I2C bus:  /dev/i2c-5
   EDID synopsis:
      Mfg id:               ACR
      Model:                V243H
      Product code:         163
      Serial number:        [REMOVED]
      Binary serial number: [REMOVED] ([REMOVED])
      Manufacture year:     2009,  Week: [REMOVED])
   DDC communication failed

$ lspci -vv | grep Elles -A 20
01:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Ellesmere [Radeon RX 470/480/570/570X/580/580X/590] (rev ef) (prog-if 00 [VGA controller])
	Subsystem: ASUSTeK Computer Inc. Ellesmere [Radeon RX 470/480/570/570X/580/580X/590]
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	...
	Capabilities: <access denied>
	Kernel driver in use: amdgpu
	Kernel modules: amdgpu
```

**With boot option amdgpu.dc=1**

```
$ dmesg | grep amdgpu.dc
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-5.4.0-86-generic root=[REMOVED] ro quiet amdgpu.dc=1
[    0.090506] Kernel command line: BOOT_IMAGE=/vmlinuz-5.4.0-86-generic root=[REMOVED] ro quiet amdgpu.dc=1

$ sudo ddcutil detect
Invalid display
   ...
   DDC communication failed
```

**With boot option amdgpu.dc=0**

```
$ dmesg | grep amdgpu.dc
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-5.4.0-86-generic root=[REMOVED] ro quiet amdgpu.dc=0
[    0.090379] Kernel command line: BOOT_IMAGE=/vmlinuz-5.4.0-86-generic root=[REMOVED] ro quiet amdgpu.dc=0
$ sudo ddcutil detect
Display 1
   I2C bus:  /dev/i2c-6
   EDID synopsis:
      Mfg id:               ACR
      Model:                V243H
      Product code:         163
      Serial number:        [REMOVED]
      Binary serial number: [REMOVED] ([REMOVED])
      Manufacture year:     2009,  Week: [REMOVED])
   VCP version:         2.0

$ sudo ddcutil setvcp 0x10 10
```

^ this works!