rockowitz/ddcutil #101: Failed building ddcutil: ddc_services.c:(.text+0x20a): undefined reference to `init_usb_display's'

lwilletts, 2019-12-31:
Hi, I believe I am missing a package to fully build ddcutil, but having a look at the building from source page shows me that I have all the requisite packages installed on my Arch system. I haven't seen anyone else report this problem so it's probably something I'm doing, thanks.

rockowitz, 2019-12-31:
First, are you building from source or using the Arch package?

The configure command should complain about missing packages. Keep in 
mind that if the "missing" package is XXX, what is probably needed is 
actually XXX-dev  or XXX-devel, depending on distribution.  (Such is the 
behavior of pkg-config.)

If there's a required package that configure is not checking for, then 
compilation should reveal the missing header file, which in turn implies 
the package.  If the solution is not apparent, send me the compiler 
error messages.

In any event, please report the missing package, once the solution is found.

Finally, a wild stab in the dark.  The Debian folks recently pointed out 
that package hwdata needs to be specifically required on that distribution.

Sanford

lwilletts, 2019-12-31:
I'm using the ddcutil-git package, which is a source build. The build doesn't fail at configure, so it's happy enough that the build can proceed. Looking over the compilation I can't see anything obvious, here's the full build log: http://ix.io/261G

hwdata doesn't seem to be a package on arch, so I don't know what it provides on debian.

rockowitz, 2019-12-31:
The problem occurs at link time because usb support is disabled (so usb 
related files are not compiled), yet there's a reference to a function 
in one of those files.  This may be related to the configure errors, 
which is surprising since that portion of the configure script has been 
untouched for years.  Perhaps is something particular about Arch.   What 
command shell are you using?  Is it something other than bash?  Please 
send me the generated configure script.

As a workaround, you might try executing "configure --enable-usb" to 
explicitly enable usb support.  This may get you past the problem while 
I figure out what is going on.

Package hwdata supplies files usb.ids, pci.ids and pnp.ids.  Any normal 
Linux distribution will install these files.  They're peculiar in that 
they are provided by multiple packages, so there can be multiple copies 
installed in different directories.

rockowitz, 2019-12-31:
A fix for the link failure has been pushed to branch 0.9.9-dev.

I still would like to see the generated configure file to try to 
understand why we're getting the configure errors.

lwilletts, 2020-01-03:
Can confirm the commit in 0.9.9-dev has fixed the issue for me. Here's the configure file that gets generated: http://ix.io/26kw

rockowitz, 2020-01-03:
What command shell are you using?

lwilletts, 2020-01-04:
zsh

rockowitz, 2020-01-04:
Thanks.  Looks like it's a bashism in the configure script.  (Using "==" 
as a synonym for "=".)  I'll fix.

rockowitz, 2020-01-04:
I've pushed an updated configure.ac file to branch 0.9.9-dev.  This 
version replaces use of "==" for string equality testing with "=".  The 
double equal sign is a bash alias for "=", and is not necessarily 
present on other shells. (In zsh it's definition is controlled by the 
setopt option EQUALS.)

Please run autoconf on the updated configure.ac, run the generated 
configure file (the error messages should be gone), and send me the 
output so that I can check.  Thank you.

- Sanford

lwilletts, 2020-01-05:
Hi, here's the new generated configure output when run: http://ix.io/26v0

Can confirm no issues building.

rockowitz, 2020-01-06:
I'm going to close this issue as the two problems both appear to be 
resolved.

Regards,
Sanford

closed 2020-01-08 by lwilletts