rockowitz/ddcutil #102: "Out of Range" after "ddcutil setvcp 60 x0f" on LG32UK650 @144Hz manufacturer unrecognized OS/setting issue?

solonlost, 2020-01-01:
I'm running win8.1 as a VM with hardware passthrough on a Ubuntu host.

I have an interesting issue where the command 

`ddcutil setvcp 60 x0f`

works as expected but the monitor starts reporting 'out of range' every few seconds, in the middle of the screen. When I operate the hardware button on the monitor, it still reports the input as HDMI2, even though the signal has been correctly switched to the DisplayPort from the passthrough card.

I expect that this is an issue stemming from the manufacturer software in the monitor. I'm wondering if anyone else has had this issue on an LG monitor and if there is perhaps a setting among the 'unrecognized features' that 'ddcutil capabalities' reports that might solve this? I have unrecognized features with options nrs F5, F6, 15, F7, F8, FD and FE. I'm hoping one of the options there can get rid of the error. It's the last step to getting full software based switching between the VM and the host with accompanying monitor configs.

It's worth noting that using clickmonitorDDC on the VM to switch it back to HDMI2 @60hz works fine, though in that case the hardware switch also continues reporting the input as DisplayPort if that's what it was manually set to.

In short, I'm hoping there's some more knowledge floating around about what the unrecognized manufacturer settings do, I'm concerned about the consequences of doing it by trial and error.

rockowitz, 2020-01-02:
Some comments.

The LG Ultrawide monitors seem to be especially problematic, 
particularly with regards to VCP feature x60, though I've not seen your 
particular report.

I have seen monitors where the value reported by DDC differs from the 
value shown in the on screen display.   In fact, I have one in front of 
me right now, a Dell 3011. Setting video gain, e.g. x18 Green, using 
setvcp changes the color of the monitor, and the correct value is shown 
in the OSD.  However, getvcp always returns the maximum value 100 (x64).

If you're feeling adventurous, you can narrow down the possible features 
to only the non-continuous ones.  getvcp for continuous features will 
report a large maximum value (e.g. 100), while getvcp for non-continuous 
features will show a small maximum value, typically less than x10.

I'd appreciate it if you could run "ddcutil interrogate" and send the 
output as an attachment.  This will enable me to add an entry for your 
monitor in the rogues gallery at www.ddcutil.com/monitor_notes.

- Sanford

solonlost, 2020-01-05:
Ok thanks for the infor: 

Here's the output of 'ddcutil interrogate':
[ddcint.txt](https://github.com/rockowitz/ddcutil/files/4023365/ddcint.txt)

The system has 4 monitors attached, this issue relates to the 2560x1440 recognized as display 1.

I've had the thought that if I leave the hardware setting on the monitor on DisplayPort, autoboot the VM and use clickmonitorDDC to switch it to HDMI2, it may function without the annoying out or range message. I'll post here if that workaround solves the problem.

rockowitz, 2020-01-18:
Can I ask you to run "ddcutil interrogate --verbose" from the latest 
0.9.9-dev branch and send the output as an attachment?  I've made two 
changes to address problems problems in your environment.

1) LG displays do not include either a binary or ascii serial number in 
the EDID.  This confused interrogate's display identifier 
cross-reference, which relied on every display having a unique EDID.

2) Better identify displays that indicate an unsupported feature in a 
getvcp reply by setting MH=ML=SH=SL=0 instead of setting the Unsupported 
VCP Code bit.

Thanks,
Sanford

atobenzo, 2020-12-24:
Hello,

I have the same situation as the original poster, but with a LG27UL850 monitor and Windows 10. I have attached the output from "ddcutil interrogate --verbose" from the master branch in the hope that it helps fix this issue.

[interrogate-output.txt](https://github.com/rockowitz/ddcutil/files/5740031/interrogate-output.txt)

Also, like the original poster if I switch to my VM's output using the monitor's interface, and then switch back using (in my case) Control My Monitor, I can switch back and forth using ddcutil and the Windows program without the "Out of Range" message. The only problems then are (1) I still need to make the change once using the monitor button and (2) when the VM shuts down I get an "Out of Range" in my Linux output.

Thanks!
Chris

rockowitz, 2020-12-26:
When you say you get an "Out of Range" message in your Linux output, which Linux component is generating the message?  Does it appear in any of your log files or in dmesg output? Or is this message generated by the monitor's firmware?

atobenzo, 2020-12-26:
Sorry, yes, by Linux output I just mean the monitor's on screen display while switched to the output that my Linux host graphics card is using.

ChinmayTyagi, 2021-08-20:
Hi, any progress with this issue? Facing the same thing myself.

Diovanti, 2022-01-08:
LG 32GK650F Monitor
Linux Mint 20.2 host, Windows 10 LTSC VM (PCI Passthrough)
ddcutil version 1.2.1 from source

I have the exact same problem myself. Any Ideas?

Also ddccontrol does the same exact thing as well “Out Of Range”

[interrogate.txt](https://github.com/rockowitz/ddcutil/files/7832638/interrogate.txt)

rockowitz, 2022-01-08:
Since you've taken the trouble to include your **interrogate** output, I do want to acknowledge your comment.  Unfortunately, not having a recent problematic LG monitor myself there's not much I can add re these monitors.

However, I've recently added a [page](https://www.ddcutil.com/sniffing) on the [website](https://www.ddcutil.com) describing some relatively inexpensive hardware that can be used sniff the I2C bus and reverse engineer manufacturers' extensions to the DDC/CI protocol.

closed 2025-02-09 by rockowitz