rockowitz/ddcutil #104: Unable to detect Cintiq CCH (A1300) despite parse-edid returning information about it

InconsolableCellist, 2020-02-01:
# Bug description 

I have a Cintiq A1300 device connected over HDMI, and it appears to be detected on the i2c bus by get-edid:

```
$ sudo get-edid -b 2 | parse-edid
2
This is read-edid version 3.0.2. Prepare for some fun.
Attempting to use i2c interface
Only trying 2 as per your request.
256-byte EDID successfully retrieved from i2c bus 2
Looks like i2c was successful. Have a good day.
Checksum Correct

Section "Monitor"
        Identifier "Cintiq A1300"
        ModelName "Cintiq A1300"
        VendorName "WAC"
        # Monitor Manufactured week 37 of 2013
        # EDID version 1.3
        # Digital Display
        DisplaySize 290 170
        Gamma 2.20
        Option "DPMS" "true"
        Horizsync 31-70
        VertRefresh 59-61
        # Maximum pixel clock is 150MHz
        #Not giving standard mode: 1280x1024, 60Hz
        #Not giving standard mode: 1280x960, 60Hz
        #Not giving standard mode: 1440x900, 60Hz
        #Not giving standard mode: 1680x1050, 60Hz

        #Extension block found. Parsing...
        Modeline        "Mode 1" 148.500 1920 2008 2052 2200 1080 1084 1089 1125 +hsync +vsync
        Modeline        "Mode 0" +hsync +vsync 
        Modeline        "Mode 2" 74.250 1280 1390 1420 1650 720 725 730 750 +hsync +vsync
        Modeline        "Mode 3" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
        Modeline        "Mode 4" 27.027 720 736 798 858 480 489 495 525 -hsync -vsync
        Modeline        "Mode 5" 25.200 640 656 752 800 480 490 492 525 -hsync -vsync
        Modeline        "Mode 6" +hsync +vsync 
        Modeline        "Mode 7" +hsync +vsync 
        Modeline        "Mode 8" +hsync +vsync 
        Modeline        "Mode 9" -hsync +vsync 
        Option "PreferredMode" "Mode 1"
EndSection
```

However, ddcutil won't detect it: 

```
$ sudo ddcutil detect
Failure getting EDID for /dev/i2c-1: status code=DDCRC_INVALID_EDID(-3010): unable to parse EDID
Failure getting EDID for /dev/i2c-2: status code=DDCRC_INVALID_EDID(-3010): unable to parse EDID
Display 1
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           SAM
      Model:            U28E590
      Serial number:    HTPK416852
      Manufacture year: 2018
      EDID version:     1.4
   VCP version:         2.0
```

I pulled the latest code from git and compiled with all the options enabled (http://www.ddcutil.com/building/) and see similar behavior:

```
$ cd ~/Programs/ddcutil/src
$ sudo ./ddcutil --version
ddcutil 0.9.8
Built without support for AMD Display Library (AMD proprietary driver).
Built with support for USB connected displays.
Built without function failure simulation.
Built without libdrm services.

Copyright (C) 2015-2018 Sanford Rockowitz
License GPLv2: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

$ sudo ./ddcutil detect -v
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Display 1
   I2C bus:             /dev/i2c-3
      I2C address 0x50 (EDID) present: true 
      Is eDP device:                   false
      /sys/bus/i2c/devices/i2c-3/name: NVIDIA i2c adapter 7 at 1:00.0
   EDID synopsis:
      Mfg id:           SAM
      Model:            U28E590
      Serial number:    HTPK416852
      Manufacture year: 2018
      EDID version:     1.4
      Product code:     3149
      Extra descriptor: 
      Video input definition: 0xb5 - Digital Input (DisplayPort)
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:2:2
         Standard sRGB color space: True
      White x,y:        0.312, 0.329
      Red   x,y:        0.634, 0.341
      Green x,y:        0.312, 0.636
      Blue  x,y:        0.158, 0.062
      Extension blocks: 1
   EDID source: 
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 4c 2d 4d 0c 4a 53 4d 30   ........L-M.JSM0
      +0010   11 1c 01 04 b5 3d 23 78 3a 5f b1 a2 57 4f a2 28   .....=#x:_..WO.(
      +0020   0f 50 54 bf ef 80 71 4f 81 00 81 c0 81 80 a9 c0   .PT...qO........
      +0030   b3 00 95 00 01 01 4d d0 00 a0 f0 70 3e 80 30 20   ......M....p>.0 
      +0040   35 00 5f 59 21 00 00 1a 00 00 00 fd 00 38 4b 1e   5._Y!........8K.
      +0050   87 3c 00 0a 20 20 20 20 20 20 00 00 00 fc 00 55   .<..      .....U
      +0060   32 38 45 35 39 30 0a 20 20 20 20 20 00 00 00 ff   28E590.     ....
      +0070   00 48 54 50 4b 34 31 36 38 35 32 0a 20 20 01 67   .HTPK416852.  .g
   VCP version:         2.0
   Controller mfg:      Mstar
   Firmware version:    0.1
   Monitor returns DDC Null Response for unsupported features: false
```

But still nothing from the other busses

If I try to force the issue by attempting to set the brightness (presumably), it also says no monitor is detected:

```
$ sudo ./ddcutil -b 2 setvcp 10 75 --noverify --nodetect
No monitor detected on I2C bus /dev/i2c-2
```

# System information 

```
$ uname -r 
4.19.0-6-amd64
```

```
$ cat /etc/os-release 
PRETTY_NAME="Devuan GNU/Linux 3 (beowulf)"
NAME="Devuan GNU/Linux"
VERSION_ID="3"
VERSION="3 (beowulf)"
VERSION_CODENAME=beowulf
ID=debian
ID_LIKE=debian
HOME_URL="https://www.devuan.org/"
SUPPORT_URL="https://devuan.org/os/community"
BUG_REPORT_URL="https://bugs.devuan.org/"
```

```
$ sudo modinfo i2c-dev 
filename:       /lib/modules/4.19.0-6-amd64/kernel/drivers/i2c/i2c-dev.ko
license:        GPL
description:    I2C /dev entries driver
author:         Frodo Looijaard <frodol@dds.nl> and Simon G. Vogl <simon@tk.uni-linz.ac.at>
depends:        
retpoline:      Y
intree:         Y
name:           i2c_dev
vermagic:       4.19.0-6-amd64 SMP mod_unload modversions 
sig_id:         PKCS#7
signer:         Debian Secure Boot CA
sig_key:        A7:46:8D:EF
sig_hashalgo:   sha256
signature:      C9:E2:E7:86:A6:8C:9B:8C:C4:18:55:A9:14:13:31:CE:1A:1C:4D:B1:
                88:66:F2:31:71:68:40:DA:5F:3A:D1:45:A6:B5:A7:34:48:05:B8:8D:
                58:EC:B5:32:4B:12:4E:1A:35:D5:95:1E:40:44:F6:16:3E:03:6F:2B:
                61:D9:B6:CE:92:5B:73:7E:CA:30:48:CE:FE:24:93:DF:0F:29:0C:52:
                15:CC:72:72:3F:C9:13:79:58:03:25:DA:8A:F1:F0:5E:88:88:4F:EE:
                76:A3:4F:4B:AA:F5:21:D3:D0:D5:9B:39:2A:FE:33:AC:F0:D8:38:81:
                98:4E:A5:93:60:F2:CE:C1:AB:D4:72:D5:8D:0B:FF:FE:CB:03:02:41:
                93:0C:C0:0F:10:E7:07:4F:18:2A:77:18:EC:40:EE:9F:FF:7D:B7:40:
                74:4C:95:A1:9F:8F:6A:DE:FA:8E:EF:3F:CB:BA:9A:69:61:E6:1C:79:
                2B:D8:91:BB:E0:91:61:0C:5B:4C:6A:BC:92:F3:31:7A:7A:4E:EF:43:
                A4:20:2D:88:6F:75:0E:93:EE:B3:39:E9:4D:58:36:C8:1F:33:51:90:
                84:37:56:BB:C6:6C:0C:11:9C:09:B9:2F:FA:76:AF:DD:15:A0:F6:A7:
                D7:43:28:25:0F:03:0D:93:54:8C:A4:DF:7A:D3:94:C5
```

----

Do you know what's going wrong? Thanks

InconsolableCellist, 2020-02-01:
Some more info, tracing the I2C-related group:

```
$ sudo ./ddcutil -b 2 setvcp 10 75 --trace=I2C
(detect_single_bus             ) Starting.  busno = 2
(i2c_check_bus                 ) Starting. busno=2, buf_info=0x560666a6a630
(is_edp_device                 ) busno=2, returning: false
(i2c_open_bus                  ) Starting. busno=2, callopts=0x80
(i2c_open_bus                  ) Returning file descriptor: 3
(i2c_get_parsed_edid_by_fd     ) Starting. fd=3
(i2c_get_raw_edid_by_fd        ) Getting EDID for file 3
(i2c_set_addr                  ) fd=3, addr=0x50, i2c_force_slave_addr_flag=false, callopts=CALLOPT_ERR_MSG
(i2c_set_addr                  ) addr = 0x50. Returning OK(0): success
(invoke_i2c_writer             ) writer=read_writer, bytes_to_write=00
(invoke_i2c_writer             ) Returning rc=OK(0): success
(invoke_i2c_reader             ) reader=read_reader, bytect=128
(invoke_i2c_reader             ) Bytes read: 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff
(invoke_i2c_reader             ) Returning rc=OK(0): success
(i2c_get_raw_edid_by_fd        ) Returning OK(0): success.  edidbuf contents:
Buffer at 0x560666a6a710,  bytes addr=0x560666a6b1c0, len=128, max_size=128
        +0          +4          +8          +c            0   4   8   c   
+0000   00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff   ................
+0010   00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff   ................
+0020   00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff   ................
+0030   00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff   ................
+0040   00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff   ................
+0050   00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff   ................
+0060   00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff   ................
+0070   00 ff ff ff 00 ff ff ff 00 ff ff ff 00 ff ff ff   ................
(i2c_get_parsed_edid_by_fd     ) Returning (nil)
(i2c_close_bus                 ) Starting. fd=3, busno=2, callopts=CALLOPT_ERR_MSG
(i2c_close_bus                 ) Returning: 0
(i2c_check_bus                 ) Done. flags=0x0cc1, bus info:
      Bus /dev/i2c-2 found:   true
      Bus /dev/i2c-2 probed:  true
      Bus accessible:          true
      Bus is eDP:              false
      Valid bus name checked:  true
      I2C bus has valid name:  true
      Address 0x50 present:    false
(detect_single_bus             ) Done.  busno=2, returning: 0x560666a6a630
No monitor detected on I2C bus /dev/i2c-2
```

InconsolableCellist, 2020-02-01:
For some comparison with get-edid's code:

`read-edid-3.0.2/get-edid/i2c.c` accesses the data off the bus with `i2c_smbus_read_byte_data`, and it's definitely getting back the EDID signature of 00,ff,ff,ff,ff,ff,ff,00, as verified on line 122 (the if statement):

```
		if (i2cfile >=0) {//no matter how many times, >=0 still looks really angry.
			for (j=0;j<256;j++)
				block[j] = i2c_smbus_read_byte_data(i2cfile, j);
		}
		close(i2cfile);
		if (block[0]==0x00&&block[7]==0x00&&block[1]==0xff&&block[2]==0xff&&block[3]==0xff&&block[4]==0xff&&block[5]==0xff&&block[6]==0xff) {
			ret = 0;
			break;
		}
```

 ddcutil is also definitely not seeing that EDID signature, as it gets just a bunch of 00,ff,ff,ff,00 data.

You can see the `j` parameter passed into that read command; that's being passed in an `i2c_smbus_ioctl_data` args struct to `ioctl`. So it would seem that to fetch the signature it's passing 0-256, presuambly saying "give me memory location j" or possibly just "hit the I2C bus 256 times, and this parameter will be ignored."

If the difference isn't in how the I2C bus is being accessed, what could be causing the device to return bad data in ddcutil but not get-edid?

InconsolableCellist, 2020-02-01:
# More system info 

```
$ sudo ./ddcutil environment
The following tests probe the runtime environment using multiple overlapping methods.

*** Basic System Information ***

ddcutil version: 0.9.8

/proc/version:
   Linux version 4.19.0-6-amd64 (debian-kernel@lists.debian.org) (gcc version 8.3.0 (Debian 8.3.0-6)) #1 SMP Debian 4.19.67-2+deb10u2 (2019-11-11)

Architecture:     x86_64
Distributor id:   Debian
Release:          3
Found a known architecture

/proc/cmdline:
   BOOT_IMAGE=/vmlinuz-4.19.0-6-amd64 root=UUID=a549d036-d666-440c-af27-55427a23e65f ro quiet

*** Primary Check 1: Identify video card and driver ***

Obtaining card and driver information from /sys...
Primary video controller at PCI address 0000:01:00.0 (boot_vga flag is set)
   Device class:        x030000    VGA compatible controller
   Vendor:              x10de      NVIDIA Corporation
   Device:              x1c82      GP107 [GeForce GTX 1050 Ti]
   Subvendor/Subdevice: 1462/8c96  Micro-Star International Co., Ltd. [MSI]
   Driver name:         nvidia
   Driver version:      440.31
   I2C device:          i2c-3      name: NVIDIA i2c adapter 7 at 1:00.0
   I2C device:          i2c-1      name: NVIDIA i2c adapter 4 at 1:00.0
   I2C device:          i2c-2      name: NVIDIA i2c adapter 6 at 1:00.0

*** Primary Check 2: Check that /dev/i2c-* exist and writable ***

Current user: root (0)

Checking /dev/i2c-* devices...

Unless the system is using the AMD proprietary driver fglrx, devices /dev/i2c-*
must exist and the logged on user must have read/write permission for those
devices (or at least those devices associated with monitors).

Typically, this access is enabled by:
  - setting the group for /dev/i2c-* to i2c
  - setting group RW permissions for /dev/i2c-*
  - making the current user a member of group i2c

Alternatively, this can be enabled by just giving everyone RW permission
The following tests probe for these conditions.

Checking for /dev/i2c-* devices...
   crw-rw---- 1 root i2c 89, 0 Feb  1 13:24 /dev/i2c-0
   crw-rw---- 1 root i2c 89, 1 Feb  1 13:24 /dev/i2c-1
   crw-rw---- 1 root i2c 89, 2 Feb  1 13:24 /dev/i2c-2
   crw-rw---- 1 root i2c 89, 3 Feb  1 13:24 /dev/i2c-3

Current user (root) has RW access to all /dev/i2c-* devices.

*** Primary Check 3: Check that module i2c_dev is loaded ***

Checking for module i2c_dev...
   Module i2c-dev is NOT built into kernel
   Loadable i2c-dev module found
   Module i2c_dev is loaded

*** Primary Check 4: Driver specific checks ***

Performing driver specific checks...

Checking for special settings for proprietary Nvidia driver 
(Needed for some newer Nvidia cards).
   grep: /etc/X11/xorg.conf: No such file or directory

*** Additional probes ***

Scanning /proc/modules for driver environment...
   Found other loaded module:      i2c_dev
   Found other loaded module:      videobuf2_vmalloc
   Found other loaded module:      videobuf2_memops
   Found other loaded module:      videobuf2_v4l2
   Found other loaded module:      videobuf2_common
   Found other loaded module:      videodev
   Found video driver module:      nvidia
   Found other loaded module:      drm_kms_helper
   Loaded drm module depends on:   nvidia_drm,drm_kms_helper,
   Found other loaded module:      i2c_i801
   Loaded video module depends on: -

Testing if modules are loaded using /sys...
   Module amdgpu           is NOT loaded
   Module fbdev            is NOT loaded
   Module fglrx            is NOT loaded
   Module fturbo           is NOT loaded
   Module i915             is NOT loaded
   Module mgag200          is NOT loaded
   Module nvidia           is loaded
   Module nouveau          is NOT loaded
   Module radeon           is NOT loaded
   Module vboxvideo        is NOT loaded
   Module vc4              is NOT loaded
   Module drm              is loaded
   Module i2c_algo_bit     is NOT loaded
   Module i2c_dev          is loaded
   Module i2c_piix4        is NOT loaded
   Module ddcci            is NOT loaded

Examining /sys/bus/i2c/devices...
   /sys/bus/i2c/devices/i2c-3/name:   NVIDIA i2c adapter 7 at 1:00.0
   /sys/bus/i2c/devices/i2c-1/name:   NVIDIA i2c adapter 4 at 1:00.0
   /sys/bus/i2c/devices/i2c-2/name:   NVIDIA i2c adapter 6 at 1:00.0
   /sys/bus/i2c/devices/i2c-0/name:   SMBus I801 adapter at f000

Env_Accumulator:
   architecture:                  x86_64
   distributor_id                 Debian
   Drivers detected:              nvidia
   /dev/i2c device numbers:       0 1 2 3
   sysfs_i2c_devices_exist:       true
   /sys/bus/i2c device numbers:   0 1 2 3
   dev_i2c_devices_required:      true
   module_i2c_dev_needed:         true
   module_i2c_dev_builtin:        false
   loadable_i2c_dev_exists:       true
   i2c_dev_loaded_or_builtin:     true
   group_i2c_checked:             true
   group_i2c_exists:              true
   dev_i2c_common_group_name:     i2c
   all_dev_i2c_has_group_i2c:     true
   any_dev_i2c_has_group_i2c:     true
   all_dev_i2c_is_group_rw:       true
   any_dev_i2c_is_group_rw:       true
   cur_uname:                     root
   cur_uid:                       0
   cur_user_in_group_i2c:         false
   cur_user_any_devi2c_rw:        true
   cur_user_all_devi2c_rw:        true

Configuration suggestions:
   Current user has RW access to all /dev/i2c-N devices.
   Skipping further group and permission checks.
```

InconsolableCellist, 2020-02-01:
So I saw in that latest command I found, the `ddcutil environment` one, that it mentioned some parameters being required for the Nvidia driver: 

```
$ cat /etc/X11/xorg.conf.d/90-nvidia_i2c.conf 
Section "Device"
   Driver "nvidia"
   Identifier "Dev0"
   Option     "RegistryDwords"  "RMUseSwI2c=0x01; RMI2cSpeed=100"
   # solves problem of i2c errors with nvidia driver
   # per https://devtalk.nvidia.com/default/topic/572292/-solved-does-gddccontrol-work-for-anyone-here-nvidia-i2c-monitor-display-ddc/#4309293
EndSection
```

I put these in, rebooted, and now ddcutil works for my monitors!

Hopefully my steps above are useful for someone. Still not sure why `get-edid` was able to access the device.

rockowitz, 2020-02-02:
First, thank you for the thorough bug report and comparison and 
analysis.  It gives me something to work with.

Here's what I believe is going on.  The original I2C bus speed is 100 
kbit/sec (standard mode).  There's also full speed (400 kbit/sec), fast 
mode (1 mbit/sec) and high speed (3.2 mbit/sec).

Apparently, the proprietary Nvidia driver defaults to full speed (400 
kbit/sec).  Some (most?) monitors work properly at full speed.  It 
appears that your Cintiq monitor does not.  Using the RMI2cSpeed=100 
option slows the I2C bus to standard mode, allowing the Cintiq monitor 
to work properly.

Whereas ddcutil reads the entire first 128 byte block of the EDID in one 
call, read-edid uses 256 calls to read single bytes.  This apparently 
slows things down enough for your monitor to work. (Why 128 vs 256 
bytes?  The original EDID was a single 128 byte block.  This was revised 
to specify a second 128 block, and further revised to allow for multiple 
second blocks, selected by writing to slave address x30.  ddcutil is 
interested only in the original 128 bytes.)

There is another way to slow things down, by increasing the duration of 
host sleeps mandated by the DDC/CI protocol.  This can be adjusted using 
the recently added ddcutil option --sleep-multiplier.

I'd appreciate it if you'd do the following:

Disable the special nvidia driver options.

Try "ddcutil detect --sleep-multiplier 4".  Does this solve the problem? 
(Probably not, but it's worth checking.)

I've uploaded a version of ddcutil to branch 0.9.9-dev that reads the 
EDID 1 byte at a time. Try running "ddcutil detect --trcfunc 
i2c_get_raw_edid_by_fd".  (--trcfunc will produce more targeted 
diagnostics than --trace.)  Does the EDID get read?  (N. the full detect 
command will likely fail during the feature reads that are part of 
detection.)  If the EDID read succeeds, I'll look into using single byte 
reads more generally.

Regards,
Sanford

InconsolableCellist, 2020-02-02:
That sounds like a good theory with the timing issue! I've disabled the Xorg conf file and I'm first trying with the master branch (a9b4a20)

```
$ sudo ./ddcutil detect
Display 1
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           SAM
      Model:            U28E590
      Serial number:    HTPK416852
      Manufacture year: 2018
      EDID version:     1.4
   VCP version:         2.0

$ sudo ./ddcutil detect --sleep-multiplier 4
Display 1
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           SAM
      Model:            U28E590
      Serial number:    HTPK416852
      Manufacture year: 2018
      EDID version:     1.4
   VCP version:         2.0
```

I recompiled and tested with 0.9.9-dev and check this out:

```
$ sudo ./ddcutil detect --trcfunc i2c_get_raw_edid_by_fd |
(i2c_get_raw_edid_by_fd        ) Getting EDID for file 3
(i2c_get_raw_edid_by_fd        ) Sleeping for 50 milliseconds. Before write
(i2c_get_raw_edid_by_fd        ) Final single byte read returned 0, ndx=128
(i2c_get_raw_edid_by_fd        ) call_read returned:
   Buffer at 0x55dcf02b2440,  bytes addr=0x55dcf02c26e0, len=128, max_size=128
           +0          +4          +8          +c            0   4   8   c   
   +0000   00 ff ff ff ff ff ff 00 10 ac 98 a0 42 33 4e 30   ............B3N0
   +0010   03 19 01 03 80 30 1b 78 ea eb f5 a6 56 51 9c 26   .....0.x....VQ.&
   +0020   10 50 54 a5 4b 00 71 4f 81 80 a9 c0 d1 c0 01 01   .PT.K.qO........
   +0030   01 01 01 01 01 01 02 3a 80 18 71 38 2d 40 58 2c   .......:..q8-@X,
   +0040   45 00 dd 0c 11 00 00 1e 00 00 00 ff 00 4b 57 31   E............KW1
   +0050   34 56 35 31 48 30 4e 33 42 0a 00 00 00 fc 00 44   4V51H0N3B......D
   +0060   45 4c 4c 20 50 32 32 31 34 48 0a 20 00 00 00 fd   ELL P2214H. ....
   +0070   00 38 4c 1e 53 11 00 0a 20 20 20 20 20 20 00 15   .8L.S...      ..
(i2c_get_raw_edid_by_fd        ) edid checksum = 0
(i2c_get_raw_edid_by_fd        ) Returning OK(0): success.  edidbuf contents:
Buffer at 0x55dcf02b2440,  bytes addr=0x55dcf02c26e0, len=128, max_size=128
        +0          +4          +8          +c            0   4   8   c   
+0000   00 ff ff ff ff ff ff 00 10 ac 98 a0 42 33 4e 30   ............B3N0
+0010   03 19 01 03 80 30 1b 78 ea eb f5 a6 56 51 9c 26   .....0.x....VQ.&
+0020   10 50 54 a5 4b 00 71 4f 81 80 a9 c0 d1 c0 01 01   .PT.K.qO........
+0030   01 01 01 01 01 01 02 3a 80 18 71 38 2d 40 58 2c   .......:..q8-@X,
+0040   45 00 dd 0c 11 00 00 1e 00 00 00 ff 00 4b 57 31   E............KW1
+0050   34 56 35 31 48 30 4e 33 42 0a 00 00 00 fc 00 44   4V51H0N3B......D
+0060   45 4c 4c 20 50 32 32 31 34 48 0a 20 00 00 00 fd   ELL P2214H. ....
+0070   00 38 4c 1e 53 11 00 0a 20 20 20 20 20 20 00 15   .8L.S...      ..
(i2c_get_raw_edid_by_fd        ) Getting EDID for file 3
(i2c_get_raw_edid_by_fd        ) Sleeping for 50 milliseconds. Before write
(i2c_get_raw_edid_by_fd        ) Final single byte read returned 0, ndx=128
(i2c_get_raw_edid_by_fd        ) call_read returned:
   Buffer at 0x55dcf02b2440,  bytes addr=0x55dcf02c3980, len=128, max_size=128
           +0          +4          +8          +c            0   4   8   c   
   +0000   00 ff ff ff ff ff ff 00 5c 23 41 10 22 00 00 00   ........\#A."...
   +0010   25 17 01 03 80 1d 11 78 aa a7 07 a6 56 54 9b 26   %......x....VT.&
   +0020   0c 50 56 21 08 00 81 80 81 40 95 00 b3 00 01 01   .PV!.....@......
   +0030   01 01 01 01 01 01 02 3a 80 18 71 38 2d 40 58 2c   .......:..q8-@X,
   +0040   45 00 25 a5 10 00 00 1e 00 00 00 fc 00 43 69 6e   E.%..........Cin
   +0050   74 69 71 20 41 31 33 30 30 0a 00 00 00 ff 00 33   tiq A1300......3
   +0060   49 41 48 30 30 30 30 33 34 0a 20 20 00 00 00 fd   IAH000034.  ....
   +0070   00 3b 3d 1f 46 0f 00 0a 20 20 20 20 20 20 01 f1   .;=.F...      ..
[... 44 lines cut ...]
      Manufacture year: 2015
      EDID version:     1.3
   DDC communication failed

Invalid display
   I2C bus:             /dev/i2c-2
   EDID synopsis:
      Mfg id:           WAC
      Model:            Cintiq A1300
      Serial number:    3IAH000034
      Manufacture year: 2013
      EDID version:     1.3
   DDC communication failed

Display 1
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           SAM
      Model:            U28E590
      Serial number:    HTPK416852
      Manufacture year: 2018
      EDID version:     1.4
   VCP version:         2.0

```

InconsolableCellist, 2020-02-02:
I should note too that the same situation was occurring with that third monitor, the DELL P2214H, which is an old 1080p monitor, from 2015 (the Cintiq 2013). I suppose the single-byte read is more compatible. I'm glad I was able to help diagnose this! I guess you have to put more of those `#ifdef SINGLE_BYTE_READ` blocks around, to enable all the functionality? Or maybe it can be a runtime arg, something like `--compatibility`. Or maybe just on by default?

rockowitz, 2020-02-03:
> I should note too that the same situation was occurring with that
> third monitor, the DELL P2214H, which is an old 1080p monitor, from
> 2015 (the Cintiq 2013). I suppose the single-byte read is [...]

Once the displays are detected, do getvcp and setvcp work? How about 
capabilities, which requires multiple write/read cycles?

I have a Dell P2411h which often fails communication with DDC data 
errors.  The --sleep-multiplier option resolves the problem, but it will 
be interesting to see if single byte reads do as well.  Can you send me 
the "ddcutil probe" output for this and the Cintiq monitor? It will be 
interesting to see if they are using the same controller chip as my P2411h.

The iftested code in i2c_get_raw_edid_by_fd() is there for quick and 
dirty testing.  Now I have to figure out whether to always use single 
byte reads, fall back to single byte reads if a multi-byte read fails, 
or as you suggest use a command line option.  I'd prefer to avoid the 
latter, as it requires people to dig into the documentation.   Given 
that 90% of ddcutil's time is spend in DDC/CI protocol mandated sleeps 
(unless shortened by a --sleep multiplier value less than 1 - I actually 
have been able to get it down to as low as .2 for some monitors), single 
byte reads shouldn't have a perceptible affect on elapsed execution 
time.  This needs to be measured.  Harder to check for will the question 
of whether there are situations where multi-byte reads work but multiple 
single byte reads do not.

Thank you for bringing this case to my attention and helping to diagnose 
it.  I'm hoping the changes will significantly reduce the number of 
problematic monitors.

Changing topics, your github userid suggests that you're a cellist. Is 
that the case?  If so, you may be amused to know that I manufacture 
Rockstops.  It's a small old family business that I maintain on the side.

Regards,
Sanford

InconsolableCellist, 2020-02-03:
>Once the displays are detected, do getvcp and setvcp work? How about
capabilities, which requires multiple write/read cycles?

It appears that it doesn't:

```
$ sudo ./ddcutil -b 2 getvcp 12
DDC communication failed for monitor on I2C bus /dev/i2c-2

$ sudo ./ddcutil -b 2 setvcp 12 50
DDC communication failed for monitor on I2C bus /dev/i2c-2
```

That's on both the Cintiq and Dell monitors. Probe also fails:

```
$ sudo ./ddcutil -b 2 probe
DDC communication failed for monitor on I2C bus /dev/i2c-2
```

`-detect` still lists the monitors though, same as above, but note that it says "DDC communication failed," both before and now:

```
$ sudo ./ddcutil detect 
Invalid display
   I2C bus:             /dev/i2c-1
   EDID synopsis:
      Mfg id:           DEL
      Model:            DELL P2214H
      Serial number:    KW14V51H0N3B
      Manufacture year: 2015
      EDID version:     1.3
   DDC communication failed

Invalid display
   I2C bus:             /dev/i2c-2
   EDID synopsis:
      Mfg id:           WAC
      Model:            Cintiq A1300
      Serial number:    3IAH000034
      Manufacture year: 2013
      EDID version:     1.3
   DDC communication failed

Display 1
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           SAM
      Model:            U28E590
      Serial number:    HTPK416852
      Manufacture year: 2018
      EDID version:     1.4
   VCP version:         2.0
```

> Changing topics, your github userid suggests that you're a cellist. Is that the case?  If so, you may be amused to know that I manufacture Rockstops.  It's a small old family business that I maintain on the side.

That's funny! Though I'm not actually a cellist, I just made a random name. That's cool though

rockowitz, 2020-02-03:
Your results are consistent with the fact that the single byte read 
change was made only to the EDID reading on the I2C bus, not to DDC 
communication. So the change needs to be more generally applied. Thanks.

rockowitz, 2020-02-09:
I've added code to further parameterize how I2C writes and reads are performed.  The latest version of ddcutil on github contains some special code that exercises these variations (16 permutations) if the command "ddcutil detect --i1 1" is issued.  I'd appreciate it if you could run this command both with and without the 90-nvidia_i2c.conf settings and send me the output. 

Running this command for several monitors here, what I observe is that all monitors return the EDID successfully no matter which variation is chosen.  Some monitors work with single byte DDC reads but others do not.  All monitors work with multi-byte DDC reads. It will be interesting to see how your system behaves. 

If anyone else besides InconsolableCellist runs the "ddcutil detect --i1 1" command I'd appreciate seeing the output.  Also, let me know your video card and driver (or better yet send the output of "ddcutil environment --verbose".  Please send the output as attachments if possible. 

--Sanford

InconsolableCellist, 2020-02-09:
Sure. Here's the run without the 90-nvidia_i2c.conf:

```
===========> IO STRATEGY 1:
   i2c_io_strategy:          FILEIO
   i2c_read_bytewise:        false
   EDID read uses I2C layer: false
   EDID read bytewise:       false

Display 1
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           SAM
      Model:            U28E590
      Serial number:    HTPK416852
      Manufacture year: 2018
      EDID version:     1.4
   VCP version:         2.0

===========> IO STRATEGY 2:
   i2c_io_strategy:          FILEIO
   i2c_read_bytewise:        false
   EDID read uses I2C layer: false
   EDID read bytewise:       true

Invalid display
   I2C bus:             /dev/i2c-1
   EDID synopsis:
      Mfg id:           DEL
      Model:            DELL P2214H
      Serial number:    KW14V51H0N3B
      Manufacture year: 2015
      EDID version:     1.3
   DDC communication failed

Invalid display
   I2C bus:             /dev/i2c-2
   EDID synopsis:
      Mfg id:           WAC
      Model:            Cintiq A1300
      Serial number:    3IAH000034
      Manufacture year: 2013
      EDID version:     1.3
   DDC communication failed

Display 2
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           SAM
      Model:            U28E590
      Serial number:    HTPK416852
      Manufacture year: 2018
      EDID version:     1.4
   VCP version:         2.0

===========> IO STRATEGY 3:
   i2c_io_strategy:          FILEIO
   i2c_read_bytewise:        false
   EDID read uses I2C layer: true
   EDID read bytewise:       false

Display 3
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           SAM
      Model:            U28E590
      Serial number:    HTPK416852
      Manufacture year: 2018
      EDID version:     1.4
   VCP version:         2.0
[... 324 lines cut ...]
      Serial number:    KW14V51H0N3B
      Manufacture year: 2015
      EDID version:     1.3
   VCP version:         2.1

Display 23
   I2C bus:             /dev/i2c-2
   EDID synopsis:
      Mfg id:           WAC
      Model:            Cintiq A1300
      Serial number:    3IAH000034
      Manufacture year: 2013
      EDID version:     1.3
   VCP version:         2.1

Display 24
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           SAM
      Model:            U28E590
      Serial number:    HTPK416852
      Manufacture year: 2018
      EDID version:     1.4
   VCP version:         2.0
```

InconsolableCellist, 2020-02-09:
Here it is after restoring the nvidia conf and rebooting: 

```

===========> IO STRATEGY 1:
   i2c_io_strategy:          FILEIO
   i2c_read_bytewise:        false
   EDID read uses I2C layer: false
   EDID read bytewise:       false

Display 1
   I2C bus:             /dev/i2c-1
   EDID synopsis:
      Mfg id:           DEL
      Model:            DELL P2214H
      Serial number:    KW14V51H0N3B
      Manufacture year: 2015
      EDID version:     1.3
   VCP version:         2.1

Display 2
   I2C bus:             /dev/i2c-2
   EDID synopsis:
      Mfg id:           WAC
      Model:            Cintiq A1300
      Serial number:    3IAH000034
      Manufacture year: 2013
      EDID version:     1.3
   VCP version:         2.1

Display 3
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           SAM
      Model:            U28E590
      Serial number:    HTPK416852
      Manufacture year: 2018
      EDID version:     1.4
   VCP version:         2.0

===========> IO STRATEGY 2:
   i2c_io_strategy:          FILEIO
   i2c_read_bytewise:        false
   EDID read uses I2C layer: false
   EDID read bytewise:       true

Display 4
   I2C bus:             /dev/i2c-1
   EDID synopsis:
      Mfg id:           DEL
      Model:            DELL P2214H
      Serial number:    KW14V51H0N3B
      Manufacture year: 2015
      EDID version:     1.3
   VCP version:         2.1

Display 5
   I2C bus:             /dev/i2c-2
   EDID synopsis:
      Mfg id:           WAC
      Model:            Cintiq A1300
      Serial number:    3IAH000034
      Manufacture year: 2013
      EDID version:     1.3
   VCP version:         2.1

Display 6
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           SAM
[... 486 lines cut ...]
      Manufacture year: 2015
      EDID version:     1.3
   VCP version:         2.1

Display 47
   I2C bus:             /dev/i2c-2
   EDID synopsis:
      Mfg id:           WAC
      Model:            Cintiq A1300
      Serial number:    3IAH000034
      Manufacture year: 2013
      EDID version:     1.3
   VCP version:         2.1

Display 48
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           SAM
      Model:            U28E590
      Serial number:    HTPK416852
      Manufacture year: 2018
      EDID version:     1.4
   VCP version:         2.0

```

InconsolableCellist, 2020-02-09:
and environment --verbose: 

```
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false

The following tests probe the runtime environment using multiple overlapping methods.

*** Basic System Information ***

ddcutil version: 0.9.2

/proc/version:
   Linux version 4.19.0-6-amd64 (debian-kernel@lists.debian.org) (gcc version 8.3.0 (Debian 8.3.0-6)) #1 SMP Debian 4.19.67-2+deb10u2 (2019-11-11)

Architecture:     x86_64
Distributor id:   Debian
Release:          3
Found a known architecture

/proc/cmdline:
   BOOT_IMAGE=/vmlinuz-4.19.0-6-amd64 root=UUID=a549d036-d666-440c-af27-55427a23e65f ro quiet

Processor information as reported by lscpu:
   Architecture:        x86_64
   CPU op-mode(s):      32-bit, 64-bit
   Byte Order:          Little Endian
   Address sizes:       36 bits physical, 48 bits virtual
   CPU(s):              4
   On-line CPU(s) list: 0-3
   Thread(s) per core:  1
   Core(s) per socket:  4
   Socket(s):           1
   NUMA node(s):        1
   Vendor ID:           GenuineIntel
   CPU family:          6
   Model:               58
   Model name:          Intel(R) Core(TM) i5-3570 CPU @ 3.40GHz
   Stepping:            9
   CPU MHz:             2346.581
   CPU max MHz:         3800.0000
   CPU min MHz:         1600.0000
   BogoMIPS:            6784.38
   Virtualization:      VT-x
   L1d cache:           32K
   L1i cache:           32K
   L2 cache:            256K
   L3 cache:            6144K
   NUMA node0 CPU(s):   0-3
   Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm cpuid_fault epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid fsgsbase smep erms xsaveopt dtherm ida arat pln pts md_clear flush_l1d

DMI Information from /sys/class/dmi/id:
   Motherboard vendor:       Dell Inc.
   Motherboard product name: 0GXM1W
   System vendor:            Dell Inc.
   System product name:      OptiPlex 7010
   Chassis vendor:           Dell Inc.
   Chassis type:             15 - Space-saving

Byte order checks:
   Is big endian (local test):       false
   WORDS_BIGENDIAN macro (autoconf): not defined
   __BYTE_ORDER__ macro (gcc):       __ORDER_LITTLE_ENDIAN__

*** Primary Check 1: Identify video card and driver ***

[... 1160 lines cut ...]
   /dev/i2c device numbers:       0 1 2 3
   sysfs_i2c_devices_exist:       true
   /sys/bus/i2c device numbers:   0 1 2 3
   dev_i2c_devices_required:      true
   module_i2c_dev_needed:         false
   module_i2c_dev_builtin:        false
   loadable_i2c_dev_exists:       true
   i2c_dev_loaded_or_builtin:     true
   group_i2c_checked:             true
   group_i2c_exists:              true
   dev_i2c_common_group_name:     i2c
   all_dev_i2c_has_group_i2c:     true
   any_dev_i2c_has_group_i2c:     true
   all_dev_i2c_is_group_rw:       true
   any_dev_i2c_is_group_rw:       true
   cur_uname:                     root
   cur_uid:                       0
   cur_user_in_group_i2c:         false
   cur_user_any_devi2c_rw:        true
   cur_user_all_devi2c_rw:        true

Configuration suggestions:
   Current user has RW access to all /dev/i2c-N devices.
   Skipping further group and permission checks.
```

rockowitz, 2020-02-10:
Thanks for running the tests.  Here are my takeaways. 

Everything works with the special Nvidia driver settings, and always works with the Samsung U238E590, so the question is whether ddcutil's I2C communication can be improved at the margin for problematic driver/monitor combinations. 

1) There is no difference depending on whether read() or ioctl() is used.

2) EDID reads succeed for the marginal monitors if multiple single byte reads are used. It does not matter whether the reads take the usual path through the I2C communication layer or are done directly the function reading the EDID.  This is consistent with your read-edid observation. 

3) DDC communication succeeds for the marginal monitors when using multiple single byte reads instead of a single multi-byte read. 

This is at variance here with what I see here with 2 marginal monitors (Acer X35W and Dell P2411h) and the amdcpu driver.  EDID reads succeed with all variants.  However, I2C communication fails for multiple single byte reads and succeeds for a single muliti-byte read - the opposite of your result.  

These results suggest that EDID reading can be marginally improved using single byte reads.  That would at least reduce the confusion caused by ddcutil not reporting a monitor that exists, whether or not DDC communication subsequently works.  However, I would like to see the result of more tests before making any changes to the low level IO. 

Regards, 
Sanford

closed 2025-02-09 by rockowitz