rockowitz/ddcutil #105: Xscreensaver blocking ddc communication

kevinsaccasyn, 2020-02-12:
System: Raspberry pi 3b+
OS: Raspbian Buster (with desktop)
Screen: Iiyama ProLite T2252MSC

TLDR: ddc communication fails when xscreensaver started to display the screensaver. When the screensaver is turned off (after keyboard input / mouse input), ddc communication is restored. I couldn't find anything about this on the web. I tried this countless times, and communication is lost each time the screensaver starts. I'm sure it's the screensaver as it also happens when you open the xscreensaver settings page (and the small preview is displayed).

/TLDR

I started using ddcutil today and it took me quite some time to figure this one out. I tried the version in the raspbian buster repo, but am now using the latest version built from source. I was not able to get ddc communication working at first. 

I'm currently replacing a wall mounted tablet that I use to control my home automation. As I did with the tablet, I would like to use the screen as a photo frame when it is idle. I'm using xscreensaver to do this. The problem is that once the screensaver is turned on, I lose ddc communication.

When screensaver is displayed
```
Invalid display
   I2C bus:             /dev/i2c-2
   EDID synopsis:
      Mfg id:           IVM
      Model:            PLT2252
      Serial number:    1151394501080
      Manufacture year: 2019
      EDID version:     1.3
   DDC communication failed
```

Screensaver not displayed
```
Display 1
   I2C bus:             /dev/i2c-2
   EDID synopsis:
      Mfg id:           IVM
      Model:            PLT2252
      Serial number:    1151394501080
      Manufacture year: 2019
      EDID version:     1.3
   VCP version:         2.1
```
Does anyone know what causes this behavior?

rockowitz, 2020-02-12:
Command line option --ddc reports DDC data errors.

Options --trace ddc, --trace ddcio, and --trace i2c turn on tracing in 
the respective layers and my give a clue, though the output is voluminous.

ksacca, 2020-02-12:
Thanks for taking the time to respond. Here are all logs you requested (both when the screensaver is active and not active). It's not really a problem as I can disable the screensaver before initiating changes to the screen, but still wanted to let you know.

[ddc WithoutScreenSaver.txt](https://github.com/rockowitz/ddcutil/files/4194343/ddc.WithoutScreenSaver.txt)
[ddc WithScreenSaver.txt](https://github.com/rockowitz/ddcutil/files/4194344/ddc.WithScreenSaver.txt)
[traceddcio WithoutScreenSaver.txt](https://github.com/rockowitz/ddcutil/files/4194345/traceddcio.WithoutScreenSaver.txt)
[traceddcio WithScreenSaver.txt](https://github.com/rockowitz/ddcutil/files/4194346/traceddcio.WithScreenSaver.txt)
[traceddc WithoutScreenSaver.txt](https://github.com/rockowitz/ddcutil/files/4194347/traceddc.WithoutScreenSaver.txt)
[traceddc WithScreenSaver.txt](https://github.com/rockowitz/ddcutil/files/4194348/traceddc.WithScreenSaver.txt)
[tracei2c WithoutScreenSaver.txt](https://github.com/rockowitz/ddcutil/files/4194349/tracei2c.WithoutScreenSaver.txt)
[tracei2c WithScreenSaver.txt](https://github.com/rockowitz/ddcutil/files/4194350/tracei2c.WithScreenSaver.txt)

rockowitz, 2020-02-12:
Even without the screen saver, there are communication errors.  I2C is 
an unreliable protocol, so ddcutil retries (10 times for a data read).  
It's just that communication is somewhat worse with the screen saver, so 
the retry limit is reached.  If you run ddcutil with the --stats option 
you'll see the retry statistics.

Typically this "dirty" communication reflects a poor I2C implementation 
in the monitor.  What happens if you plug in a regular monitor?

Option --sleep-multiplier increases (or decreases) the amount of time 
ddcutil waits for the monitor form a response.  The default times are 
those in the DDC spec.   You could try something like 
"--sleep-multiplier 4" to quadruple the sleep times. Sometimes this 
helps with marginal monitors.

Finally, it appears that you can alter the I2C bus speed. This is new to 
me.   See: 
https://www.raspberrypi-spy.co.uk/2018/02/change-raspberry-pi-i2c-bus-speed/ 
The default speed is 100k.  I would try something like 50k, i.e. 50000.

Let me know how what you find.

-- Sanford

closed 2020-05-01 by rockowitz