rockowitz/ddcutil #106: Failed to detect a monitor

linuxhw, 2020-02-17:
`ddcutil` can't find a monitor (DELL via HDMI on Fedora 31 X11).

`sudo ddcutil environment`:

```
The following tests probe the runtime environment using multiple overlapping methods.

** Basic System Information **

ddcutil version: 0.9.9

/proc/version:
   Linux version 5.4.13-201.fc31.x86_64 (mockbuild@bkernel03.phx2.fedoraproject.org) (gcc version 9.2.1 20190827 (Red Hat 9.2.1-1) (GCC)) #1 SMP Tue Jan 21 17:21:47 UTC 2020

Architecture:     x86_64
Distributor id:   Fedora
Release:          31
Found a known architecture

/proc/cmdline:
   BOOT_IMAGE=(hd1,gpt2)/vmlinuz-5.4.13-201.fc31.x86_64 root=/dev/mapper/fedora--ssd-root ro resume=/dev/mapper/fedora--ssd-swap rd.lvm.lv=fedora-ssd/root rd.lvm.lv=fedora-ssd/swap rhgb quiet

** Primary Check 1: Identify video card and driver **

Obtaining card and driver information from /sys...
Primary video controller at PCI address 0000:00:02.0 (boot_vga flag is set)
   Device class:        x030000    VGA compatible controller
   Vendor:              x8086      Intel Corporation
   Device:              x0412      Xeon E3-1200 v3/4th Gen Core Processor Integrated Graphics Controller
   Subvendor/Subdevice: 1458/d000  Gigabyte Technology Co., Ltd
   Driver name:         i915
   Driver version:      Unable to determine
   I2C device:          i2c-3      name: i915 gmbus dpc
   I2C device:          i2c-1      name: i915 gmbus vga
   I2C device:          i2c-4      name: i915 gmbus dpb
   I2C device:          i2c-2      name: i915 gmbus panel
   I2C device:          i2c-0      name: i915 gmbus ssc
   I2C device:          i2c-5      name: i915 gmbus dpd

** Primary Check 2: Check that /dev/i2c- exist and writable ***

Current user: root (0)

Checking /dev/i2c-* devices...

Unless the system is using the AMD proprietary driver fglrx, devices /dev/i2c-*
must exist and the logged on user must have read/write permission for those
devices (or at least those devices associated with monitors).

Typically, this access is enabled by:
  - setting the group for /dev/i2c-* to i2c
  - setting group RW permissions for /dev/i2c-*
  - making the current user a member of group i2c

Alternatively, this can be enabled by just giving everyone RW permission
The following tests probe for these conditions.

Checking for /dev/i2c-* devices...
   crw------- 1 root root 89, 0 фев 17 13:45 /dev/i2c-0
   crw------- 1 root root 89, 1 фев 17 13:45 /dev/i2c-1
   crw------- 1 root root 89, 2 фев 17 13:45 /dev/i2c-2
   crw------- 1 root root 89, 3 фев 17 13:45 /dev/i2c-3
   crw------- 1 root root 89, 4 фев 17 13:45 /dev/i2c-4
   crw------- 1 root root 89, 5 фев 17 13:45 /dev/i2c-5
   crw------- 1 root root 89, 6 фев 17 13:45 /dev/i2c-6

Current user (root) has RW access to all /dev/i2c-* devices.

** Primary Check 3: Check that module i2c_dev is loaded **

Checking for module i2c_dev...
   Module i2c-dev is NOT built into kernel
   Loadable i2c-dev module found
   Module i2c_dev is loaded

** Primary Check 4: Driver specific checks **

Performing driver specific checks...
No driver specific checks apply.

** Additional probes **

Scanning /proc/modules for driver environment...
   Found other loaded module:      i2c_dev
   Found other loaded module:      i2c_i801
   Found video driver module:      i915
   Found other loaded module:      i2c_algo_bit
   Found other loaded module:      drm_kms_helper
   Loaded drm module depends on:   i915,drm_kms_helper,
   Loaded video module depends on: i915,

Testing if modules are loaded using /sys...
   Module amdgpu           is NOT loaded
   Module fbdev            is NOT loaded
   Module fglrx            is NOT loaded
   Module fturbo           is NOT loaded
   Module i915             is loaded
   Module mgag200          is NOT loaded
   Module nvidia           is NOT loaded
   Module nouveau          is NOT loaded
   Module radeon           is NOT loaded
   Module vboxvideo        is NOT loaded
   Module vc4              is NOT loaded
   Module drm              is loaded
   Module i2c_algo_bit     is loaded
   Module i2c_dev          is loaded
   Module i2c_piix4        is NOT loaded
   Module ddcci            is NOT loaded

Examining /sys/bus/i2c/devices...
   /sys/bus/i2c/devices/i2c-3/name:   i915 gmbus dpc
   /sys/bus/i2c/devices/i2c-1/name:   i915 gmbus vga
   /sys/bus/i2c/devices/i2c-6/name:   SMBus I801 adapter at f040
   /sys/bus/i2c/devices/i2c-4/name:   i915 gmbus dpb
   /sys/bus/i2c/devices/i2c-2/name:   i915 gmbus panel
   /sys/bus/i2c/devices/i2c-0/name:   i915 gmbus ssc
   /sys/bus/i2c/devices/i2c-5/name:   i915 gmbus dpd

Env_Accumulator:
   architecture:                  x86_64
   distributor_id                 Fedora
   Drivers detected:              i915
   /dev/i2c device numbers:       0 1 2 3 4 5 6
   sysfs_i2c_devices_exist:       true
   /sys/bus/i2c device numbers:   0 1 2 3 4 5 6
   dev_i2c_devices_required:      true
   module_i2c_dev_needed:         true
   module_i2c_dev_builtin:        false
   loadable_i2c_dev_exists:       true
   i2c_dev_loaded_or_builtin:     true
   group_i2c_checked:             true
   group_i2c_exists:              false
   dev_i2c_common_group_name:     root
   all_dev_i2c_has_group_i2c:     false
   any_dev_i2c_has_group_i2c:     false
   all_dev_i2c_is_group_rw:       true
   any_dev_i2c_is_group_rw:       false
   cur_uname:                     root
   cur_uid:                       0
   cur_user_in_group_i2c:         false
   cur_user_any_devi2c_rw:        true
   cur_user_all_devi2c_rw:        true

Configuration suggestions:
   Current user has RW access to all /dev/i2c-N devices.
   Skipping further group and permission checks.
```

linuxhw, 2020-02-17:
Collected logs: https://linux-hardware.org/index.php?probe=fc7ce0211e

linuxhw, 2020-02-17:
It's Dell U2417H DEL40E8 connected to integrated graphics card.

linuxhw, 2020-02-17:
Error: "No displays found".

linuxhw, 2020-02-17:
Fixed by:

`xhost +local:`

Is it possible to add this command to `ddcutil`?

rockowitz, 2020-02-17:
Please send the output of "ddcutil detect --verbose" for the failure 
situation.

Also, when sending the output of "ddcutil environment"  use the 
""--verbose" option as well.  The basic environment output is intended 
to help users diagnose ddcutil configuration problems (e.g. drivers, 
permissions).  With" --verbose", it includes detailed information to 
help with remote debugging.

rockowitz, 2020-02-17:
What is your configuration that requires this command?   What is the 
output of "xhost" both before and after you issue the "xhost +local" 
command?

linuxhw, 2020-02-18:
> What is your configuration that requires this command?   What is the output of "xhost" both before and after you issue the "xhost +local" command?

Most of modern Linux distributions deny access to X11 from root, so we need to allow access by `xhost +local:root` before executing `ddcutil ...` via `sudo` or `su` or use `sudo -E ddcutil ...`.

Reproduced on Debian 8 and 10. Output of `xhost` is usually:

    SI:localuser:USERNAME

This affects also `x*` tools like `xrandr` or `xdpyinfo` as well.

`xhost +local:` adds `LOCAL:` to the output list.

rockowitz, 2020-02-18:
This is the first error report I've received regarding xhost.   It would 
appear that most users' systems are properly configured automatically.

xhost settings fall into the category of execution environment, along 
with things like driver setup and group permissions.  These are not 
things that ddcutil should change.  It is, however, the sort of thing 
that the "ddcutil environment" command is intended to check for in order 
to help users configure their systems.

> What is your configuration that requires this command? What is the
>     output of "xhost" both before and after you issue the "xhost
>     +local" command?
>
> Most of modern Linux distributions deny [...]
Is this before or after the "xhost +" command?

On my present system (Ubuntu 19.04) there is somewhat more output:

   access control enabled, only authorized clients can connect
   SI:localuser:rock

Please describe what the symptoms of bad xhost configuration would be.  
What would the output of xhost look like in this situation? What does it 
need to look like?  How to make the "xhost +" change persistent.   This 
is the information needed to document the issue and create a check in 
"ddcutil environment".

closed 2020-11-16 by rockowitz