rockowitz/ddcutil #109: Dual Boot install broke ddcutil, "No monitors found"

Twiggeh, 2020-03-04:
First of all tank you (and everyone involved) for making this. I was able to change my brightness / color temperature on my Cintiq for the first time in 5 years :)

The issue : I installed ddcutil on my usb (Manjaro installation, not live usb) and other things like xf86-input-wacom, which require a restart to work. Ddcutil is working on my usb installed os like expected.
But when I installed the os to my ssd in a dual boot configuration only xf86-input-wacom started working after a reboot, ddcutil did not (Error : No monitors found).

`sudo ddcutil detect --verbose`
```shell
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

```

`ddcutil interrogate 2>&1`
[interrogateOutput.txt](https://github.com/rockowitz/ddcutil/files/4287524/interrogateOutput.txt)

`sudo ddcutil environment --verbose`
[environmentOutput.txt](https://github.com/rockowitz/ddcutil/files/4287533/environmentOutput.txt)

rockowitz, 2020-03-04:
It looks like module dev-i2c is required but not loaded.  See the 
configuration suggestion at the end of the output of the environment 
command.

Twiggeh, 2020-03-04:
It works! Thank you so much :)

closed 2020-03-04 by Twiggeh