rockowitz/ddcutil #113: Invalid display - DDC communication failed

jmaltis, 2020-03-16:
Hello,
I'm facing the "Invalid display - DDC communication failed" issue and not sure how to fix it.
My setup:
- Recent screen (AU0 M270HVN02.0)
- Raspberry Pi (3B+) 
- HDMI connection

I was following this page https://www.ddcutil.com/raspberry/#background.
Problem, when running `ddcutil detect`, here's my output:
```
$ sudo ddcutil detect
Invalid display
   I2C bus:             /dev/i2c-2
   EDID synopsis:
      Mfg id:           ROW
      Model:            AAA
      Serial number:    Unspecified
      Manufacture year: 2016
      EDID version:     1.3
   DDC communication failed
```

If it can help, here is the output of `ddcutil environment` -> [environment.txt](https://github.com/rockowitz/ddcutil/files/4336001/environment.txt)

I believe the screen is supposed to be supporting DDC/CI, so I'm not sure what went wrong here.
Let me know if you need more information.
Sorry if it's a duplicate, but I couldn't find a working solution/workaround/lead on the similar issues.
Thank you!

rockowitz, 2020-03-16:
1) AUO N270HVN02.0 is the model number for the panel in a monitor, not 
the monitor.  See 
(http://www.panelook.com/M270HVN02.0_AUO_27.0_LCM_overview_17595.html). 
What is the monitor make and model?

2) If a monitor supports DDC/CI, typically but not always this will be 
indicated in the on-screen display.

3) Run command "i2cdetect 2". Does it show slave addresses 50 and 37 as 
active?

4) Please run command "ddcutil environment /verbose" and send me the 
output.

Sanford

jmaltis, 2020-03-16:
1) I'm actually directly plugged onto the panel (using m.nt68676.2a control board, if it matters) and not using an actual monitor
3) Output of `i2cdetect 2`
```
$ i2cdetect 2
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-2.
I will probe address range 0x03-0x77.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- 37 -- -- 3a -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: 50 51 52 53 54 55 56 57 -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
4) Output of `ddcutil environment --verbose`: [environment.txt](https://github.com/rockowitz/ddcutil/files/4339826/environment.txt)

Thank you for the support!
Jérémy

rockowitz, 2020-03-16:
>  2. Output of |i2cdetect 2|

Further thought.  Really quite basic. What happens if you plug the 
control board into a regular computer, and what happens if you plug a 
normal monitor into the Pi?

jmaltis, 2020-03-17:
- Normal monitor plugged into the Pi -> working perfectly
```
sudo ddcutil detect
Display 1
   I2C bus:             /dev/i2c-2
   EDID synopsis:
      Mfg id:           LEN
      Model:            L24q-20
      Serial number:    U5P0N506
      Manufacture year: 2017
      EDID version:     1.3
   VCP version:         2.2

$ sudo ddcutil getvcp 10
VCP code 0x10 (Brightness                    ): current value =    70, max value =   100
$ sudo ddcutil setvcp 10 50
```
Output of `ddcutil environment` just in case -> [environment.txt](https://github.com/rockowitz/ddcutil/files/4340840/environment.txt)

- Control board plugged to another computer -> no way to control its brightness either
The other computer is a Mac, so I was running some other tool which is supposed to control the external monitor brightness, but it didn't work either.

Which leads me to believe the issue is coming from my panel itself...

closed 2020-03-24 by rockowitz