rockowitz/ddcutil #114: Fedora 32 ddutil Copr repository

amarshall, 2020-03-22:
Fedora 32 is now in beta! Would it be possible to enable Fedora 32 for the [ddcutil Copr repository](https://copr.fedorainfracloud.org/coprs/rockowitz/ddcutil/)? It's enabled for ddcui, but not ddcutil. I believe there is also a setting to auto-enable for new branches that may be worth enabling. Thanks!

rockowitz, 2020-03-23:
Updated.

Note that COPR package ddcui is now a second package in project ddcutil 
instead of having its own project ddcui.   The "Follow Fedora branching" 
option in project ddcutil's setting has been enabled, as per your 
suggestion.

Thank you for bringing this to my attention.

-- Sanford

closed 2020-03-23 by rockowitz

amarshall, 2020-03-24:
Thank you for the prompt resolution, and for ddcutil in general!