rockowitz/ddcutil #116: Display not found

Anyeos, 2020-03-27:
I have a monitor Samsung 24" connected to DVI and a projector EPSON (old 1999) connected to VGA with a DisplayPort adapter.
amdgpu open driver with a Radeon RTX 580 Nitro+ PCI-E videocard.
Both detections fails with a "Display not found". I cannot do anything at all. But as you can see there are some responses. 
And as I verified the response is not always the same. With the same command there are different responses. It appears to be random but the last response always begin with "6e 6e". So there are a pattern and can mean something.

detect:
```
Invalid display
   I2C bus:             /dev/i2c-8
   EDID synopsis:
      Mfg id:           SEC
      Model:            
      Serial number:    
      Manufacture year: 1999
      EDID version:     1.1
   DDC communication failed
   This appears to be a laptop display. Laptop displays do not support DDC/CI.

Invalid display
   I2C bus:             /dev/i2c-11
   EDID synopsis:
      Mfg id:           SAM
      Model:            SyncMaster
      Serial number:    0
      Manufacture year: 2010
      EDID version:     1.3
   DDC communication failed

```
The verbose output of i2c: 
sudo ddcutil -v --ddc --mfg SEC getvcp ALL > [ddcutil_SEC.txt](https://github.com/rockowitz/ddcutil/files/4391644/ddcutil_SEC.txt)
sudo ddcutil -v --ddc --mfg SAM getvcp ALL > [ddcutil_SAM.txt](https://github.com/rockowitz/ddcutil/files/4391645/ddcutil_SAM.txt)

I guess there are more research to do about how works the communication with monitors in practice.

rockowitz, 2020-03-27:
With displays that old the first question is whether DDC is supported at 
all.  The first check is to use i2cdetect (in package i2c-tools) to see 
if I2C slave address x37 (DDC) is even responsive.

Also, the DisplayPort-VGA combination is problematic.  VGA, DVI, HDMI 
have dedicated pins for I2C. DisplayPort implements I2C communication in 
software using its AUX channel, so both sides need to understand how to 
do the conversion.

That said, it does look like I2C communication is taking place, but is 
garbled.  The double bytes in what otherwise would be a proper response 
packet is a giveaway.  Some monitors, particularly older ones, cannot 
keep up with the 100mhz speed of the I2C bus. Unfortunately, there's no 
way for ddcutil to tell the video drivers to slow down.  You could try 
using the --sleep-multiplier option, e.g. "--sleep multiplier 4.0" to 
tell ddcutil to wait longer than the spec requires between sending a 
request to the monitor and reading the response.  This occasionally helps.

If you'd like, send the output of "ddcutil environment --verbose" (as an 
attachment please, not inline) and I'll take a look, but I doubt there 
will be more to add.  Do make sure that command i2cdetect exists on your 
system.

Sanford

Anyeos, 2020-03-29:
I tested with Windows 10 and the SoftMCCS and it say that eventually the EPSON projector does not support control anything. But my Samsung monitor supported some things (not all). It can control the brightness, contrast and colors. And it can be reset to factory too.

It is a clean Windows without drivers installed, it worked as is out of the box.

I think there are a need to make some low level research. It cannot be so complicated because Windows is not so complicated for doing that things.
And the SoftMCCS app appears to have a complete database only. It does not appear to be a high level engineering app. 

I think some better algorithms must be implemented over i2c to try different workarounds. I don't say it is posible with this app. It will be neccesary to work near driver/module or low level access too.
There appears that the i2c commands are being send at wrong speed or wrong bit level like all serial communication protocols that must be equal on both sides. But i2c is so simple that I don't think it must be very difficult to someone with an oscilloscope and computers and electronics knowledge.

If there are no manner to control speed from userspace apps, we are on a lack of features from the module itself. i2c is low level so it is neccesary to control each aspect of it at low level. There are no definitive way of doing that. It is a trial and error only. Because there are no enforced rules for that each manufacturer wants to do what he want with this. So we must do the same, trial and error.

Thank you for your support I just wanted to put to factory reset my EPSON projector because I don't have the remote control and it is configured not so good and I have no form of put it to factory defaults without the remote. I just thinked that it supported some DDC control to send that message but eventually it is not the case.

I tried with a very high value of --sleep-multiplier without success in both cases. The response is bad and I think it is because there are some misuse of the physics layer of the communication. Maybe there are some EMI produced because a wrong usage of the i2c communication over the wires. And if that is the case it is a fail of the module itself. I am electronic so I know what I am talking about. But I don't have oscilloscope, sorry.

rockowitz, 2020-03-30:
Aneyos,

If you'd like to tackle the problem, have at it. All I can say is that 
DDC communication has not been a priority for the video driver folks.  
See, for example, https://bugs.freedesktop.org/show_bug.cgi?id=100954 
regarding the i915 driver, particularly the exchange starting with 
comment 15.

Sanford

closed 2025-02-09 by rockowitz