rockowitz/ddcutil #117: Validate setvcp

KeyofBlueS, 2020-04-15:
Hi,

is it possible to make an option for validating setvcp?
I made a script (https://github.com/KeyofBlueS/yddcutil) and I would like to insert a validation for setvcp, this is a proof of concept i.e. for brightness:
```
MAX_RETRIES=10
while true; do
	ddcutil --bus=2 setvcp 10 80
	RETRY="$(expr "$RETRY" + 1)"
	if ddcutil -t --bus=2 getvcp 10 | awk -F' ' '{print $4}' | grep -xq "80"; then
		RETRY=0
		break
	elif [ "$RETRY" -eq "$MAX_RETRIES" ]; then
		RETRY=0
		break
	fi
done
```

But I was wondering if this kind of option could be included directly in ddcutil, usage would be like:
ddcutil --validate --validate-max-retries=10 --bus=2 setvcp 10 80

Thanks for your attention

rockowitz, 2020-04-15:
By default, setvcp reads the value after writing it to verify that the 
monitor has acted properly on the command.  Option --noverify turns off 
this check. See http://www.ddcutil.com/cmdopts/#option-noverify. So what 
you're calling the --validate option is already there - it's the default 
--verify.

If verification fails, ddcutil returns 1.  If you execute ddcutil with 
option --exec it will display internal exception reports before 
releasing them.  You'll see the string DDCRC_VERIFY.  (Bear in mind that 
checking for DDCRC_VERIFY is brittle.  The --exec option is a 
development aid.)

The --validate-max-retries option is interesting.  I can see its 
usefulness.  As you clearly know, ddcutil's retry functionality exists 
at the I2c level to manage the unreliability of I2C.   Value 
verification failure is at a higher level.  Adding retries at that level 
is conceivable, but is non-trivial.  Integration with thread-specific 
settings (important for the API), statistics, and command loadvcp come 
to mind immediately. I'll give it some further thought, but it would be 
a while before it got to the top of the queue.

setvcp verification failure is rare.  I'd like to know more about the 
context in which this occurs - monitor, video card, etc.  Can I ask you 
to send the output of the interrogate command (preferable as a file)?

Your yddcutil project is an interesting approach to creating a GUI for 
ddcutil using shell scripting. It's clearly a non-trivial amount of 
work.  What's the impetus for its development? I'll continue to watch 
it, and do be in touch with any questions.

Regards,
- Sanford

On 4/14/20 10:34 PM, KeyofBlueS wrote:

KeyofBlueS, 2020-04-16:
Thanks for the detailed response @rockowitz.

The default --verify option... I don't know how I managed not to see it.

I don't "really" have setvcp verification failure or similar. They pop up just when a too low sleep-multiplier value is set, I think it's normal. And yes in this case a "--validate-max-retries" option could be useful, If you think it's worth it.
Anyway ddcutil.out attached, maybe You could find interesting things in it.

About my project, all started with my half broken monitor physical buttons. As I can hardly configure the monitor with those nasty buttons, I wanted to recreate the same menu on pc side. Then I got a second monitor and started to adapt the script on it, but I thought "what if the menu could be dynamically created based on monitor capabilities?"
It's just an exercise for me, and could be useful for someone, so I decided to share it.

Thanks and Best Regards.

rockowitz, 2020-04-17:
Thank you for sending the interrogate output.  Nothing particularly 
noteworthy, though that in itself is noteworthy given the age of the 
monitors.

And yes, those buttons on monitors are bloody annoying.

Regards,
Sanford

closed 2020-06-05 by rockowitz