rockowitz/ddcutil #119: All displays not found after system update(s). (I2C device: Not found)

Okxa, 2020-04-20:
I do not know the exact point this happened, but not all of my displays are detected. IIRC it worked before I updated to kernel 5.7rc1.

```
[2020-04-06T16:55:19+0300] [ALPM] upgraded linux-mainline (5.6rc6-1 -> 5.6-1)
[2020-04-18T14:41:24+0300] [ALPM] upgraded linux-mainline (5.6-1 -> 5.7rc1-1)
[2020-04-20T19:55:33+0300] [ALPM] upgraded linux-mainline (5.7rc1-1 -> 5.7rc2-1)
```

The displays that do not work are the same model as one which works. (minus the Valve index VR headset, but I could detect the other displays before even with it connected.)

 Using `ddcutil interrogate 2>&1`, I found that I2C device is not found.

```
   EDID: ...3020003C  Mfg: ACR  Model: Acer AL2216W   SN: L74090466440 
      I2C device:     /dev/i2c-3
      XrandR output:  DVI-1-0
      DRM connector:  DVI-D-1
      UDEV name:      Radeon i2c bit bus 0x91
      UDEV syspath:   /devices/pci0000:00/0000:00:01.3/0000:03:00.2/0000:1d:04.0/0000:2c:00.0/i2c-3/i2c-dev/i2c-3
      sysfs drm path: /sys/class/drm/card0/card0-DVI-D-1
      sysfs drm I2C:  (null)

   EDID: ...300A005A  Mfg: ACR  Model: G246HYL        SN: LZ5EE0072400 
      I2C device:     /dev/i2c-5
      XrandR output:  HDMI-1-0
      DRM connector:  HDMI-1
      UDEV name:      Radeon i2c bit bus 0x93
      UDEV syspath:   /devices/pci0000:00/0000:00:01.3/0000:03:00.2/0000:1d:04.0/0000:2c:00.0/i2c-5/i2c-dev/i2c-5
      sysfs drm path: /sys/class/drm/card0/card0-HDMI-A-1
      sysfs drm I2C:  (null)

   EDID: ...35380036  Mfg: PHL  Model: PHL 240S4QY    SN: ZV81847000858
      I2C device:     /dev/i2c-13
      XrandR output:  HDMI-A-1
      DRM connector:  HDMI-2
      UDEV name:      AMDGPU DM i2c hw bus 3
      UDEV syspath:   /devices/pci0000:00/0000:00:03.1/0000:2e:00.0/0000:2f:00.0/0000:30:00.0/i2c-13/i2c-dev/i2c-13
      sysfs drm path: /sys/class/drm/card1/card1-HDMI-A-2
      sysfs drm I2C:  (null)

   EDID: ...38370148  Mfg: PHL  Model: PHL 240S4QY    SN: AU81820000087
      I2C device:    Not found
      XrandR output:  DisplayPort-0
      DRM connector:  DP-1
      UDEV name:      (null)
      UDEV syspath:   (null)
      sysfs drm path: /sys/class/drm/card1/card1-DP-1
      sysfs drm I2C:  (null)

   EDID: ...363401FB  Mfg: PHL  Model: PHL 240S4QY    SN: ZV81847000864
      I2C device:    Not found
      XrandR output:  DisplayPort-2
      DRM connector:  DP-3
      UDEV name:      (null)
      UDEV syspath:   (null)
      sysfs drm path: /sys/class/drm/card1/card1-DP-3
      sysfs drm I2C:  (null)

   EDID: ...000001C3  Mfg: VLV  Model: Index HMD      SN:              
      I2C device:    Not found
      XrandR output:  (null)
      DRM connector:  DP-2
      UDEV name:      (null)
      UDEV syspath:   (null)
      sysfs drm path: /sys/class/drm/card1/card1-DP-2
      sysfs drm I2C:  (null)`
```
Full output: https://gist.github.com/Okxa/72d944b65d94dc3daff14b30976d9300

Also various i2c bus examinations fail:
```
Examining device /dev/i2c-12...
      Functionality: I2C_FUNC_I2C, I2C_FUNC_SMBUS_PEC, I2C_FUNC_SMBUS_QUICK, 
                     I2C_FUNC_SMBUS_READ_BYTE, I2C_FUNC_SMBUS_WRITE_BYTE, 
                     I2C_FUNC_SMBUS_READ_BYTE_DATA, I2C_FUNC_SMBUS_WRITE_BYTE_DATA, 
                     I2C_FUNC_SMBUS_READ_WORD_DATA, I2C_FUNC_SMBUS_WRITE_WORD_DATA, 
                     I2C_FUNC_SMBUS_PROC_CALL, I2C_FUNC_SMBUS_WRITE_BLOCK_DATA, 
                     I2C_FUNC_SMBUS_READ_I2C_BLOCK, I2C_FUNC_SMBUS_WRITE_I2C_BLOCK
      Unable to read EDID, psc=EIO(-5): Input/output error

      Trying simple VCP read of feature 0x10...
      Attempt 1 to read feature returned hard error: EIO(-5): Input/output error
      DDC communication failed.
```

rockowitz, 2020-04-20:
Clearly a driver issue, probably for the amdgpu folks.  The problem only 
occurs for DisplayPort devices. The device id cross reference table 
shows no  UDEV syspath, which should end with "i2c-dev/ i2c-n".  Also 
DisplayPort devices should have an i2-n directory under the sys drm path 
directory.

As confirmation, if you have sufficiently old kernels around, try 
booting with one of those.

Did check that you built the amdpu driver for 5.7rc2?  That may well 
solve the problem.  If you post to the amdgpu list, the first thing 
they'll ask is whether you're running the latest driver.

Okxa, 2020-04-21:
Can confirm that it finds the displays when on kernel `5.6.5.arch3-1`.

As for the gpu driver, I currently had latest in the official arch repos:

```
mesa 20.0.4-2
xf86-video-amdgpu 19.1.0-1 (xorg-drivers)
```

I can test development versions with the [mesa-git](https://aur.archlinux.org/packages/mesa-git/) AUR package, however it will take a while to do so.

I would need to either compile it myself (just takes a longish while), or wait until the [unofficial repo](https://wiki.archlinux.org/index.php/Unofficial_user_repositories#mesa-git) updates its packages, as currently there is an issue with `llvm-libs` dependency.

As for now, because 5.6.5 kernel is properly released (which had some important fixes for my GPU), I do not necessarily need to use the mainline kernel. Therefore *my* issue is sort of "fixed", **although I will test with the mainline kernel and git drivers when I have time to do so**, to see if the issue remains.

closed 2020-05-01 by rockowitz