rockowitz/ddcutil #120: Problem with AOC monitor with ADMI/DVI adapter. - Error DDC communication failed

dqtup, 2020-05-17:
I can't execute ddcutils, please in external monitor from my note Dell.
HDMI out Dell and DVI in AOC monitor.
Please see stdoud of ddc enviroment:
Invalid display
   I2C bus:             /dev/i2c-5
      I2C address 0x50 (EDID) present: true 
      Is eDP device:                   false
      /sys/bus/i2c/devices/i2c-5/name: i915 gmbus dpb
   EDID synopsis:
      Mfg id:           AOC
      Model:            2470W
      Serial number:    E72G6XA003087
      Manufacture year: 2016
      EDID version:     1.3
      Product code:     9328
      Extra descriptor: 
      Video input definition: 0x80 - Digital Input
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.641, 0.338
      Green x,y:        0.311, 0.619
      Blue  x,y:        0.159, 0.059
      Extension blocks: 0
   EDID source: 
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 05 e3 70 24 0f 0c 00 00   ..........p$....
      +0010   18 1a 01 03 80 34 1d 78 2a 2a c5 a4 56 4f 9e 28   .....4.x**..VO.(
      +0020   0f 50 54 bf ef 00 d1 c0 b3 00 95 00 81 80 81 40   .PT............@
      +0030   81 c0 01 01 01 01 02 3a 80 18 71 38 2d 40 58 2c   .......:..q8-@X,
      +0040   45 00 09 25 21 00 00 1e 00 00 00 fd 00 32 4c 1e   E..%!........2L.
      +0050   53 11 00 0a 20 20 20 20 20 20 00 00 00 fc 00 32   S...      .....2
      +0060   34 37 30 57 0a 20 20 20 20 20 20 20 00 00 00 ff   470W.       ....
      +0070   00 45 37 32 47 36 58 41 30 30 33 30 38 37 00 a1   .E72G6XA003087..
   DDC communication failed
   Is DDC/CI enabled in the monitor's on-screen display?

And i2cdetect 5
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- 37 -- -- 3a -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- 49 -- -- -- -- -- -- 
50: 50 -- -- -- -- -- -- -- -- 59 -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

rockowitz, 2020-05-18:
Please run "sudo ddcutil environment --verbose" and send the output as 
an attachment, not inline.

Have you verified that DDC is enabled in the monitor's on screen display.

Are you using a docking station?

rockowitz, 2020-09-11:
No user response to request for detail.

closed 2020-09-11 by rockowitz