rockowitz/ddcutil #121: 0 brightness via ddc equals 30 brightness in monitor menu (Dell 2407wfp)

Epskampie, 2020-05-20:
Just discoverd ddc/ci, really cool, just what I needed.
One problem, when I set the brightness and contrast to 0 using `ddcutil`:

```
sudo ddcutil --bus 5 setvcp 12 0
sudo ddcutil --bus 5 setvcp 10 0
```

The screen doesn't become completely dark, and the monitor menu shows that the brightness is actually `30`. Setting is to `100` does work correctly, and values in between "scale" correctly, so it seems the monitor is enforcing some minumum brightness.

Is this probably just a quirk of the monitor firmware or is there something I can do?

Thanks!

rockowitz, 2020-05-21:
> Just discovered ddc/ci, really cool, just what I needed.

Welcome to the world of DDC.   There are standards, but different 
companies interpret them differently, and there are many "corner cases" 
where implementers are simply on their own.

"Continuous" features like brightness and contrast have maximum value 
that the monitor reports.  The minimum value is implicitly 0.  It looks 
like the firmware is designed to not let the user set a blank display.

If you're trying to blank the monitor while not turning it off, here are 
a couple things you could try:

- Set the values of features x16, x18, x1a (red, green, and blue gain) 
to 0.  The monitor might let you do that.  You might need to set the 
value of feature x14 (select color preset) to a User value  (x0b, x0c, 
x0d) before changing the  color gains.  Of course, you'll need to know 
the values you want to reset the features to.

- Play with feature xd6 (power mode).  The DPMS spec is defined for 
CRTs, but values 2 or 3 might dim the display but allow it to be 
reawakened,

You wrote that "the monitor menu is actually 30".   What does getvcp 
report?  It should be consistent with the OSD, but I've seen cases where 
they get out of sync.

What is the mfg/model of the monitor?  Better yet, run "ddcutil probe" 
as root and send the output as an attachment.

Sanford

Epskampie, 2020-05-25:
Thanks for your response! It was a long weekend here, so I could only try out your suggestions today, since this is my work PC.

I'm not trying to blank the display, simply set it's brightness quite low, since bright displays annoy my eyes.

> Set the values of features x16, x18, x1a

I tried this, and it works quite well! A nice hack to sort of "go below" brightness 30, thanks!

> You wrote that "the monitor menu is actually 30".   What does getvcp
report?  It should be consistent with the OSD, but I've seen cases where
they get out of sync.

It's not consistent. Out of interest, i mapped the full relationship in the table below. As you can see, ddcutil values 0 - 50 are mapped to the OSD 30 - 50 range. It seems as dell was afraid that people would set their brightness too low using ddc, then call tech support or something. They must have changed their mind later, because I've got another newer dell monitor without this issue.

|ddcutil|OSD|
|--------|------|
|0|30|
|10|34|
|20|38|
|30|42|
|40|46|
|50|50|
|60|60|
|70|70|
|80|80|
|90|90|
|100|100|

> What is the mfg/model of the monitor?  Better yet, run "ddcutil probe"
as root and send the output as an attachment.

[probe.txt](https://github.com/rockowitz/ddcutil/files/4675955/probe.txt)

Don't think there is a lot that can be done here, seems like Dell just built the firmware this way. Feel free to close this issue if you agree.

closed 2020-11-16 by rockowitz