rockowitz/ddcutil #122: FindDDCUtil.cmake is installed to the wrong location

simotek, 2020-05-25:
Generally only Modules shipped by cmake should be installed into `/usr/share/cmake/modules` Modules shipped by 3rd partys should be installed into `<libdir>/cmake/libraryname` in the same way autotools .pc files should be installed into `<libdir>/pkgconfig` the correct example for openSUSE would be `/usr/lib64/cmake/ddcutil/`

rockowitz, 2020-05-26:
Thank you for the clarification.  I had puzzled over the proper location 
for .pc files.  Integrating all the conflicting information I found on 
the web, I concluded that architecture agnostic .pc files were installed 
in /usr/share/pkgconfig, while architecture specific files belonged 
under <libdir>.   Are there definitive specifications for the proper 
location for .pc and .cmake files, or is this just generally known?

I'm wrapping up the next releases of ddcutil (0.9.9) and ddcui (0.1.1) 
.  I expect they'll be out in a couple weeks.  I'll address the location 
of the .cmake (and .pc) files as part of the release process.

Sanford

simotek, 2020-05-26:
> Thank you for the clarification.  I had puzzled over the proper location for .pc files.  Integrating all the conflicting information I found on the web, I concluded that architecture agnostic .pc [...]

for .pc files I think the info you already have is right, for cmake I also spent a while searching to see if its documented anywhere, it certainly isn't for openSUSE, cmake's tutorial[1] uses `<prefix>/lib/foo*/` but also says "If your project does not already have a /lib/Foo*/ directory you may prefer to put the package file in /lib/cmake/Foo*/".

Beyond that I found a mail thread that said not to use `/usr/share/cmake` [2].  On openSUSE almost everything is using `/usr/lib64/cmake/Foo' (with no version numbers), although there are a couple using `/usr/lib64/Foo` only boost and openjpeg are putting in version numbers as it may make sense to have multiple versions of them at some point I might clean up some of the other packages.

1. https://gitlab.kitware.com/cmake/community/-/wikis/doc/tutorials/Packaging
2. https://gitlab.kitware.com/cmake/cmake/issues/17530

closed 2025-02-11 by rockowitz