rockowitz/ddcutil #125: Validation fail for setvcp (value does not match value set)

slavov-vili, 2020-06-01:
Hello,
I am using two monitors (BenQ GW2480, Lenovo L22e-20). I have set the udev rules for i2c group access, added my user to the group and everything. ddcutil runs and seems to detect the monitors fine.

However, when I try to set the brightness (ddcutil setvcp 10 70) I get a validation error. I made sure that DDC/CI is on for both monitors. I attached the outputs of my setvcp and also of interrogate. 

Could the problem be caused because I am using two different monitors with different versions of EDID? I tried running only one monitor at the time, but got the same result.

Thank you very much in advance.

[interrogate.txt](https://github.com/rockowitz/ddcutil/files/4712609/interrogate.txt)
[setvcp_verbose.txt](https://github.com/rockowitz/ddcutil/files/4712610/setvcp_verbose.txt)

rockowitz, 2020-06-01:
First, thank you taking the trouble send the interrogate output.  I don't see anything out of the ordinary there, but that itself is information.

I've seen situations where setvcp appears to work yet the monitor brightness doesn't change, and conversely situations where setvcp appears to fail yet the monitor brightness does change. Sometimes the monitor overrides values.  See for example issue #121.

As background, the setvcp command causes a DDC Set Feature Request packet to be sent.  At the I2C level the monitor acknowledges that it has successfully received that packet, but there's no guarantee that it has successfully executed the request.  Therefore, ddcutil internally performs a getvcp to check that the change has been made successfully (unless ***--noverify*** is specified).

Some things to check: 

- Do you see a change in the monitor brightness?  (Testing with very high or low values makes it apparent whether a change has occurred.)
- Does the value in the On Screen Display change?
- Try using the ***--noverify*** option on the setvcp command, then explicitly issue a getvcp command.
- DDC mandates that the host give the monitor enough time to perform an operation.  The ***--sleep-multiplier*** option will increase those delays, e.g. ***--sleep-multiplier 2*** will double the delays from those in the DDC/CI specification. 
- Are multiple ddcutil calls occurring in a script?  Inserting a delay between calls may help.

Let me know if any of these actions gives a clue.

slavov-vili, 2020-06-02:
Thank you very much for the super quick and detailed response.

It turned out that the Dynamic Contract Ration (DCR on Lenovo, Eye Care on BenQ) was overriding the value that I was trying to set, therefore turning off the feature ended up solving my problem. Thank you for explaining what could be causing it, I would not have thought to check this otherwise.
I might have missed this in the official Documentation and I apologize if I did.

For the sake of completeness:
- At the time of me reporting the issue the monitor brightness did not change at all
- The Value on the OSD also did not change
- I did not end up needing to use the --noverify option
- The default sleep modifier seemed to work quite well too, I did occasionally get console messages that the set operation had succeeded after n sleeps (usually 1-3)
- I was trying to get ddcutil to work before integrating it into a script. I am still planning to do this so I will keep this in mind.

Thank you again. I am closing the issue since the problem was not with your utility.

closed 2020-06-02 by slavov-vili

rockowitz, 2020-06-03:
Thank you for letting me know how you solved the problem and what you 
observed.  I'll add DCR to the list of possible reasons for setvcp failure.

Re the retry message, was it more specifically  "ddc_write_read() 
succeeded after N sleep and retry for DDC Null Response", where of 
course N is replaced by the number of retries?

This trace message is issued unconditionally so that it stands out.  
It's an unusual situation.  It's useful to know that it occurred. Given 
now recent your monitors are I'm surprised that it occurs.

Sanford

slavov-vili, 2020-06-03:
Yes, that is exactly the message that I received, it seems to only show up when I set the brightness of the Lenovo monitor. I don't know if that itself could be the cause or some other factor such as using HDMI or the length of the cable.

rockowitz, 2020-06-05:
This is not a physical communication problem.  In this case a DDC Null 
Message might be returned instead of a the expected reply message to 
indicate that for some reason the monitor was unable to process the 
request.  There's some special code for handling the Null Message to 
improve performance.  The purpose of the trace message is to noisily 
report that the  special case path through the code occurred. It's hard 
to properly test, since it's an artifact of how a particular monitor 
behaves. So I appreciate your reporting it.

Sanford