rockowitz/ddcutil #126: In man page for setvcp, position of [+|-] is misleading

Ergo42, 2020-06-06:
The man page suggests,
> **setvcp** [+|-] **feature-code** **new-value**

but only this form runs,
> **setvcp** **feature-code** [+|-] **new-value**

So, this command won't work,
`# ddcutil setvcp + 10 50`
but, this one will.
`# ddcutil setvcp 10 + 50`

Ergo42, 2020-06-06:
Nevermind. I had some after thoughts so I searched through branch 9.9. There it was fixed. I had tested 9.2 on pi3, and 9.8 on my desktop so I just assumed. :)
My bad.

closed 2020-06-06 by Ergo42

rockowitz, 2020-06-06:
No problem.  The man page had gotten badly out of sync with the actual 
code. As you found, the version in the upcoming 0.9.9 is extensively 
revised.  I wasn't sure that people actually used it.

Sanford