rockowitz/ddcutil #128: Compilation failure when building with cython

jpalus, 2020-06-22:
Since eed8e62 ddca_show_stats() was updated and includes new "by_thread" argument. However this change was not reflected in src/cython/cyddc.pyx causing build failures if building with --enable-cython.

rockowitz, 2020-06-22:
Jan,

cython, cffi, swig, and gobject_api were experiments in layering a 
Python API over the C API, with the expectation that Python would be the 
best language to use for creating a GUI.  In the end I decided that it 
was easier to use C++ for the GUI than to build and maintain a Python 
layer.  I've kept the directories in case I decide to revisit a Python 
interface, but I should have eliminated the configure options.  I 
wouldn't expect anything to build at this point.   I'll remove the 
options in the next release.

Having worked on ddcui for over a year now, I think that decision was 
correct.   When I have time, I hope to give a talk comparing the various 
ways to layer a Python API on top of a C API.  It would be subtitled 
"TNTAAFL - YMMV".  (There's No Such Thing As A Free Lunch, Your Mileage 
May Vary)  Each of the solutions was going to require hand editing some 
files for each API change.

Regards,
Sanford

jpalus, 2020-06-22:
That is perfectly fine. Just needed a clear statement from upstream that we should abandon Python API.

To give some background -- as a Linux distribution we tend to enable most available features in packages so that's why we're still building with Python API. But now that it's officially confirmed that it's unmaintained/scheduled for removal we'll just disable it.

rockowitz, 2020-06-22:
ddcutil's configure has numerous options that are useful only in the 
development environment.  The options you should build with are the ones 
that default to yes: --enable-lib, ---enable-usb, --enable-drm, 
--enable-x11.

I'll also add documentation for downstream.

closed 2020-07-05 by rockowitz