rockowitz/ddcutil #129: FindDDCUtil.cmake does not provide ddcutil version

ypnos, 2020-06-28:
In your API you have breaking changes based on version. However, the supplied Find script for CMake does not determine nor expose the installed version in a variable.

I would like to patch KDE's powerdevil to support version 0.9.9 without breaking compatibility to older versions. The only change needed to be based on version number would be a call to `ddca_feature_list_string`.

rockowitz, 2020-06-28:
Header file ddcutil_macros.h has #defines for the version number, and 
ddca_ddcutil_version()/ddca_ddcutil_version_string() report the version 
at execution time.

Can you describe in more detail what you want to do?   If 
FindDDCUtil.cmake could report the version, how would you use it?

I have to admit that FindDDCUtil.cmake was supplied by a user some years 
ago, and I haven't given it much thought since.  ddcui's CMakeList.txt 
uses pkg_check_modules() to check for libddcutil.

Sanford

ypnos, 2020-06-29:
Thank you. I was not aware of the macro so that would solve the problem for my case.

Exposing the version through CMake has the advantage that it can be used within CMake, e.g. to ensure that a compatible version is installed/selected for compilation.
However it is rather involved, you would need to put a version file somewhere that is used by the find script. So I'm not sure it's worth the work.

rockowitz, 2020-06-29:
FindDDCUtil.cmake is installed as part of the development package, as 
are the include files.  So I would think that all that's required is for 
FindDDCUtil.cmake to read ddcutil_macros.h.  It already looks for 
ddcutil_c_api.h to identify the include directory.  Or is there 
something more subtle going on here?

ypnos, 2020-06-29:
I edited my comment to reflect that `pkg_check_modules` already does provide version information.

To answer your question:

1. Yes, the macros header file could be parsed to extract the version.
2. Mind that the presence of FindDDCUtil.cmake does not imply the presence of the development package. Software using it would typically ship its own version of it. It's how the CMake crowd does it. Not a big deal though, just in case you weren't aware.

closed 2025-02-09 by rockowitz