rockowitz/ddcutil #130: Assertion fails in parse_capabilities.c:722

jthomme1, 2020-07-13:
Hi everyone,
I just discovered this tool and found out that my monitor indeed does support configuration via ddc/ci.
However, when I try to run `# ddcutil capabilities` it fails with:
```ddcutil: parse_capabilities.c:722: next_capabilities_segment: Assertion `*pos == '('' failed.```
Issuing `ddcutil detect --verbose` yields:
```
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  false
   Dynamic sleep adjustment enabled:            false

Display 1
   I2C bus:             /dev/i2c-1
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      false
      I2C address 0x37 (DDC) responsive:  true 
      /sys/bus/i2c/devices/i2c-1/name:   i915 gmbus dpc
   EDID synopsis:
      Mfg id:           ACI
      Model:            ASUS PB287Q
      Serial number:    
      Manufacture year: 2015
      EDID version:     1.3
      Product code:     10403
      Binary sn:        11717 (0x00002dc5)
      Extra descriptor: 
      Video input definition: 0x80 - Digital Input
      Supported features:
         DPMS active-off
         Digital display type: RGB 4:4:4 + YCrCb 4:2:2
         Standard sRGB color space: True
      White x,y:        0.313, 0.329
      Red   x,y:        0.637, 0.341
      Green x,y:        0.312, 0.625
      Blue  x,y:        0.154, 0.054
      Extension blocks: 1
   EDID source: 
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 04 69 a3 28 c5 2d 00 00   .........i.(.-..
      +0010   2d 19 01 03 80 3e 22 78 3a 1c b5 a3 57 4f a0 27   -....>"x:...WO.'
      +0020   0d 50 54 bf ef 00 d1 c0 81 40 81 80 95 00 b3 00   .PT......@......
      +0030   71 4f 81 c0 01 01 04 74 00 30 f2 70 5a 80 b0 58   qO.....t.0.pZ..X
      +0040   8a 00 6d 55 21 00 00 1a 02 3a 80 18 71 38 2d 40   ..mU!....:..q8-@
      +0050   58 2c 45 00 6d 55 21 00 00 1e 00 00 00 fd 00 1e   X,E.mU!.........
      +0060   50 18 a0 1e 00 0a 20 20 20 20 20 20 00 00 00 fc   P.....      ....
      +0070   00 41 53 55 53 20 50 42 32 38 37 51 0a 20 01 e9   .ASUS PB287Q. ..
   VCP version:         2.1
   Controller mfg:      Mstar
   Firmware version:    0.0
   Monitor returns DDC Null Response for unsupported features: false

Invalid display
   I2C bus:             /dev/i2c-4
      I2C address 0x50 (EDID) responsive: true 
      Is eDP device:                      true 
      /sys/bus/i2c/devices/i2c-4/name:   AUX A/port A
   EDID synopsis:
      Mfg id:           CMN
      Model:            
      Serial number:    
      Manufacture year: 2018
      EDID version:     1.4
      Product code:     5362
      Binary sn:        0 (0x00000000)
      Extra descriptor: N140HCG-GQ2
      Video input definition: 0xa5 - Digital Input (DisplayPort)
      Supported features:
         Digital display type: RGB 4:4:4
         Standard sRGB color space: False
      White x,y:        0.313, 0.329
      Red   x,y:        0.640, 0.330
      Green x,y:        0.300, 0.600
      Blue  x,y:        0.150, 0.060
      Extension blocks: 0
   EDID source: 
   EDID hex dump:
              +0          +4          +8          +c            0   4   8   c   
      +0000   00 ff ff ff ff ff ff 00 0d ae f2 14 00 00 00 00   ................
      +0010   16 1c 01 04 a5 1f 11 78 02 ee 95 a3 54 4c 99 26   .......x....TL.&
      +0020   0f 50 54 00 00 00 01 01 01 01 01 01 01 01 01 01   .PT.............
      +0030   01 01 01 01 01 01 36 36 80 a0 70 38 20 40 2e 1e   ......66..p8 @..
      +0040   24 00 35 ad 10 00 00 18 00 00 00 fe 00 4e 31 34   $.5..........N14
      +0050   30 48 43 47 2d 47 51 32 0a 20 00 00 00 fe 00 43   0HCG-GQ2. .....C
      +0060   4d 4e 0a 20 20 20 20 20 20 20 20 20 00 00 00 fe   MN.         ....
      +0070   00 4e 31 34 30 48 43 47 2d 47 51 32 0a 20 00 bb   .N140HCG-GQ2. ..
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.

```
Just querying the capabilities for `--bus=1` also does not succeed. But, for example:
```
ddcutil --bus=1 getvcp 60
VCP code 0x60 (Input Source                  ): HDMI-2 (sl=0x12)
```
As you see, the monitor is connected via HDMI directly to my Laptop, a Lenovo t490, running arch on kernel `5.7.7-arch1-1`.
If you need any other information, let me know.
Have a nice day.

rockowitz, 2020-07-13:
There's an error parsing the capabilities string.  Please send the 
output of "ddcutil capabilities --brief".

jthomme1, 2020-07-14:
Hi, the output of `ddcutil capabilities --brief` is exactly the same:
```
ddcutil: parse_capabilities.c:722: next_capabilities_segment: Assertion `*pos == '('' failed.
zsh: abort      sudo ddcutil capabilities --brief
```
Another thing: through my dock I get a `DDC communication failed`. I have read the list of comments in the freedesktop gitlab concerning this issue and I suppose that - as of right now - there is nothing to do about it until the kernel display port code gets fixed, right?

rockowitz, 2020-07-14:
I have pushed out changes on ddcutil development branch 1.0.0-dev to 
avoid parsing the capabilities string when the --brief option is used.   
If you build from the development branch "ddcutil capabilities --brief" 
should now report the raw capabilities string.

Re the docking station problem, it really is a driver issue.  I'd 
suggest adding yourself to the CC list, as an indication of interest in 
the problem.

jthomme1, 2020-07-16:
Hi, sorry for the long response time, but I now have the capabilities string.
```
Unparsed capabilities string: (prot(monitor) type(LCD)model LCDPB287 cmds(01 02 03 07 0C F3) vcp(02 04 05 08 0B 0C 10 12 14(05 06 08 0B) 16 18 1A 60(11 12 0F) 62 6C 6E 70 8D(01 02) A8 AC AE B6 C6 C8 C9 D6(01 04) DF) mccs_ver(2.1)asset_eep(32)mpu(01)mswhql(1))
```

rockowitz, 2020-07-16:
Branch 1.1.0-dev has been updated.

The  capabilities string returned by the monitor is invalid.  The value 
in the model segment, i.e. "LCDP287" is not parenthesized. I have 
modified the code to no longer trigger an assert(), instead an error 
message is issued and parsing terminates.

The capabilities command is purely informational, as the capabilities 
string returned by monitors is in general unreliable.

Thank you for bringing this case to my attention and your help in 
debugging it.

Sanford

On 7/16/20 6:44 AM, jthomme1 wrote:
> |(prot(monitor) type(LCD)model LCDPB287 cmds(01 02 03 07 0C F3) vcp(02 
> 04 05 08 0B 0C 10 12 14(05 06 08 0B) 16 18 1A 60(11 12 0F) 62 6C 6E 70 
> 8D(01 02) A8 AC AE B6 C6 C8 C9 D6(01 04) DF) 
> mccs_ver(2.1)asset_eep(32)mpu(01)mswhql(1)) |

closed 2020-08-03 by rockowitz