rockowitz/ddcutil #131: Program Hangs after --async

jcstill, 2020-07-18:
After using `--async`, all subsequent commands with or without `--async` spawn second process and whole program hangs. I am not sure if the second process is normal as I only noticed it after having to forcefully kill ddcutil with `sudo kill -9 PID`.

Version 0.8.6:
[image]

Version 0.9.9:
[image]

I only got verbose outputs after this started happening (just the detect command):
Version 0.8.6:
[image]

Version 0.9.9:
[image]

Unfortunately the process is in state D (uninterruptible sleep). I assume a reboot will get rid of these and fix the problem altogether, though haven't tried yet. I will reboot and update.

rockowitz, 2020-07-18:
> ddcutil_HEADSHOTS
> ddcutil_HEADSHOTS_2
> ddcutil_verbose_detect
> ddcutil_verbose_detect_2

Thank you for reporting the hang. There have been similar reports in the 
past, but never have been able to reproduce the problem.  Are you able 
to reproduce it?  If so, does it occur when you invoke ddcutil without 
using sudo?

If you can reproduce the problem, try usingrecently added option 
<http://www.ddcutil.com/release_notes/> "--timeout-i2c-io".  This option 
causes most I2C IO calls to time out instead of waiting forever.  If 
timeout occurs, a debug message is issued to draw attention to the event.

I'm not sure what you're getting at regarding the "--verbose" option.  
It should cause the settings information to be shown at the start of any 
command with that option.

-- Sanford

jcstill, 2020-07-19:
Rebooting fixed the issue, as expected. However, the kernel sat waiting for those processes to end for about 15 min before I decided to cut power.

As for the verbose thing, I meant that I did not run the program with verbose before the hang, so there is no before/after reference for this machine (not sure if it matters).

I am not sure if this occurs without sudo, as I get permission denied errors. I will test if this occurs running from the root user as well.

I was able to reproduce this, though I did not document and it was almost exactly the same as:
>Correct. If I call this:
>
>```
>/usr/bin/ddcutil -d 1 setvcp 10 50 &
>/usr/bin/ddcutil -d 2 setvcp 10 50 &
>```
>The second command will hang.
>
>However, changing it to this:
>
>```
>/usr/bin/ddcutil -d 1 setvcp 10 50
>sleep .1
>/usr/bin/ddcutil -d 2 setvcp 10 50
>```
>Has resulted in no hangs when calling the batch command.

_Originally posted by @nvahalik in https://github.com/rockowitz/ddcutil/issues/86#issuecomment-508773371_

Ill do some more testing and document it.

jcstill, 2020-07-19:
I think I may have been incorrect in assuming this was caused by `async`. It seems that `async` works as intended:

>note: all file with "baseline" were run before running `async`
>every run as non-priveleged user looks identical to `detect_baseline_noprivs.txt`
>every run as root user looks identical to the respective `sudo` run

command | result
-|-
`ddcutil -v detect` | [detect_baseline_noprivs.txt](https://github.com/rockowitz/ddcutil/files/4943153/detect_baseline_noprivs.txt)
`sudo ddcutil -v detect` | [detect_baseline_sudo.txt](https://github.com/rockowitz/ddcutil/files/4943154/detect_baseline_sudo.txt)
`sudo ddcutil -v -d 2 setvcp 10 75` | [setvcp_baseline_sudo.txt](https://github.com/rockowitz/ddcutil/files/4943159/setvcp_baseline_sudo.txt)
`sudo ddcutil -v -d 2 --timeout-i2c-io setvcp 10 75` | [setvcp_timeout-i2c-io_sudo.txt](https://github.com/rockowitz/ddcutil/files/4943160/setvcp_timeout-i2c-io_sudo.txt)
`sudo ddcutil -v --async -d 2 setvcp 10 75` | [setvcp_async_baseline_sudo.txt](https://github.com/rockowitz/ddcutil/files/4943157/setvcp_async_baseline_sudo.txt)
`ddcutil -v --async --timeout-i2c-io -d 2 setvcp 10 75` | [setvcp_async_timeout_sudo.txt](https://github.com/rockowitz/ddcutil/files/4943158/setvcp_async_timeout_sudo.txt)

Basically everything appears to be working as normal. Here is the interrogate:
[interrogate_baseline_sudo.txt](https://github.com/rockowitz/ddcutil/files/4943175/interrogate_baseline_sudo.txt)

I think this is actually the same problem as #86. 

This works in a script (ran with root privs):
```shell
ddcutil -v -d 2 setvcp 10 $1
ddcutil -v -d 2 setvcp 12 $1
```

This causes the hang:
```shell
ddcutil -v --timeout-i2c-io -d 2 setvcp 10 $1 &
ddcutil -v --timeout-i2c-io -d 2 setvcp 12 $1 &
```
[image]
[image]

I will update with `--async` and also adding a `sleep` similar to what @nvahalik did to test further.

This is the script I am using:

```shell
#!/bin/bash

# Tested, Working
#ddcutil -v -d 2 setvcp 10 $1
#ddcutil -v -d 2 setvcp 12 $1

#--------------------------------------------------------

# Tested, Causes Hang
#ddcutil -v --timeout-i2c-io -d 2 setvcp 10 $1 &
#ddcutil -v --timeout-i2c-io -d 2 setvcp 12 $1 &

# Untested
ddcutil -v --timeout-i2c-io -d 2 setvcp 10 $1 &
sleep 1
ddcutil -v --timeout-i2c-io -d 2 setvcp 12 $1 &

#--------------------------------------------------------

# Untested
#ddcutil -v --async --timeout-i2c-io -d 2 setvcp 10 $1 &
#ddcutil -v --async --timeout-i2c-io -d 2 setvcp 12 $1 &

# Untested
#ddcutil -v --async --timeout-i2c-io -d 2 setvcp 10 $1 &
#sleep 1
#ddcutil -v --async --timeout-i2c-io -d 2 setvcp 12 $1 &

#--------------------------------------------------------

# Untested
#ddcutil -v --async -d 2 setvcp 10 $1 &
#ddcutil -v --async -d 2 setvcp 12 $1 &

# Untested
#ddcutil -v --async -d 2 setvcp 10 $1 &
#sleep 1
#ddcutil -v --async -d 2 setvcp 12 $1 &
```

jcstill, 2020-07-19:
I tested all the cases in the script and here are the results:

`async` Hang:
[image]
This one was weird as the first time I ran it, nothing happened (display didn't change) and it did not hang. But, the second time hung:
[image]

`async+sleep` Hang
[image]
[image]

`async+timeout` Hang:
[image]
[image]

`timeout` Hang:
[image]
[image]

`timeout+sleep` Hang:
[image]
[image]

Wokring:

`async+timeout+sleep` Works:
[image]

rockowitz, 2020-07-20:
> (comment)

Jacob,

Thank you for the detailed tests.  I will review what you've observed.

One minor thing to point out is that you can run ddcutil as a non-root 
user.  From the interrogate output, I see that group i2c exists and the 
/dev/i2c devices are assigned to that group, which has RW permission for 
the devices.  All you need to do is add your user to that group.

Alternatively, and you can simply "sudo chmod a+rw /dev/i2c*", though 
you'll have to do this every time you log on.

Regards,
Sanford

jcstill, 2020-07-20:
> One minor thing to point out is that you can run ddcutil as a non-root
user.  From the interrogate output, I see that group i2c exists and the
/dev/i2c devices are assigned to that group, which has RW permission for
the devices.  All you need to do is add your user to that group.

I guess that completely flew over my head. Makes perfect sense though. Thanks.

rockowitz, 2020-07-21:
> timeout_hang_output
> timeout_hang_ps

What your tests show is that the hang happens when running 2 instances 
of ddcutil in parallel (& at the end of command), not when running them 
sequentially.  Based on the prior email and this one, it's unclear 
whether inserting a sleep between the 2 sequential ddcutil calls is 
unnecessary.

Please run a failure case with the following command options: "--trace 
i2c --tid --ts".  This will turn on tracing in the i2c layer, and 
include the thread id and timestamp.

If you can, please run the tests using the current 1.0.0 branch. I've 
made minor changes to tracing that should make the output easier to analyze.

Regards,
Sanford

closed 2025-02-09 by rockowitz