rockowitz/ddcutil #132: "SETVCP command requires even number of arguments" warning while trying to increase or decrease brightness

JakeVanWinkle, 2020-07-26:
The _ddcutil_ [documentation](https://www.ddcutil.com/command_setvcp/) says that you can run the _setvcp_ command with a relative value by putting " + " or " - " between the feature id and value. I ran the command ```ddcutil setvcp 10 + 10``` but _ddcutil_ returns the following warning: _SETVCP command requires even number of arguments_. Is there a problem with the documentation or it this a bug?

rockowitz, 2020-07-27:
Thank you for reporting the problem.

  The "even number of arguments"  message is misleading. It predates 
relative value processing and needs to be updated, but it does indicate 
an error.   setvcp parsing was reworked in release 0.9.9, which added 
special handling in the parser for relative setvcp values.

What version of ddcutil are you running?

If you are running release 0.9.9 then you've encountered a bug. Please 
run the following command and send the output:

$  sudo ddcutil setvcp <your arguments here> --debug-parse --trace top

Be sure to  include the exact command executed.

Sanford

JakeVanWinkle, 2020-07-27:
I just realized that I was running an outdated version of _ddcutil_. After installing the most recent version, the command does run successfully. I apologize for not adequately researching the problem before opening this issue.

closed 2020-07-27 by JakeVanWinkle