rockowitz/ddcutil #133: Assert fails in vcp_feature_codes.c

AtesComp, 2020-07-29:
Getting:
```
ddcutil: vcp_feature_codes.c:xxxx: format_feature_detail_audio_treble_bass: Assertion `code_info->vcp_code == 0x8f || code_info->vcp_code == 0x91' failed.
```

So, **vcp_feature_codes.c** sets up function **format_feature_detail_audio_treble_bass()** for **0x8f**, **0x91**, and **0x93**.  I suspect **0x93** is used and the assert fails.

Change:
```
assert (code_info->vcp_code == 0x8f || code_info->vcp_code == 0x91);
```
to 
```
assert (code_info->vcp_code == 0x8f || code_info->vcp_code == 0x91 || code_info->vcp_code == 0x93);
```

rockowitz, 2020-07-30:
Thank you for reporting the bug.

Function format_feature_detail_audio_treble_bass() was incorrectly being 
called for feature x93 - audio balance. Function 
format_feature_detail_x93_audio_balance()
is now called for this feature when the VCP version is 2.2 or 3.0.  
(Feature x93 is a standard continuous function in VCP 2.0 and 2.1, so 
does not require a special formatting function in that case.)

A fix for this bug has been pushed to branch 1.0.0-dev.

(If you are looking at the changed code, note that 
format_feature_detail_audio_treble_bass() has been renamed 
format_feature_detail_x8f_x91 audio_treble bass()).

Can you confirm that the bug has been fixed?

Regards,
Sanford

AtesComp, 2020-08-25:
Confirmed.  After compile and install, ddcutil (1.0.0) completes the VCP code listing for my monitors.

closed 2020-08-25 by AtesComp