rockowitz/ddcutil #134: Can't use chinese external driver board DDCCI enabled

samuele2723, 2020-07-31:
Dear all,

i'm a happy user of DDCUTIL on some display at home but now i am planning to use it to control a LCD external display (surface pro 3 screen replace 1631 V1.1 LTL120QL01-003) that i am controlling through a dedicated driver board [link here ](https://www.aliexpress.com/item/Sucface-3-pro-LCD-Display-Driver-Board-LTL120QL01-003-Dedicated-2160-1440-2K/32994273859.html?spm=a2g0s.9042311.0.0.18164c4dFYhEXZ)

Once turned on, the OSD menu calls for a Realtek logo and even have DDCCI option to turn on\off.

Here the result i get from the ddcutil interrogate.

Please anyone can help me to understand how to enable DDCUTIL on this board\lcd?

Regards
[interrogate.txt](https://github.com/rockowitz/ddcutil/files/5009017/interrogate.txt)

rockowitz, 2020-07-31:
The i2cdetect output indicates that slave address x50 (EDID) is present on /dev/i2c-2, but slave address x37 (DDC) is not.  So this appears to be a problem in the driver and/or the board.  I assume that the configuration options at https://www.ddcutil.com/raspberry/ haven't worked for you. (That page is somewhat dated.) I suggest you post on a Raspberry Pi list.

(I note that gddccontrol is running.  I don't believe it is the cause of your problem, but as a precaution I suggest exiting the program before tryiing to use ddcutil.)

samuele2723, 2020-08-01:
Hello Rocko! thanks for the quick answer. I went through the configuration when i used the first time ddcutil for my Benq display, and there is currently working when i plug to my pi, therefore i didnt think is a rasp-thing unless may need different configurations for different attache displays.

So the problem is that x37 address missing? Is there anything i can do? On the OSD menu everything looks alright.

Please where do you suggest to post my question?

rockowitz, 2020-08-01:
Your driver board is generating the Realtek logo and OSD.  Since it 
reports DDC/CI enabled, it at least thinks it's responding to I2C slave 
address x37 on the I2C bus. Whether it actually is, of course, is 
another question.  To test would require a hardware analyzer on the I2C 
lines.  Let's assume that the driver board is properly responding to 
signals on slave address x37.

When i2cdetect tests /dev/i2c-2, there is no response.  (This is 
consistent with what ddcutil finds.) Somewhere in the layers of drivers 
between i2c_dev, which i2cdetect talks to, and the driver that actually 
talks to the board,  one of the drivers is not (properly) implementing 
I2C.  It may be that your setup requires some special configuration. Or 
it may simply be that one of the drivers simply does not support I2C in 
your configuration. Assuming again that the board properly implements 
DDC, this is definitely a Raspian driver issue.

I am not familiar with the Raspian mailing lists.

Good luck.

- Sanford

closed 2021-03-13 by rockowitz