rockowitz/ddcutil #135: DELL P2415Q datas for support

cyrinux, 2020-08-02:
Please find data needed for the Dell P2415Q support
[ddutil.log](https://github.com/rockowitz/ddcutil/files/5012449/ddutil.log)

rockowitz, 2020-08-02:
Please describe how the monitor is connected to the laptop?  Which 
connector on the laptop? Which on the monitor?  What kind of cable? Are  
you connected through a docking station?   Has DDC/CI been enabled in 
the monitor's OSD?

cyrinux, 2020-08-03:
Hi @rockowitz,

The laptop is connected with thunderbolt cable on a Dell WD19TB (thunderbolt so) docking station.
Monitor is connected on the dock with a DisplayPort.
Thunderbolt is enable is bios, i'm running well 4k on 60fps.
DDC/CI is well enabled on the monitor.

Do you need something more?

Regards,

rockowitz, 2020-08-03:
It's a well known problem with the i915 driver and docking stations.  
See ddcutil FAQ question *ddcutil* can't communicate with a monitor when 
it's plugged into a docking station <http://www.ddcutil.com/faq/>and bug 
report i915 i2c-dev failures with recent chips and docking stations. 
<https://gitlab.freedesktop.org/drm/intel/-/issues/37>

I suggest you add yourself to the CC list for the bug as a way to 
indicate that many people need this fixed.

Sanford

cyrinux, 2020-08-03:
Thanks for your quick reply, I subscribe to the intel issue so 👍

cyrinux, 2020-08-03:
Should i keep this issue open?

closed 2020-08-03 by cyrinux

rockowitz, 2020-08-03:
The issue is appropriately closed. There's nothing in ddcutil that can 
be done to solve a driver problem.