rockowitz/ddcutil #139: Build on Mac OS X?

mamullen13316, 2020-08-08:
Is it possible to build ddcutil in OS X?   I am trying to build from the tarball and getting stuck at the UDEV dependency...

configure: The package providing libudev.h varies by Linux distribution and release.
configure: It may be a udev specific package, e.g. libudev-dev, libudev-devel
configure: or it may be part of systemd, e.g systemd-devel
configure: error: libudev not found

rockowitz, 2020-08-09:
Porting ddcutil to Mac OS would be a major project.  The udev dependency 
could be eliminated, but that is relatively minor compared to everything 
else.  For starters, you'd have to rip out the i2c layer and parts of 
the ddc layer and replace them with Mac specific code.  (See 
https://github.com/MonitorControl/MonitorControl) .

mamullen13316, 2020-08-09:
Thanks for the reply and the pointer to MonitorControl.

closed 2020-08-09 by mamullen13316