rockowitz/ddcutil #140: Samsung Curved Monitor No DDC/CI

FibroJedi, 2020-08-17:
Hey,

So I just switched to Linux and all bar one thing is working fine: DDC/CI. 

Monitor: Samsung CF390 (FYI this is attacked by HDMI to my laptop)

In Windows this worked most of the time, had to turn the monitor off and on occasionally to refresh it. But with softMCCS (A windows app) and a DDC/CI app I was able to control the backlight and colour tinting.

However in Linux, ddcutil says the monitor is in the database but DDC/CI is not active. While I agree it's not active, it should be possible to activate it. And in Linux I don't know what I'm doing with this one. Hope you can help. From what little I've managed to follow here's some output.

```
sudo ddcutil detect        
Invalid display
   I2C bus:             /dev/i2c-3
   EDID synopsis:
      Mfg id:           SAM
      Model:            C24F390
      Serial number:    H4ZM900890
      Manufacture year: 2019
      EDID version:     1.3
   DDC communication failed

Invalid display
   I2C bus:             /dev/i2c-6
   EDID synopsis:
      Mfg id:           CMN
      Model:            
      Serial number:    
      Manufacture year: 2015
      EDID version:     1.4
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI
```

Great start, eh?

`$ ddcutil getvcp known --bus 6`
Most of this returned "Invalid Response"

`ddcutil interrogate 2>&1`
Attached as a txt file. (Scrap that I couldn't find how to attach, output is at the bottom of my post)

If there's anything else you need to help you, let me know. I'm moving house tomorrow but from Wednesday I should be able to reply again, if our ISP does its job.

Thanks in advance.
_________
```
sudo ddcutil interrogate 2>&1
Setting output level verbose...
Setting maximum retries...
Forcing --stats...
Forcing --force-slave-address..
This command will take a while to run...

   Current time (local): 2020-08-17 22:46:02 BST
   Current time (UTC):   2020-08-17 21:46:02
   Seconds since boot:   12197

*** Basic System Information ***

ddcutil version: 0.9.8

/proc/version:
   Linux version 5.4.0-42-generic (buildd@lgw01-amd64-038) (gcc version 9.3.0 (Ubuntu 9.3.0-10ubuntu2)) #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020

Architecture:     x86_64
Distributor id:   Linuxmint
Release:          20
Found a known architecture

[... 2087 lines cut ...]
   SE_POST_READ           1372
   SE_DDC_NULL               0
   SE_POST_SAVE_SETTINGS     0

Call Stats:
   Type                                     Count    Millisec  (      Nanosec)
   write calls       (IE_WRITE)              1589        1944  (   1944833102)
   read calls        (IE_READ)               1372        3726  (   3726720853)
   open file calls   (IE_OPEN)                  1           0  (         5397)
   close file calls  (IE_CLOSE)                 1           0  (         7418)
   other I/O calls   (IE_OTHER)                 1           0  (         2008)
   Totals:                                   2964        5671  (   5671568778)

Sleep Call Stats:
   Total sleep calls:                                    2771
   Requested sleep time milliseconds :                 138700
   Actual sleep milliseconds (nanosec):                139047  ( 139047358276)

Elapsed milliseconds since last reset (nanosec):    143615  ( 143615965050)
Total elapsed milliseconds (nanoseconds):              157719  ( 157719964930)

Skipping invalid display on bus /dev/i2c-6

Display scanning complete.`
```

rockowitz, 2020-08-18:
To clarify, there's no database that ddcutil consults to activate a 
monitor.  It "simply" attempts to communicate with the monitor's Virtual 
Control Panel (VCP)  using DDC/CI.  It interprets VCP features (e.g. 
that feature x10 is brightness) per the definitions in the Monitor 
Command Control Set (MCCS) specification.

What you've got is a marginal DDC/CI implementation in the monitor/video 
driver combination. Sometimes it works, sometimes it doesn't .   DDC/CI 
is an inherently  unreliable protocol (think UDP).  The client, ddcutil 
in this case, is responsible for retrying failed operations.  With most 
monitors, retries are seldom needed.  If you look at the statistics 
sections of the interrogate output, you'll see that operations commonly 
require multiple retries to succeed, and even then often do not.

Why do things work on Windows but not Linux?  (And I take "In Windows 
this worked most of the time" to indicate that it does not always work 
even in Windows.)  First, I suspect the Windows drivers are just better 
at handling marginal monitors.  It may be that Windows drivers slow down 
the I2C bus speed when errors are encountered.  LInux drivers do not.  
Second, I doubt that monitor manufacturers test on Linux.

Of note, despite being a recent model, the monitor reports its MCCS 
version as 2.0.  That is very old.  The most recent version (for many 
years now) is 2.2.  That suggests that the monitor may be using an old 
DDC/CI implementation as well.

Things to try:

Use the --sleep-multiplier option, (e.g. --sleep-multiplier 3) to 
increase the time that ddcutil waits, as per the DDC/CI spec, between 
sending a request to the monitor and reading the response. The delay is 
intended to give the monitor time to process the request.

If you have another monitor available, try connecting that one instead.  
Does it have similar problems?

The interrogate output shows that the monitor is connected to the on 
chip video implementation, which uses the i915 driver.  What happens 
when you switch the monitor to the discrete Nvidia chip, which currently 
uses the Nvidia proprietary driver?  If you try that, be sure to use the 
Spedial Nvidia Driver Settings <http://www.ddcutil.com/nvidia/>.  You 
might also try using the open source nouveau driver instead of the 
proprietary Nvidia driver.

Good luck and let me know your results.

-- Sanford

On 8/17/20 5:55 PM, FibroJedi wrote:
> Hey,
>
> So I just switched to Linux and all bar one thing is working fine: DDC/CI.
>
> Monitor: Samsung CF390 (FYI this is attacked by HDMI to my laptop)
>
> In Windows this worked most of the time, had to [...]

FibroJedi, 2020-08-19:
Thanks for the explanation, that helps. So, two numpty questions

1. Which command do I apply the --sleep-multiplier to?
2. How do I force the monitor to use the discreet NVIDIA chip?

Sadly I don't own another monitor so I can't check that one. I'll have a nosey around to see if I can force NVIDIA on that screen until you have a chance to get back to me. Thanks again!

rockowitz, 2020-08-19:
>
> Thanks for the explanation, that helps. So, two numpty questions
>
>  1. Which command do I apply the --sleep-multiplier to?
Option --sleep-multiplier applies to any command that communicates with 
the monitor, e.g. detect, setvcp, getvcp.
>
>  1. How do I force the monitor to use the discreet NVIDIA chip?

Probably the simplest way is to edit the kernel command line to affect 
the current boot only.  (You can edit the command line if you use the 
Advanced boot option at the GRUB menu.)  Add "rd.blacklist=i915 
modprobe.blacklist=i915"  (without the quotes). This tells the kernel to 
ignore the i915 driver, so the Nvidia driver will be used.

FibroJedi, 2020-08-25:
Hey, sadly I don't have another monitor. Well, I do, but it's 500 miles south at my Dad's house, so not exactly useful right now. And that monitor is pretty old, so not sure it'd be a valid test.

I'll just have to deal with my "fake" dimming on QRedshift instead of actually using less power to the monitor. Thanks for taking the time to try to get this working for me.

Martin aka Fibro Jedi
Website<https://fibrojedi.me.uk/> | Twitter<https://twitter.com/FibroJedi> | DeviantArt<https://www.deviantart.com/fibrojedi>

|What command did you issue that returned this response?|

|You could try setting the --sleep-multiplier value greater than 3, but
I very much doubt that would have much effect.  In fact, from your
descriptions it appears that the apparent success of --sleep-multiplier
just reflects the random success of DDC/CI commands. My best guess is
that monitor is marginal at the standard (100 mhz) I2C bus speed.
Unfortunately, there's no way to the Linux drivers to slow down.  Again,
if you're able to test with another monitor and that monitor works ok
then the problem is definitely with  your monitor.|

captn3m0, 2021-01-18:
I have the same monitor (C24F390) connected to a Intel NUC (with integrated Intel graphics). The second monitor I have (LG) works perfectly with ddcutil. I have so far only managed one single successful `detect` call:

```
sudo ddcutil detect
Display 1
   I2C bus:             /dev/i2c-5
   EDID synopsis:
      Mfg id:           SAM
      Model:            C24F390
      Serial number:    HTQK801650
      Manufacture year: 2018
      EDID version:     1.3
   VCP version:         2.0
```
This was just after I re-plugged the monitor. Every subsequent call has failed, even with sleep multipliers:

```
sudo ddcutil detect --sleep-multiplier=10
Invalid display
   I2C bus:             /dev/i2c-5
   EDID synopsis:
      Mfg id:           SAM
      Model:            C24F390
      Serial number:    HTQK801650
      Manufacture year: 2018
      EDID version:     1.3
   DDC communication failed
```

[Output for sudo ddcutil --bus 5 interrogate](https://github.com/rockowitz/ddcutil/files/5832026/i.txt)

I'm gonna blame samsung here, but happy to provide more details if they can be of help in maybe a workaround?

Thanks for making `ddcutil`, I'd been trying to dim my monitors for months via a script, and I was so disappointed when I found out that they didn't support HDMI-CEC. `ddcutil` worked great for my usecase 🤗 (atleast on one monitor!)

rockowitz, 2021-01-18:
The DDC read is returning junk bytes. If I'm reading your hardware right 
it looks like your NUC board has HDMI and USB-C outputs, but the monitor 
has only HDMI and (gasp) VGA inputs.  You're connecting the USB-C output 
to the HDMI input.  Type C connectors should be dual-mode.  (See 
www.ddcutil.com/displayport).  How are you making the connection? What 
kind of cable?  Are you using an active or passive adapter? This 
connection could be problematic.  Try connecting the HDMI output to the 
HDMI input.  Let me know if that works before I suggest anything more.

captn3m0, 2021-01-19:
I was first trying it via a USB-C-HDMI —->HDMI, but the above commands were run on direct HDMI—HDMI with no adapter involved. 

Could the cable itself make a difference? I could try a different one.

closed 2021-03-13 by rockowitz

mirh, 2024-06-16:
I'd be really curious to know if even with Samsung's own MagicTune you'd have this unreliability.

captn3m0, 2024-06-17:
I never tried MagicTune  - don't run Windows, and the [monitor's page at samsung website](https://www.samsung.com/us/computing/monitors/curved/samsung-24-curved-led-monitor-lc24f390fhnxza/) never listed it as supported.

I can't find a MagicTune listing on the samsung website - is the tool still supported?

mirh, 2024-06-17:
Mhh, right. It's not a recent monitor by any stretch, but it's still 3 years newer than the [last version](https://www.samsung.com/uk/support/model/LS22A100NS/EN/#downloads) of MT (which may or may not require a bit of [legwork](https://www.overclock.net/threads/calling-all-samsung-monitor-owners-magictune-premium-support.1163166/#post-25823768) to install on W10). 
Still, who knows? There's also supposedly a [firmware](https://www.samsung.com/us/support/downloads/?model=N0020373&modelCode=LC24F390FHNXZA) update I think?

captn3m0, 2024-06-17:
> Upgrade File(USB type)
> ver 1007.0 0.22 MB Dec 05. 2023 All OS 

The firmware gets me a .bin file, but the monitor doesn't have a USB socket, and the manual makes no mention of a firmware upgrade.