rockowitz/ddcutil #143: AOC 24G2W1G4 Invalid display DDC communication failed

tuv7041, 2020-09-01:
Hi, I recently received a AOC 24G2W1G4 monitor. I'm able to make DDC work on windows with AOC's GMenu, but cant make it work on debian. I made sure that DDC was enabled on the monitor's settings. Im running Debian Sid, and this monitor is connected to a Dell Latitude 7490 laptop (used with screen closed).
ddcutil detect:
```
Invalid display
   I2C bus:             /dev/i2c-8
   EDID synopsis:
      Mfg id:           CMN
      Model:
      Serial number:
      Manufacture year: 2016
      EDID version:     1.4
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.

Invalid display
   I2C bus:             /dev/i2c-9
   EDID synopsis:
      Mfg id:           AOC
      Model:            24G2W1G4
      Serial number:
      Manufacture year: 2020
      EDID version:     1.4
   DDC communication failed
```
ddcutil interrogate > 
[interrogate.log](https://github.com/rockowitz/ddcutil/files/5158777/interrogate.log)

Any idea on how to make it work? Thanks!

rockowitz, 2020-09-02:
Looks like you're using the Type-C connector on your laptop connected to 
DisplayPort on the monitor. Have your tried a HDMI to HDMI connection?

tuv7041, 2020-09-02:
Thanks @rockowitz that worked great! 
For some odd reason xrandr shows only 120hz available using hdmi to hdmi when I had 144hz using a usb C adapter. But ddc is working perfectly. Thanks for your help.

closed 2020-09-02 by tuv7041

rockowitz, 2020-09-02:
The problem is that the i915 driver does/did not implement I2C 
communication over DisplayPort's Mulch-Stream-Transport (MST).   
Internally, this is used for all connections, not just the DisplayPort 
connection, on docking stations.  It also is being used for DisplayPort 
in the Type-C connection.

This has been a long standing problem with the i915 driver.  For 
details, see i915 bug report 
<https://gitlab.freedesktop.org/drm/intel/-/issues/37> that I opened 
over 3 years ago.  As it turns out, a fix for this problem just dropped 
today (see the end of the bug report).  You should see this fix in the 
next kernel release.

tuv7041, 2020-09-02:
That's a coincidence! Ill wait for the kernel update and try it. Thanks again @rockowitz for the great software, quick support and the detailed explanation.