rockowitz/ddcutil #144: No Display Found

aspate, 2020-09-21:
Hi All,

I am very new with Linux and RaspberryPi, but I would like to use ddcutil with it.
I tried to use ddcutil detect, and keep getting: No displays found  
I am running a Raspberry Pi Zero attached to a Dell P2219H via HDMI.  
The DDCI is turned on for the monitor.  

I think it may be an issue with i2c, but I am not sure.

I ran the ddcutil interrogate, 
[ddcutil.txt](https://github.com/rockowitz/ddcutil/files/5256625/ddcutil.txt)

I started to spread them back out into lines, but feared messing things up for anyone who knew what words to look for.  

If anyone could provide some help, I would appreciative.  

Thanks!

rockowitz, 2020-09-23:
The interrogate output shows that /dev/i2c-2 does not exist.  On the Pi 
3, this is used for DDC communication with the HDMI connected monitor.  
I do not have experience with the Pi Zero, but it appears similar.  For 
documentation on configuring the Pi (though it has not been updated in 
some time), see http://www.ddcutil.com/raspberry/#raspberry-pi.

kode54, 2020-10-04:
I have a similar problem on Arch with an RX 480, there are no i2c devices in the system, even though I have two displays connected.

rockowitz, 2020-10-04:
Have you run "ddcutil environment"? That likely will diagnose the 
problem.   If not, run "sudo ddcutil interrogate" and submit the output 
as an attachment.

kode54, 2020-10-04:
Yeah, that showed me that I need to have i2c-dev loaded, and that I needed to do the i2c group permission dance. All fixed.

closed 2020-11-15 by rockowitz