rockowitz/ddcutil #146: DDC communication failed over USB-C Dock

criemen, 2020-10-05:
Hi,

I tried using ddcutil in the following setup:
Thinkpad T14s laptop with AMD Ryzen processor+GPU
external monitor connected to the Lenovo USB-C Gen 2 dock via DisplayPort.

However, I get the following output
```
sudo ddcutil detect
Invalid display
   I2C bus:             /dev/i2c-5
   EDID synopsis:
      Mfg id:           AUO
      Model:
      Serial number:
      Manufacture year: 2018
      EDID version:     1.4
   DDC communication failed
   This is an eDP laptop display. Laptop displays do not support DDC/CI.

Invalid display
   I2C bus:             /dev/i2c-6
   EDID synopsis:
      Mfg id:           DEL
      Model:            DELL U2713HM
      Serial number:    PRIVATE
      Manufacture year: 2012
      EDID version:     1.4
   DDC communication failed
```

Is there anything I can provide to help debug the issue?

I did run modprobe i2c-dev, and I'm running ddcutil using sudo to alleviate the permission problems.
I am using ddcutil 0.9.9 from the arch linux repositories.

rockowitz, 2020-10-06:
Have you tried connecting without the dock, using  HDMI?

Please send the output of "sudo ddcutil interrogate" as an attachment.

criemen, 2020-10-07:
Attaching the monitor via HDMI directly to the computer works.
However, the monitor is useless connected via HDMI, because the native resolution (2k) of the monitor is not supported via HDMI.

I attached the output of ddcutil interrogate both with HDMI (working) and displayport over dock (not working).

[ddcutil-interrogate-hdmi.txt](https://github.com/rockowitz/ddcutil/files/5339777/ddcutil-interrogate-hdmi.txt)
[ddcutil-interrogate-displayport-dock.txt](https://github.com/rockowitz/ddcutil/files/5339778/ddcutil-interrogate-displayport-dock.txt)

rockowitz, 2020-10-09:
i2c failure on recent docking stations (i.e. using DisplayPort MST) is a 
well known problem.   See the bug report 
<https://gitlab.freedesktop.org/drm/intel/-/issues/37> that I submitted 
3 years ago. If you scroll to the end, you'll see that a driver fix was 
recently submitted.  It's in the drm driver, so hopefully it fixes the 
problem for amdgpu as well as i2c.   The fix may be in the 5.8 kernel, 
it certainly will be in 5.9.

In the meantime, you might try connecting a USB Type-C connector on the 
laptop to the DisplayPort connector on the monitor.

criemen, 2020-10-09:
5.8.14 does not help, but I expect 5.9 to land somewhat soon, and I'll report back.
Unfortunately I don't have a USB-C to DP cable to test that case.

Thanks for the support!

frogmaster, 2020-10-22:
Using 5.9.1 on arch and Dell WD19TB thunderbolt dock. The issue is unfortunately still not resolved. I've attached the interrogation log.
[ddc_interrogate.txt](https://github.com/rockowitz/ddcutil/files/5423452/ddc_interrogate.txt)

rockowitz, 2020-10-23:
Commit  adb48b26985686f93f20ca71c16c067d790e7af3  (drm/dp_mst: Support remote i2c writes) is not in kernel release 5.9.1.
 
More specifically, it is not currently in 
- branch linux-5.9.y at  git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git

The commit is found in: 
- branch master at  git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git
- branch master at  https://github.com/torvalds/linux.git

frogmaster, 2020-10-23:
Thanks for the clarification. Installed the linux-git arch package. Can confirm that, at least for my use case, ddcutil works great with tb dock and intel graphics.

rockowitz, 2020-12-15:
The changes to file drm_dp_mst_topology.c are in just released kernel branch 5.10.

ChrisAJS, 2021-01-05:
I'll hopefully be able to echo frogmaster's success-  also experiencing an issue with the exact same Dock, (Dell WD19TB), so when I get the latest kernel I'll report whether or not it fixes my issue also. Freaking amazing tool!

whirm, 2021-01-11:
Quick line to confirm it works just fine with my WD15TB dock! Thanks for this great tool!

closed 2025-02-09 by rockowitz

bjmc, 2025-02-11:
I think I'm still seeing this issue on Ubuntu 24.04 on a ThinkPad X1 Carbon Gen 10 with a Dell Thunderbolt dock. External monitor works fine when connected directly to the laptop over HDMI, but fails when connected via the dock.

`ddcutil detect` reports as:

```
Invalid display
   I2C bus:  /dev/i2c-14
   DRM connector:           card1-DP-3
   EDID synopsis:
      Mfg id:               DEL - Dell Inc.
      Model:                DELL S2721QS
      Product code:         41366  (0xa196)
      Serial number:        <redacted>
      Binary serial number: <redacted>
      Manufacture year:     2021,  Week: 12
   DDC communication failed
```

I'm unclear on the status of the resolution [here.](https://gitlab.freedesktop.org/drm/i915/kernel/-/issues/37) Can anyone comment if there's a patch to be applied somewhere, or if this still unresolved for certain docks/laptops?