rockowitz/ddcutil #147: Cannot detect monitors via Thunderbolt dock - Unexpected /sys/bus/i2c/devices file name

JustGitting, 2020-10-08:
Hi,

I've been trying to get a System76 Darter Pro to detect and use two external monitors connected to a thunderbolt 3 docking station (Dell WD19TB) via DisplayPort.

Using System76's own PopOS 20 (Ubuntu based) it works fine, with the external monitors detected automagically. However, Devuan Beowulf (Debian based) does not work.

I've authorized the docking station using bolt (https://gitlab.freedesktop.org/bolt/bolt) for both PopOS and Devuan.

```
# boltctl
 ● Dell WD19TB Thunderbolt Dock
   ├─ type:          peripheral
   ├─ name:          WD19TB Thunderbolt Dock
   ├─ vendor:        Dell
   ├─ uuid:          <UUID> 
   ├─ generation:    Thunderbolt 3
   ├─ status:        authorized
   │  ├─ domain:     <DOMAIN>
   │  ├─ rx speed:   40 Gb/s = 2 lanes * 20 Gb/s
   │  ├─ tx speed:   40 Gb/s = 2 lanes * 20 Gb/s
   │  └─ authflags:  none
   ├─ authorized:    Fri 03 Oct 2020 15:16:49 UTC
   ├─ connected:     Fri 03 Oct 2020 15:16:48 UTC
   └─ stored:        Tue 08 Sep 2020 11:25:17 UTC
      ├─ policy:     auto
      └─ key:        no
```

Using `$ sudo ddcutil interrogate` it's apparent that PopOS finds extra i2c buses on the docking stations, whilst Devuan does not.

Both OS's have the thunderbolt kernel module loaded. The only difference I can see is that PopOS has i2c-dev built in, whilst in Devuan it is an external module that I needed to load manually. 

PopOS finds the DPMST/DPDDC over i2c from the monitors:

```
/sys/bus/i2c/devices/i2c-8/name: DPMST
/sys/bus/i2c/devices/i2c-6/name: DPDDC-A

```
Whilst Devuan only has an AUX port.

`/sys/bus/i2c/devices/i2c-6/name: AUX A/port A`

And the unexpected bus:

```
/sys/bus/i2c/devices/0-0050/name: ee1004
(each_i2c_device ) Unexpected /sys/bus/i2c/devices file name: 0-0050
```
Because I cannot find documentation on how the i915, i2c, thunderbolt, DP detection is meant to work, I'm stuck. Given @rockowitz has made the only tool that has help me so far, I hope I can get some guidance here.

Exerts from the ddcutil command below:
 
PopOS with 5.4 kernel:
```
*** Primary Check 1: Identify video card and driver ***

Obtaining card and driver information from /sys...
Primary video controller at PCI address 0000:00:02.0 (boot_vga flag is set)
Device class: x030000 VGA compatible controller
Vendor: x8086 Intel Corporation
Device: x9b41 UHD Graphics
Subvendor/Subdevice: 8086/2212 Intel Corporation
Driver name: i915
Driver version: Unable to determine
I2C device: i2c-3 name: i915 gmbus dpc
I2C device: i2c-8 name: DPMST
I2C device: i2c-4 name: i915 gmbus misc
I2C device: i2c-2 name: i915 gmbus dpb
I2C device: i2c-9 name: DPMST
I2C device: i2c-5 name: i915 gmbus dpd

<SNIP>

Examining /sys/bus/i2c/devices...
/sys/bus/i2c/devices/i2c-3/name: i915 gmbus dpc
/sys/bus/i2c/devices/i2c-1/name: Synopsys DesignWare I2C adapter
/sys/bus/i2c/devices/i2c-8/name: DPMST
/sys/bus/i2c/devices/i2c-6/name: DPDDC-A
/sys/bus/i2c/devices/i2c-4/name: i915 gmbus misc
/sys/bus/i2c/devices/i2c-2/name: i915 gmbus dpb
/sys/bus/i2c/devices/i2c-0/name: SMBus I801 adapter at efa0
/sys/bus/i2c/devices/i2c-9/name: DPMST
/sys/bus/i2c/devices/i2c-7/name: DPDDC-B
/sys/bus/i2c/devices/i2c-5/name: i915 gmbus dpd
```

Devuan with 5.8 kernel:

```
*** Primary Check 1: Identify video card and driver ***

Obtaining card and driver information from /sys...
Primary video controller at PCI address 0000:00:02.0 (boot_vga flag is set)
Device class: x030000 VGA compatible controller
Vendor: x8086 Intel Corporation
Device: x9b41 UHD Graphics
Subvendor/Subdevice: 8086/2212 Intel Corporation
Driver name: i915
Driver version: Unable to determine
I2C device: i2c-3 name: i915 gmbus dpc
I2C device: i2c-4 name: i915 gmbus misc
I2C device: i2c-2 name: i915 gmbus dpb
I2C device: i2c-5 name: i915 gmbus dpd

<SNIP>

Examining /sys/bus/i2c/devices...
/sys/bus/i2c/devices/i2c-3/name: i915 gmbus dpc
/sys/bus/i2c/devices/0-0050/name: ee1004
(each_i2c_device ) Unexpected /sys/bus/i2c/devices file name: 0-0050
/sys/bus/i2c/devices/i2c-1/name: Synopsys DesignWare I2C adapter
/sys/bus/i2c/devices/i2c-6/name: AUX A/port A
/sys/bus/i2c/devices/i2c-4/name: i915 gmbus misc
/sys/bus/i2c/devices/i2c-2/name: i915 gmbus dpb
/sys/bus/i2c/devices/i2c-0/name: SMBus I801 adapter at efa0
/sys/bus/i2c/devices/i2c-7/name: AUX B/port B
/sys/bus/i2c/devices/i2c-5/name: i915 gmbus dpd
```
Thank you

rockowitz, 2020-10-09:
I2C support on recent docking stations is a long standing problem.  A 
fix recently went into the drm driver.  I'm not sure where it is in the 
release cycle.

See the following freedesktop.org bug report: 
https://gitlab.freedesktop.org/drm/intel/-/issues/37

JustGitting, 2020-10-09:
Thanks @rockowitz,

Looking into [issue 37](https://gitlab.freedesktop.org/drm/intel/-/issues/37), the actual changes to the kernel source are given at [drm/dp_mst: Support remote i2c writes](https://gitlab.freedesktop.org/lyudess/linux/-/commit/adb48b26985686f93f20ca71c16c067d790e7af3)

I've just checked both the Devuan 5.8 kernel source and the PopOS 5.4 kernel source ([https://github.com/pop-os/linux](https://github.com/pop-os/linux)) for the changes shown in the linked commit. However, neither seem to contain the changes shown in the above commit. So it does not explain why PopOS/Ubuntu can detect the monitors using an older kernel than Devuan with a newer kernel.

lauritsriple, 2020-10-09:
I tested the patch on a custom kernel (based on 5.8.14) . It works! Finally I can adjust the brightness of both my monitors using ddcutil. Looking forward to seeing this released!

rockowitz, 2020-10-09:
Are you using the i915 driver, amdgpu, or some other video driver?

Can you run "sudo ddcutil environment --verbose" and submit the output 
as an attachment?  I'd like to see the udev and sysfs probes.  Thanks.

lauritsriple, 2020-10-09:
I think it is the i915. Using optimus-manager to toggle between them. Attached output from both cases, with the intel enabled and with the nvidia enabled. It seems like the i915 is enabled in both cases. Did a reboot between the two logs. For some reason `ddcutil detect` reports 4 displays even though I only have 3 (laptop+2external). Seems like the serial number is the same on two of them. setvcp works on all of them setting `--display` from 1 to 3. Even though 1 and 2 are the same monitor.
[ddc_detect_nvidia.txt](https://github.com/rockowitz/ddcutil/files/5357871/ddc_detect_nvidia.txt)
[ddc_environment_intel_sudo.txt](https://github.com/rockowitz/ddcutil/files/5357873/ddc_environment_intel_sudo.txt)
[ddc_environment_nvidia_sudo.txt](https://github.com/rockowitz/ddcutil/files/5357875/ddc_environment_nvidia_sudo.txt)

rockowitz, 2020-10-10:
@JustGitting , was asking @lauritsriple since the driver update works for him

JustGitting, 2020-10-10:
@rockowitz okay

So what else can I do to investigate why I cannot detect and use the external monitors?

rockowitz, 2020-10-11:
@lauritstiple thanks for sending the output.  It gives me some hints as to how to eliminate that duplicate /dev/i2c devices. 

In both cases the monitors are connected using the i915 driver, not the nvidia driver.   To see this, search for card0 and card1 in the environment output.  All monitors are connected to card0, the i915 "card", not card1, the nvidia card.  I need to improve the environment command to better indicate which hardware/driver is providing which I2c device. 

Can you write up the steps you went through to build the custom kernel with the patch?  That would be generally useful.

rockowitz, 2020-10-11:
@JustGitting. That the drm patch does not explain the absence of I2C devices for the docking station on your system, but solves the problem for i915, suggests your problem is amdgpu specific.  I'd suggest raising the question on an amdgpu mailing list and/or filing a bug report.

JustGitting, 2020-10-11:
Hi @rockowitz,

I'm sorry but I don't understand how amdgpu is involved, given the System76 Darter Pro does not have any AMD components. The Darter Pro (using a i7-10510U CPU) only has an iGPU provided by Intel's UHD Graphics as listed above ( Device: x9b41 UHD Graphics ).

Does the amdgpu kernel module provide other services that is even used by Intel only systems?

rockowitz, 2020-10-11:
My apologies.  I was thinking that you had a system with an amd cpu/gpu when I wrote the response.

I also realize that I may have misunderstood what you mean when you write that you "cannot detect and use the external monitors".  This could mean several things: 
1) The monitors do not appear to exist.  xrandr does not detect them.
2) The monitors appear to exist, but no image displays on them. 
3) An image displays, but there appears to be no I2C devices associated with the monitor, 
   i.e. an I2C device that reports the EDID at slave address x50
4) The EDID can be read over i2c, but slave address x37 for DDC does not exist. 
    i2c_detect does  not find slave address x37. 
5) Slave address x37 exists, but DDC communication fails.

If (1) or (2), the problem is outside my area of expertise.
Otherwise, please run "sudo ddcutil environment --verbose" for both PopOs and Devuan and include the output as attachments. Ensure that command i2c_detect (package i2c-tools) exists.

JustGitting, 2020-10-11:
Hi @rockowitz,

No worries :+1: 

My problem is 1) The monitors do not appear to exist. xrandr does not detect them (when using Devuan/Debian)

Whilst in PopOS the monitors exist (detected by xrandr) and can be used automatically without any intervention. 

There are no errors or obvious messages in the kernel logs in Devuan as to why the monitors were not detected. The only error I've been able to find so far was detected by your wonderful tool, ddcutil, with the i2c bus. Hence, I hoped that given your understanding of the system you might be able to help.

Q. Do you know any references that explain how monitors are detected and used through a thunderbolt 3 dock in Linux?

The only reference I can find about how the different components fit together is from the [thunderspy website](https://thunderspy.io/). e.g. https://thunderspy.io/assets/img/tb3-controller-architecture-alpine-ridge.png

It's a bit frustrating that there is no clear explanation or overview of the Linux architecture and how it works to detect and use different components and hardware...it's almost as opaque as MS Windows :)

No wonder there is yet to be the mythical "year of the GNU/Linux desktop".

lauritsriple, 2020-10-13:
Sorry for hijaking your issue, @JustGitting,I thought your issue was related to the patch of the i915drm driver.  

@rockowitz I used the Arch Build System (https://wiki.archlinux.org/index.php/Kernel/Arch_Build_System) to build the kernel. Followed the steps there. Before building I downloaded the patch for **issue 37** on [patchwork](https://patchwork.freedesktop.org/patch/388575/) (there is a button there) saved it in the same folder as the PKGBUILD. Edited the PKGBUILD where I added a entry for the name of the patch file under sources, and added a `SKIP` on the sha256sum. That part of the PKGBUILD file now looked something like this for me:
```
source=(
  "$_srcname::git+https://git.archlinux.org/linux.git?signed#tag=$_srctag"
  config         # the main kernel config file
  sphinx-workaround.patch
  drm-dp_mst-Support-remote-i2c-writes.patch                                                                                          
)

... validpkgKeys lines between here ...

sha256sums=('SKIP'
  '40c80a5e6ef76702021abd10e5a9bf07c68530cb2c4868154c1b5adf4bf3b917'
  '8cb21e0b3411327b627a9dd15b8eb773295a0d2782b1a41b2a8839d1b2f5778c'
  'SKIP'
   )
```
Also change `pkgbase=linux` to `pkgbase=linux-custom` in the PKGBUILD to get another name for it.
Then I built with `makepkg -s` and installed with `pacman` after a couple of hours (as indicated on the arch wiki page). Probably a faster way to do this, but I don't know how. 

Oh, and I added a entry to my bootloader to be able to run the custom kernel after reboot. In `systemd-boot` that would be to make a file `/boot/loader/entries/customkernel.conf` with the following content:
```
title Arch Linux drmPatch                                                                                                                                                                                                         
linux /vmlinuz-linux-custom
initrd /initramfs-linux-custom.img
options root=UUID=PLACE_YOUR_UUID_HERE rw

```

closed 2020-10-16 by JustGitting

rockowitz, 2020-10-19:
@lauritsriple First, thank you for posting the detailed description of how you built a kernel with the i915 patch. 

The latest 1.0.0-dev branch contains additional probing of i2c related attributes in the /sys file system.  I'd appreciate it if you could build it and send the output from "ddcutil environment --verbose".  I believe it will enable me to more properly address the problem of the same monitor appearing on 2 different i2c buses. Thanks.

lauritsriple, 2020-10-19:
Built the 1.0.0-dev branch. Output from `ddcutil environment --verbose` attached in logfile.
[ddc_environment_1.0.0-dev.txt](https://github.com/rockowitz/ddcutil/files/5402008/ddc_environment_1.0.0-dev.txt)

rockowitz, 2020-10-24:
@lauritsriple Can I ask you to build the latest 1.0.0-dev branch, run "sudo ddcutil environment --verbose" and send the output?  Thanks.

lauritsriple, 2020-10-26:
I updated my source from git (commit: b51e28c11aaeceaf4fa047eead4f3a4f05cc0f00) and compiled again. Output attached. 
[ddc_environment_1.0.0-dev.txt](https://github.com/rockowitz/ddcutil/files/5438154/ddc_environment_1.0.0-dev.txt)

rockowitz, 2020-10-30:
Can I ask you to build the latest ddcutil 1.0.0-dev branch?  This time 
please run command "sudo ddcutil detect --vv".  ("--vv""  stands for 
"very verbose"). Unless your display configuration has changed, there's 
no need to run "sudo ddcutil environment --verbose" again, as I have 
that output.  Thanks.

lauritsriple, 2020-11-05:
Hi! Sure! I seem to be getting some error with `--vv` attached output from `-v` as well
[ddc_detect_verbose_1.0.0-dev.txt](https://github.com/rockowitz/ddcutil/files/5493743/ddc_detect_verbose_1.0.0-dev.txt)
[ddc_detect_veryVerbose_1.0.0-dev.txt](https://github.com/rockowitz/ddcutil/files/5493744/ddc_detect_veryVerbose_1.0.0-dev.txt)

rockowitz, 2020-11-07:
The /sys query tripped up because I hadn't handled the case of the eDP 
display in the laptop.  A fix for this, along with some cleanup, is now 
in branch 1.0.0-dev. Can I ask you to run "sudo ddcutil detect -vv" and 
"sudo ddcutil environment --verbose".  Thanks

lauritsriple, 2020-11-25:
Ey! I forgot about this. Ran the commands now. Here are the output.
[ddc_detect_verbose_1.0.0-dev.txt](https://github.com/rockowitz/ddcutil/files/5596009/ddc_detect_verbose_1.0.0-dev.txt)
[ddc_detect_veryVerbose_1.0.0-dev.txt](https://github.com/rockowitz/ddcutil/files/5596010/ddc_detect_veryVerbose_1.0.0-dev.txt)
[ddc_environment_1.0.0-dev.txt](https://github.com/rockowitz/ddcutil/files/5596011/ddc_environment_1.0.0-dev.txt)

rockowitz, 2020-11-29:
Which connector on the dock is the display with serial number ending 
70Z5L connected to, and which is the one with serial number ending I0GCL 
connected to?

rockowitz, 2020-11-29:
Also, what is the docking station manufacturer/model?